Koenkk/zigbee2mqtt #13041: IKEA E1812 action not resetting since 1.26.0

eloo, 2022-07-05:
### What happened?

Hi, 
since the update to 1.26.0 i see an unexpected behaviour for my ikea shortcut buttons which is also breaking my automations.

When the button is pressed i see a mqtt message like this:
```
{"action":"on","battery":87,"last_seen":"2022-07-05T09:44:32+02:00","linkquality":255,"update":{"state":"idle"},"update_available":false}
```

but no "clearing" message afterwards which has no action field set.

Comparing to the messages of my mijia switch which sends a message with "click" and then one without "click" i would expect to get here also a "clearing" message.

Further the automation i'm using is also expecting this clearing message so i would guess it was available in 1.25.0 
https://github.com/EPMatt/awesome-ha-blueprints/blob/main/blueprints/controllers/ikea_e1812/ikea_e1812.yaml#L197

Any ideas what could have changed here?

Best regards
eloo

### Zigbee2MQTT version

1.26.0

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

eloo, 2022-07-05:
Here is the debug log
```
debug 2022-07-05 20:24:10: Received Zigbee message from 'IKEA ShortCut 3', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID null
info  2022-07-05 20:24:11: MQTT publish: topic 'zigbee2mqtt/IKEA ShortCut 3', payload '{"action":"on","battery":87,"last_seen":"2022-07-05T20:24:10+02:00","linkquality":255,"update":{"state":"idle"},"update_available":false}'
info  2022-07-05 20:24:11: MQTT publish: topic 'zigbee2mqtt/IKEA ShortCut 3', payload '{"action":"","battery":87,"last_seen":"2022-07-05T20:24:10+02:00","linkquality":255,"update":{"state":"idle"},"update_available":false}'
info  2022-07-05 20:24:11: MQTT publish: topic 'zigbee2mqtt/IKEA ShortCut 3/action', payload 'on'
debug 2022-07-05 20:24:16: Received Zigbee message from 'IKEA ShortCut 3', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID null
info  2022-07-05 20:24:17: MQTT publish: topic 'zigbee2mqtt/IKEA ShortCut 3', payload '{"action":"on","battery":87,"last_seen":"2022-07-05T20:24:16+02:00","linkquality":255,"update":{"state":"idle"},"update_available":false}'
info  2022-07-05 20:24:17: MQTT publish: topic 'zigbee2mqtt/IKEA ShortCut 3', payload '{"action":"","battery":87,"last_seen":"2022-07-05T20:24:16+02:00","linkquality":255,"update":{"state":"idle"},"update_available":false}'
info  2022-07-05 20:24:17: MQTT publish: topic 'zigbee2mqtt/IKEA ShortCut 3/action', payload 'on'
debug 2022-07-05 20:24:20: Received Zigbee message from 'IKEA ShortCut 3', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID null
info  2022-07-05 20:24:21: MQTT publish: topic 'zigbee2mqtt/IKEA ShortCut 3', payload '{"action":"on","battery":87,"last_seen":"2022-07-05T20:24:20+02:00","linkquality":255,"update":{"state":"idle"},"update_available":false}'
info  2022-07-05 20:24:21: MQTT publish: topic 'zigbee2mqtt/IKEA ShortCut 3', payload '{"action":"","battery":87,"last_seen":"2022-07-05T20:24:20+02:00","linkquality":255,"update":{"state":"idle"},"update_available":false}'
info  2022-07-05 20:24:21: MQTT publish: topic 'zigbee2mqtt/IKEA ShortCut 3/action', payload 'on'
```

i have grepped for the the ikea shortcut button

eloo, 2022-07-05:
ah okay.. it seems to work again..

maybe this was related to the disabled homeassistant setting...

sorry for that but it seems to be a side affect of the lost configuration from the file based addon config to the application based one. 

i had to reconfigure some settings.. seems like the homeassistant config was also affected

Koenkk, 2022-07-06:
Enabling the HA integration will indeed clear the actions, so that was the root cause :)

closed 2022-07-06 by Koenkk