Koenkk/zigbee2mqtt #13044: 1.26 breaks IKEA E1524/E1810 compatibility

ministryofsillywalks, 2022-07-05:
### What happened?

I upgraded to 1.26 and all my IKEA E1524/E1810 stopped working.
They were still connected and reporting battery and LQI but no actions were being sent when pressing a button.
Nothing was to be seen in the logs. 
Downgraded to 1.25.2 and everything is once again working as before.

### What did you expect to happen?

The Remotes should continue sending actions

### How to reproduce it (minimal and precise)

Upgrade from 1.25.2 to 1.26

### Zigbee2MQTT version

1.26

### Adapter firmware version

6.7.8.0 build 373

### Adapter

EZSP v8 (Sonoff Bridge)

oystermon, 2022-07-07:
Hi I recently updated to 1.26 and I got the following in the log for my E1524/E1810 remote:

`Error 2022-07-07 21:23:48Failed to configure 'Ikea_Remote', attempt 1 (Error: Bind 0x84fd27fffe2469b8/1 genOnOff from '0x00124b00258d4225/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:559:17) at processTimers (node:internal/timers:502:7))`

But it still seems to work ok. Will report back if I have any issues.

Using Sonoff Zigbee Dongle USB 3.0 Plus.

4tbs, 2022-07-24:
Also found some issues with these remotes. Just switched from Deconz to Zigbee2mqtt for my Home assistant instance and after joining all devices to Zigbee2mqtt all worked fine. I then started to update fw on the devices (which solved some battery level reporting issues), but after that all E1524/E1810 remotes failed to work - with one exception. The middle button still worked, but not the other buttons. Update was from 2.2.? to 2.3.080.
I then tried to delete and re-join the devices, and that solved the issue. After that all seems to work just fine.

MattWestb, 2022-07-25:
Some devices then updating the firmware is changing its configuration and need being reconfigured for working OK and the last IKEA controllers is one of this,
Also the controllers (not styrbar) is loosing group binding in the last OTA and need doing new bindings with devices instead.

prutseltje, 2022-08-22:
I have tried to reconfigure the device, but no luck.

ahd71, 2022-08-29:
I have the same or similar problem. Several of my remotes are online but do not send any keypresses. All are on same firmware, some few works. I'm on 1.27 and it stopped working at that or 1.26. Rejoining them doesn't solve the problem.

MattWestb, 2022-08-29:
Perhaps updating the very old ZLL firmware to new Zigbee 3 one and getting many bugs away in the device ??

closed 2022-10-07 by github-actions[bot]