Koenkk/zigbee2mqtt #13054: GLEDOPTO GL-S-004Z - it does not support `color_temp_startup`

gorbunovav, 2022-07-05:
### What happened?

I see the error `UNSUPPORTED_ATTRIBUTE` on attempt to set this parameter.

### Zigbee2MQTT version

1.26.0 commit: cd23aea

### Adapter firmware version

20220219

### Adapter

CC2652P stick

gorbunovav, 2022-07-05:
@Koenkk I've noticed that the device definition lacks the `colorTemp` parameter:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/bb01ed184425e7a000437c0a6ed253354edff3a7/devices/gledopto.js#L351

`colorTempPhysicalMin` and `colorTempPhysicalMax` values reported by the bulb are `155` and `495` (the device manual states that the supported color range is `2700 - 6500K`).

Is it safe to add these values to the profile?

Koenkk, 2022-07-06:
> Is it safe to add these values to the profile?

Yes

closed 2022-07-06 by Koenkk