Koenkk/zigbee2mqtt #13069: Bosch Plug Compact BSP-FZ2

UrsusS, 2022-07-07:
### Link

https://www.bosch-smarthome.com/uk/en/products/devices/plug-compact/

### Database entry

{"friendly_name":"0x50325ffffe72b022","ieee_address":"0x50325ffffe72b022",

### Comments

I tried to to pair the plug, but the interview fails.
According the FAQ from Bosch, thew plug should be able to work as range extender for zigbee.

```
info  2022-06-21 19:55:16: Device '0x50325ffffe72b022' joined
info  2022-06-21 19:55:16: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x50325ffffe72b022","ieee_address":"0x50325ffffe72b022"},"type":"device_joined"}'
info  2022-06-21 19:55:16: MQTT publish: topic 'zigbee2mqtt/0x50325ffffe72b022/availability', payload 'online'
info  2022-06-21 19:55:16: Starting interview of '0x50325ffffe72b022'
info  2022-06-21 19:55:16: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x50325ffffe72b022","ieee_address":"0x50325ffffe72b022","status":"started"},"type":"device_interview"
}'
debug 2022-06-21 19:57:12: Saving state to file /opt/zigbee2mqtt/data/state.json
error 2022-06-21 19:57:35: Failed to interview '0x50325ffffe72b022', device has not successfully been paired
info  2022-06-21 19:57:35: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x50325ffffe72b022","ieee_address":"0x50325ffffe72b022","status":"failed"},"type":"device_interview"}'
info  2022-06-21 19:57:40: Starting interview of '0x50325ffffe72b022'
info  2022-06-21 19:57:40: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x50325ffffe72b022","ieee_address":"0x50325ffffe72b022","status":"started"},"type":"device_interview"}'

```

UrsusS, 2022-07-08:
Here you'll find the debuglog
[debuglog.txt](https://github.com/Koenkk/zigbee2mqtt/files/9074390/debuglog.txt)

Koenkk, 2022-07-09:
Seems the device doesn't connect at all, can you try pairing it closer to the coordinator?

UrsusS, 2022-07-09:
I moved it now quite close to the coordinator. But it seems to as nothing has changed
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9077776/log.txt)
[debuglog1.txt](https://github.com/Koenkk/zigbee2mqtt/files/9077777/debuglog1.txt)

UrsusS, 2022-07-27:
I've updated the firmware, but that didn't change anything
I'll try the sniffing, but this will take a while until I'll do it

closed 2022-09-04 by github-actions[bot]

danieledwardgeorgehitchcock, 2022-09-06:
@UrsusS - you might want to check out the activity that has been going on in #13520 - There is now support for adding install codes which should get you passed the interview stage so that you can get a proper DB entry

UrsusS, 2022-10-01:
@danieledwardgeorgehitchcock - thanks, I followed the procedure at #13520 so I was able to pair.
So I was following the "support new devices" procedure, but I'm stuck as it does not complain about missing converters.

```
debug 2022-10-01 15:21:30: Received Zigbee message from '0x50325ffffe72b022', type 'read', cluster 'genBasic', data '["zclVersion"]' from endpoint 1 with groupID 0
debug 2022-10-01 15:21:30: Received Zigbee message from '0x50325ffffe72b022', type 'read', cluster 'genBasic', data '["zclVersion"]' from endpoint 1 with groupID 0
debug 2022-10-01 15:21:30: Received Zigbee message from '0x50325ffffe72b022', type 'read', cluster 'genBasic', data '["zclVersion"]' from endpoint 1 with groupID 0
debug 2022-10-01 15:21:30: Received Zigbee message from '0x50325ffffe72b022', type 'read', cluster 'genBasic', data '["zclVersion"]' from endpoint 1 with groupID 0
debug 2022-10-01 15:21:30: Received Zigbee message from '0x50325ffffe72b022', type 'read', cluster 'genBasic', data '["zclVersion"]' from endpoint 1 with groupID 0
debug 2022-10-01 15:21:30: Received Zigbee message from '0x50325ffffe72b022', type 'read', cluster 'genBasic', data '["zclVersion"]' from endpoint 1 with groupID 0
debug 2022-10-01 15:21:30: Received Zigbee message from '0x50325ffffe72b022', type 'read', cluster 'genBasic', data '["zclVersion"]' from endpoint 1 with groupID 0
debug 2022-10-01 15:21:30: Received Zigbee message from '0x50325ffffe72b022', type 'read', cluster 'genBasic', data '["zclVersion"]' from endpoint 1 with groupID 0
debug 2022-10-01 15:21:30: Received Zigbee message from '0x50325ffffe72b022', type 'read', cluster 'genBasic', data '["zclVersion"]' from endpoint 1 with groupID 0
debug 2022-10-01 15:21:30: Received Zigbee message from '0x50325ffffe72b022', type 'read', cluster 'genBasic', data '["zclVersion"]' from endpoint 1 with groupID 0
debug 2022-10-01 15:21:30: Received Zigbee message from '0x50325ffffe72b022', type 'read', cluster 'genBasic', data '["zclVersion"]' from endpoint 1 with groupID 0
debug 2022-10-01 15:21:30: Received Zigbee message from '0x50325ffffe72b022', type 'read', cluster 'genBasic', data '["zclVersion"]' from endpoint 1 with groupID 0
debug 2022-10-01 15:21:30: Received Zigbee message from '0x50325ffffe72b022', type 'read', cluster 'genBasic', data '["zclVersion"]' from endpoint 1 with groupID 0
debug 2022-10-01 15:21:30: Received Zigbee message from '0x50325ffffe72b022', type 'read', cluster 'genBasic', data '["zclVersion"]' from endpoint 1 with groupID 0
debug 2022-10-01 15:21:31: Device '0x50325ffffe72b022' announced itself
info  2022-10-01 15:21:31: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x50325ffffe72b022","ieee_address":"0x50325ffffe72b022"},"type":"device_announce"}'
debug 2022-10-01 15:21:33: Received Zigbee message from '0x50325ffffe72b022', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":621110800,"imageType":12290,"manufacturerCode":4617}' from endpoint 1 with groupID 0
debug 2022-10-01 15:21:33: Device '0x50325ffffe72b022' requested OTA
debug 2022-10-01 15:21:33: Retrieving state of '0x50325ffffe72b022' after reconnect
```

danieledwardgeorgehitchcock, 2022-10-03:
Could you post up your database.db entry so I can have a look at the device state, etc?

on-jz, 2022-10-08:
I started using Home Assistant about a week ago and I'm not a pro in Javascript. This is my very first attempt to dive deep into stuff under the hood here. So the converter might be improved by more savvy people out there. "Settings (specific)" in Zigbee4MQTT offers "power_calibration", "current_calibration",  "voltage_calibration" which are afaik all pointless for this device. 

I bought the plug because it's specified as 16A max. To be honest I am pretty disappointed as compared to the e.g. Aqara EU smart plug (lumi.plug.maeu01) which is only 10A max. readings and settings are poor. Most disappointing atm is that I could not find any option to set "power_outage_memory" to on/off/last state. Missing "device_temperature" for a wall plug also is a minus in my opinion. 

All the findings and the code are status "works for me" for der German schuko version of the Plug (Bosch Smart Home Plug Compact, Model no.: BSP-FZ2, Manufacturer no.: 8750001300; EAN: 4062321299019)

Anyway, I hope the stuff below is helpful.

### Link

**Device:**
-  https://www.bosch-smarthome.com/uk/en/products/devices/plug-compact/ 
-  https://www.bosch-smarthome.com/de/de/produkte/geraete/zwischenstecker-kompakt/
 
**Device FAQs:** 
- https://www.bosch-smarthome.com/uk/en/service/help/product-help/plug-compact-help/
- https://www.bosch-smarthome.com/de/de/service/hilfe/hilfe-zum-produkt/hilfe-zum-zwischenstecker-kompakt/

**Device manuals:**
- https://tt-smarthome.resource.bosch.com/nrt/web/produkt/plug_compact/BSP-QSG_EN_FR_DK_OSS_20200924_web.pdf
- https://tt-smarthome.resource.bosch.com/nrt/web/produkt/plug_compact/BSP-QSG_DE_IT_ES_PT_20200924_web.pdf

### Database entry

```json
{
	"id": 32,
	"type": "Router",
	"ieeeAddr": "0x0000000000000000",
	"nwkAddr": 462,
	"manufId": 4617,
	"manufName": "BOSCH",
	"powerSource": "Mains (single phase)",
	"modelId": "RBSH-SP-ZB-EU",
	"epList": [
		1,
		242
	],
	"endpoints": {
		"1": {
			"profId": 260,
			"epId": 1,
			"devId": 81,
			"inClusterList": [
				0,
				3,
				4,
				5,
				6,
				1794,
				2820,
				2821
			],
			"outClusterList": [
				3,
				10,
				25
			],
			"clusters": {
				"seMetering": {
					"attributes": {
						"currentSummDelivered": [
							0,
							8
						],
						"multiplier": 1,
						"divisor": 100
					}
				},
				"haElectricalMeasurement": {
[... 200 lines cut ...]
        await reporting.onOff(endpoint);

		try {
			await reporting.bind(endpoint, coordinatorEndpoint, ['seMetering']);
			await reporting.readMeteringMultiplierDivisor(endpoint);
			await reporting.currentSummDelivered(endpoint, {change: 0});
		} catch (e) {
			logger.warn(`RBSH-SP-ZB-EU failed to setup metering (${e.message})`);
			logger.debug(e.stack);
		}

		try {
			await reporting.bind(endpoint, coordinatorEndpoint, ['haElectricalMeasurement']);
			// await endpoint.read('haElectricalMeasurement', ['acPowerMultiplier', 'acPowerDivisor', 'activePower']);
			await endpoint.read('haElectricalMeasurement', ['acPowerMultiplier', 'acPowerDivisor']);
			await reporting.activePower(endpoint, {change: 0});
		} catch (e) {
			logger.warn(`RBSH-SP-ZB-EU failed to setup electricity measurements (${e.message})`);
			logger.debug(e.stack);
		}
    },
};

module.exports = definition;
```

stefano-bortolotti, 2023-07-14:
Hi, 

I bought a UK version(BSP-GZ2) of this device and thanks to the above info I managed to set it up in my Home Assistant. 
I only had to change the name of the JS file ("RBSH-SP-ZB-GB.js") and change the references in the JS file: 

```javascript
// /**
//  * Bosch Smart Home Plug Compact, Model no.: BSP-GZ2, Manufacturer no.: 8750001300; EAN: 4062321299019
//  * 
//  * @file config/zigbee2mqtt/RBSH-SP-ZB-GB.js
//  * @version 0.0.1
//  * @since 20221008
//  * @author https://github.com/on-jz
//  */
// 
// ## Device:
// -  https://www.bosch-smarthome.com/uk/en/products/devices/plug-compact/ 
// -  https://www.bosch-smarthome.com/de/de/produkte/geraete/zwischenstecker-kompakt/
//  
// ## Device FAQs: 
// - https://www.bosch-smarthome.com/uk/en/service/help/product-help/plug-compact-help/
// - https://www.bosch-smarthome.com/de/de/service/hilfe/hilfe-zum-produkt/hilfe-zum-zwischenstecker-kompakt/
//
// ## Device manuals:
// - https://tt-smarthome.resource.bosch.com/nrt/web/produkt/plug_compact/BSP-QSG_EN_FR_DK_OSS_20200924_web.pdf
// - https://tt-smarthome.resource.bosch.com/nrt/web/produkt/plug_compact/BSP-QSG_DE_IT_ES_PT_20200924_web.pdf
//
// ## Setup Zigbee2MQTT in Home Assistant:
// - save this file to Home Assistant "config/zigbee2mqtt/RBSH-SP-ZB-GB.js"
// - in Home Assistant Web UI
//   go to Zigbee2MQTT > Settings > Settings > External converters > click [+]
// - enter "RBSH-SP-ZB-GB.js" in text input
// - click [Submit]
// - go to Home Assistant Web UI > Settings > Add-ons > Zigbee2MQTT > click [RESTART]
// 
// ## Paring:
// read this: https://github.com/Koenkk/zigbee2mqtt/issues/13520#issuecomment-1228882633
// - BEFORE pairing: in Home Assistant Web UI go to: Zigbee2MQTT > Settings > Tools > click [add install code]
// - scan QR code from bottom of device
// - paste entire string read from QR code into text field "enter install code"
// - click [OK]
// - in Zigbee2MQTT click [Permit join (All)] 
// - press button on device until led flashes
// - wait for interview to finish
// 
// ## Tested and working with:
// - Rapsberry Pi 4
// - Sonoff Zigbee 3.0 USB Dongle Plus CC2652P + CP2102(N): Z-Stack-firmware/coordinator/Z-Stack_3.x.0/20220219
// - Home Assistant: 2022.10.1
// - Mosquitto broker: 6.1.3
// - Zigbee2MQTT: 1.28.0-1
// 
// ## Known issues / findings:
// - device does not seem to support reading device_temperature 
// - device does not seem to support reading voltage
// - device does not seem to support reading current
// - device does not seem to support reading/setting power_outage_memory (no information found at Bosch either)
// - electricity measurement "power" (unit W) is reported as int only

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const globalStore = require('zigbee-herdsman-converters/lib/store');

const definition = {
    zigbeeModel: ['RBSH-SP-ZB-GB'],
    model: 'BSP-GZ2',
    vendor: 'Bosch Smart Home',
    description: 'Bosch Smart Home Plug Compact UK',
    toZigbee: [tz.on_off],
    fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering],
	exposes: [e.switch(), e.power(), e.energy()],
    // The configure method below is needed to make the device reports on/off state changes
    // when the device is controlled manually through the button on it.
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(endpoint);

		try {
			await reporting.bind(endpoint, coordinatorEndpoint, ['seMetering']);
			await reporting.readMeteringMultiplierDivisor(endpoint);
			await reporting.currentSummDelivered(endpoint, {change: 0});
		} catch (e) {
			logger.warn("RBSH-SP-ZB-GB failed to setup metering (${e.message})");
			logger.debug(e.stack);
		}

		try {
			await reporting.bind(endpoint, coordinatorEndpoint, ['haElectricalMeasurement']);
			// await endpoint.read('haElectricalMeasurement', ['acPowerMultiplier', 'acPowerDivisor', 'activePower']);
			await endpoint.read('haElectricalMeasurement', ['acPowerMultiplier', 'acPowerDivisor']);
			await reporting.activePower(endpoint, {change: 0});
		} catch (e) {
			logger.warn("RBSH-SP-ZB-GB failed to setup electricity measurements (${e.message})");
			logger.debug(e.stack);
		}
    },
};

module.exports = definition;
```

Koenkk, 2023-07-15:
@stefano-bortolotti thanks, added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

stefano-bortolotti, 2023-07-24:
> @stefano-bortolotti thanks, added!
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

thank for that @Koenkk 
Here is the product image if you need to add it to the products DB: 
[image]