Koenkk/zigbee2mqtt #13070: Müller Light remote 404011 no longer working

MHeuvel, 2022-07-07:
### What happened?

After a re-install of Zigbee2MQTT (latest) the Muller 404011 does pair (as Muller 404011/4040(something) but no buttons in any group show up in MQTT. Reverted to version 1.24 without any configuration changes and the same setup works.

### How to reproduce it (minimal and precise)

Install newest version and it will stop working

### Zigbee2MQTT version

1.24.p

### Adapter firmware version

6.7.6.0 build 327

### Adapter

Zbbridge

Koenkk, 2022-07-08:
@kirovilya I think this is due to https://github.com/Koenkk/zigbee-herdsman/pull/551 . Seems something goes wrong with the group message handling.

kirovilya, 2022-07-09:
@MHeuvel Can you describe the problem in more detail?
I have groups works. What errors appear in the logs? Please record debug logs

MHeuvel, 2022-07-10:
I would like to give more input if I could. There are no errors, not even any messages in the debug logging. Like buttons are never pressed.

closed 2022-08-19 by github-actions[bot]