Koenkk/zigbee2mqtt #13071: Failed to Start - Error: Error: Is a directory, cannot open /dev/ttyUSB0

abishur, 2022-07-07:
### What happened?

I'm running Zigbee2MQTT on Truenas.  It was working but I turned it off and installed a new 5 port USB PCIe card.  I plugged my sonoff 3.0 zigbee stick into the same location, but when I start my container I get the following in my log

```
2022-07-07 20:50:30.560943+00:00Zigbee2MQTT:info  2022-07-07 15:50:30: Starting Zigbee2MQTT version 1.25.2 (commit #3c5854fa)
2022-07-07 20:50:30.563544+00:00Zigbee2MQTT:info  2022-07-07 15:50:30: Starting zigbee-herdsman (0.14.34)
2022-07-07 20:50:30.565886+00:00Zigbee2MQTT:debug 2022-07-07 15:50:30: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/data/coordinator_backup.json","databaseBackupPath":"/data/database.db.backup","databasePath":"/data/database.db","network":{"channelList":[25],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"auto","path":"/dev/ttyUSB0"}}'
2022-07-07 20:50:31.006327+00:00Zigbee2MQTT:error 2022-07-07 15:50:31: Error while starting zigbee-herdsman
2022-07-07 20:50:31.007302+00:00Zigbee2MQTT:error 2022-07-07 15:50:31: Failed to start zigbee
2022-07-07 20:50:31.007866+00:00Zigbee2MQTT:error 2022-07-07 15:50:31: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
2022-07-07 20:50:31.008629+00:00Zigbee2MQTT:error 2022-07-07 15:50:31: Exiting...
2022-07-07 20:50:31.043749+00:00Zigbee2MQTT:error 2022-07-07 15:50:31: Error: Error while opening serialport 'Error: Error: Is a directory, cannot open /dev/ttyUSB0'
2022-07-07 20:50:31.043821+00:00at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:146:28)
2022-07-07 20:50:31.043834+00:00at SerialPort._error (/app/node_modules/@serialport/stream/lib/index.js:198:14)
2022-07-07 20:50:31.043864+00:00at /app/node_modules/@serialport/stream/lib/index.js:242:12
```

I thought maybe I was on a new port since I installed the new pcie card, but doing a ls -l /dev/serial/by-id I get

```
lrwxrwxrwx 1 root root 13 Jul  7 15:39 usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_748ce0695198ec11a2a5aad044d80d13-if00-port0 -> ../../ttyUSB0
```

So I'm still on USB0.  What am I doing wrong all of the sudden?  The only thing I haven't tried yet it removing the new pcie card, I won't be able to try that until late tonight since the server is actively being used.

### Zigbee2MQTT version

1.25.2

### Adapter firmware version

not certain

### Adapter

Sonoff 3.0

ashokaiyar, 2022-07-08:
What are the permissions and ownership on `/dev/ttyUSB0`. For example, this is what it looks like on my system:

`crw-rw---- 1 root dialout 188, 0 Jul  7 21:37 /dev/ttyUSB0`

abishur, 2022-07-08:
the permissions are the same on mine.  Also, I was able to reboot the machine and the problem vanished.  I've tried to get it to come back, but so far nothing I've done has been able to reproduce my own issue 🤦‍♂️If I can't get it to come back in the next day or two I'll close the issue.

ashokaiyar, 2022-07-11:
@abishur 
Can you close this issue if it hasn't reproduced?

abishur, 2022-07-11:
You bet, I've tried everything I could to reproduce it, I even reinstalled truenas and imported in my container to see if I could break it that way, but it loaded up just fine.  It must have been a random fluke, thanks!

closed 2022-07-11 by abishur

MattInternet, 2023-03-07:
Just encountered this same issue, and it was fixed by a reboot.

Running `ls -l /dev/serial/by-id` gave me the expected output: `lrwxrwxrwx 1 root root 13 Mar  6 18:17 usb-1a86_USB_Serial-if00-port0 -> ../../ttyUSB0`

But the tell was running `ls -l /dev/ttyUSB0`. This should give you like permission and stuff but mine was blank

rclarsfull, 2023-12-07:
Im having this issue now for cule op days and cant resume it. I run trunas scale as host system. It runs as application. This Error only happens when I bind the port in the pod, to the port /dev/serial/by-id/... on the host machine. When i bind it to an /dev/ttyACM0 i geht the error message: "Error: Is a directory," im a little helpless at the moment. Do you kow what the problem could be? btw i alredy did chmod 777 (port) to ensure its not en rights problem

XSAM, 2025-09-29:
I encountered the same issue recently. I was able to fix it without needing to reboot.

It is actually caused by the behavior of the container (containerd or Docker) when you are trying to mount `/dev/ttyUSB0`. When you mount `/dev/ttyUSB0` by using `-v /dev/ttyUSB0:/dev/ttyUSB0`, the container would create a folder if `/dev/ttyUSB0` does not exist.

My theory is that at some point, the USB connection somehow became disconnected. During this short moment, the `zigbee2mqtt` container restarts due to the lost device. And, this restart triggers the folder creation for `/dev/ttyUSB0`, as this path does not exist anymore (USB device disconnected). After `zigbee2mqtt` starts, it finds that it is reading a folder instead of a device, so it throws the error and exits.

**For those who want to resolve this issue without rebooting**:
- Stop `zigbee2mqtt` container (this prevents it from continuously creating the `/dev/ttyUSB0` folder).
- Remove `/dev/ttyUSB0` folder.
- Disconnect the USB device, and reconnect the USB device to the host. This should make the functional `/dev/ttyUSB0` entry show again.
- Start `zigbee2mqtt` container.

**A long-term fix (quick way)**: mount `/dev` instead of `/dev/ttyUSB0`, although it may raise security concerns by exposing the entire `dev` folder.