Koenkk/zigbee2mqtt #13073: Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms - Zigbee2mqtt Crashing

Devon1984, 2022-07-07:
### What happened?

After upgrading from 1.25.2 to 1.26.0 my zigbee2mqtt container crashes at least a few times per day.

### What did you expect to happen?

Nothing. Just a smooth upgrade.

### How to reproduce it (minimal and precise)

Upgrade from 1.25.2 to 1.26.0 and let it run.

### Zigbee2MQTT version

1.26.0

### Adapter firmware version

20220219

### Adapter

Electrolama zig-a-zig-ah! CC2652R

### Debug log

[08-07-22-log.zip](https://github.com/Koenkk/zigbee2mqtt/files/9067838/08-07-22-log.zip)

Devon1984, 2022-07-07:
Hi @Koenkk since updating to the new Z2M version of 1.26.0 Z2M keeps on crashing daily.
I have attached the Herdsman logs for you but the snippet of the error is below.
My coordinator is still on this FW version https://github.com/Koenkk/zigbee2mqtt/issues/12439#issuecomment-1136258830

2022-07-07T15:12:01.812Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtLeaveReq - {"dstaddr":40538,"deviceaddress":"0x0000000000000000","removechildrenRejoin":0}
2022-07-07T15:12:01.812Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,11,37,52,90,158,0,0,0,0,0,0,0,0,0,222]
Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:559:17)
    at processTimers (node:internal/timers:502:7)

Im not sure if the issue is caused by this but i have a spurious device "deviceaddress":"0x0000000000000000" that tries to join every so often so i have placed it on the banned list. This device just started to appear for no reason 1 day after an update.

Also, for a long time now my samsung door sensors kept dropping off of my zigbee network for no reason so i have been leaving my "allow zigbee devices to join" set to always allow which has mostly fixed the issue and i do see devices drop off and then rejoin each day or so... so not sure if this is also contributing to the issue.

Hoping you can help me out again this time with this :-)

Koenkk, 2022-07-09:
Should be fixed now, please confirm.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Devon1984, 2022-07-10:
Legend!! As always, thanks for the quick responce and the  quick fix @Koenkk  :-)
Iv updated to the latest dev branch and will monitor and report back.

Devon1984, 2022-07-14:
Since updating to the new dev version of Z2M this issue is now solved.
Thanks @Koenkk

closed 2022-07-14 by Devon1984

silkyclouds, 2024-09-17:
Gents,

since I removed some of my devices to try to fix things today, I also have this happening. 

zh:controller: Failed to remove rejected device: Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms

Current version: 1.40.1-1