Koenkk/zigbee2mqtt #13074: Log via syslog suddenly stop

zen2, 2022-07-08:
### What happened?

My zigbee2mqtt configuration related to logging was:
```
  log_level: info
  log_directory: data
  log_file: log.txt
  log_output:
    - syslog
  log_syslog:
    protocol: udp
    host: 127.0.0.1
    port: 514
    path: /dev/log
    app_name: zigbee2mqtt
    eol: /n
    localhost: localhost
    pid: process.pid
    type: '5424'
```

Everything was logging perfectly for days then I have configured Availability (advanced) via the frontend:
```
  active:
    timeout: 10
  passive:
    timeout: 120
```

And on frontend, the restart button appears so I press it to restart z2m to take in consideration new options.
I was watching z2m log (via syslog): the shutdown of z2m was logged but it never log anymore once started...

Notes:
- while no log anymore via syslog I could see log via frontend.
- I have re-added "file" target in log_output as temporary work around
- I've restarted several times the service from system tty and still no log via syslog while file target work as expected

### Zigbee2MQTT version

1.26.0-dev (commit #ee227d85)

### Adapter firmware version

20220219

### Adapter

CC1352P-2

zen2, 2022-08-08:
Still no syslog log with current stable version 1.27.0.

astapelfeld, 2022-10-08:
Not working for me either 1.28

zen2, 2022-11-08:
Still an issue witn z2m 1.28.2.

Btw, logging documentation https://www.zigbee2mqtt.io/guide/configuration/logging.html need a fix:
`type: 5424 ` should be `type: '5424' ` or z2m refuses to start telling that this value need to be a string.

magcode, 2022-11-14:
I'm also having issues getting syslog to work with the latest code.
Nothing arrives at the remote syslog server via tcp4

magcode, 2022-11-14:
whats interesting: if z2m fails to start I can actually see a syslog entry!

zen2, 2022-12-16:
it is still an issue.

zen2, 2023-01-17:
Issue still exists.

magcode, 2023-01-17:
I've changed to udp/rsyslog (which again forwards promtail/loki via TCP) and now its working for me...

```
  log_syslog:
    app_name: Zigbee2MQTT
    eol: ' '
    host: localhost
    localhost: myhost
    path: /dev/log
    pid: process.pid
    port: 50514
    protocol: udp4
    type: '5424'
```

closed 2023-02-24 by github-actions[bot]