Koenkk/zigbee2mqtt #13075: Lellki WK34 doesn't show Energy (kWh)

rauljerico, 2022-07-08:
### What happened?

I've added 2 times a Lellki WK34 with Z2MQTT and the Energy value is always with Null value, the other parameters are OK

### Zigbee2MQTT version

1.26.0

### Adapter firmware version

20210708

### Adapter

Sonoff Dongle Plus

### Debug log

2022-07-08 13:17:36MQTT publish: topic 'zigbee2mqtt/Nevera', payload '{"current":0.23,"energy":null,"linkquality":81,"power":33,"power_outage_memory":"on","state":"ON","voltage":230}'

nmaster2042, 2022-07-09:
Same issue for me.

gelokatil, 2022-07-28:
Yes, same issue for me

Koenkk, 2022-07-28:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition =     {
    fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_0yxeawjt'}],
    model: 'WK34-EU',
    description: 'Power socket EU (with power monitoring)',
    vendor: 'LELLKI',
    fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report, fz.tuya_switch_power_outage_memory],
    toZigbee: [tz.on_off, tz.tuya_switch_power_outage_memory],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await endpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
        endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
        device.save();
    },
    options: [exposes.options.measurement_poll_interval()],
    exposes: [e.switch(), e.power(), e.current(), e.voltage().withAccess(ea.STATE),
        e.energy(), exposes.enum('power_outage_memory', ea.ALL, ['on', 'off', 'restore'])
            .withDescription('Recover state after power outage')],
    onEvent: tuya.onEventMeasurementPoll,
};

module.exports = definition;
```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

gelokatil, 2022-07-28:
I just placed the file and it seems resolved, these are the states now. The energy field already shows numerical value.

    "current": 0.38,
    "energy": 120.88,
    "linkquality": 60,
    "power": 11,
    "power_outage_memory": "on",
    "state": "ON",
    "voltage": 229,

Koenkk, 2022-07-28:
@rauljerico can you also check?

rauljerico, 2022-07-28:
> @rauljerico can you also check?

{
    "current": 0.03,
    "energy": 23.05,
    "linkquality": 78,
    "power": 0,
    "power_outage_memory": "on",
    "state": "ON",
    "voltage": 227
}

Now it works. Thank you

Koenkk, 2022-07-28:
Great, pushed the fix.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-07-28 by Koenkk