Koenkk/zigbee2mqtt #13077: zigbee_publish_error HOME CONTROL AS -  HC-SLM-1

nagydavid, 2022-07-08:
### What happened?

I have added the device to my network, but I am keep getting timeout publish error. @[TheStigh](https://github.com/TheStigh), [jdoe0000000](https://github.com/Koenkk/zigbee-herdsman-converters/issues?q=is%3Apr+author%3Ajdoe0000000),@Koenkk

### What did you expect to happen?

I expect to lock and unlock the device.

### How to reproduce it (minimal and precise)

Adding device, try to lock and unlock through web UI.

### Zigbee2MQTT version

 1.26.0

### Adapter firmware version

 20210708

### Adapter

Slaesh's stick

### Debug log

```
error 2022-07-08 18:05:06: Publish 'set' 'state' to 'Door Lock Hallway' failed: 'Error: Command 0x680ae2fffe7b5e49/1 closuresDoorLock.lockDoor({"pincodevalue":""}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 12982 - 1 - 165 - 257 - 0 after 10000ms)'
debug 2022-07-08 18:05:06: Error: Command 0x680ae2fffe7b5e49/1 closuresDoorLock.lockDoor({"pincodevalue":""}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 12982 - 1 - 165 - 257 - 0 after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:559:17)
    at processTimers (node:internal/timers:502:7)
info  2022-07-08 18:05:06: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Door Lock Hallway' failed: 'Error: Command 0x680ae2fffe7b5e49/1 closuresDoorLock.lockDoor({\"pincodevalue\":\"\"}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 12982 - 1 - 165 - 257 - 0 after 10000ms)'","meta":{"friendly_name":"Door Lock Hallway"},"type":"zigbee_publish_error"}'

```

nagydavid, 2022-07-17:
@Koenkk how can I pass pin code to the lock converter?

Koenkk, 2022-07-17:
@nagydavid that is not possible but I don't think that is the problem. Try pairing the device very close to the coordinator and try controlling it then.

nagydavid, 2022-07-18:
I did try, the repairing, Coordinator is connected to the device (2 m apart from each other).

I can see the state, it updates if the door open but the lock function throws an error, constantly .
[image]

Koenkk, 2022-07-19:
So if I understand correctly, the lock physically unlocks but z2m throws an error?

nagydavid, 2022-07-19:
No, the lock does not react for the command. But if I physically unlock it the state updates as unlocked as it should

Koenkk, 2022-07-19:
@thestigh does this device still work correctly for you with the latest z2m?

TheStigh, 2022-07-19:
> @TheStigh does this device still work correctly for you with the latest z2m?

Don't know, sold it as it was a piece of ****.  ID LOCK 150 are the best. Are there any updates to the converter lately, some that has been added by a script ?

nagydavid, 2022-07-20:
Can you please confirm that state in UI updates depending of manual lock AND unlock?
_**Yes, lock state updates instantaneously while is manually opened, regardless it is code or fingerprint is used.**_
Are you sure this is from HOME CONTROL? 
_**Yes it is, the HEIMGUARD/WATTLE/(HOME CONTROL?, they merged I guess)**_
It exist several versions if exactly same physical model but firmwares are different.
_**Since, I cannot see the exact firmware. I cannot confirm. Tomorrow I spin up a dev container to test and see the available attributes.**_

Can you - and don't skip any of the a-b-c-d below:;
_**There is a reset button top right from the zigbee module. Is that it? (Make sense, any extra procedure I should be aware of apart from pushing the button?)**_
a: Reset the lock itself to factory default
b: Delete the entity from Z2M and use FORCE REMOVE option
c: Restart Z2M
d: Re-pair

Thanks for the answers.

nagydavid, 2022-07-22:
So I did the following:

1. reseted the device,
2.  force removed
3. restart z2m
4. re-paired

no success.

I have updated the latest firmware (https://heimgard.com/support/firmware-filer-til-vare-produkter). 
I repeated the above. 

Here is the log of repairing:

```
Zigbee2MQTT:debug 2022-07-22 19:34:30: Received MQTT message on 'testmqtt/bridge/groups' with data '[]'
Zigbee2MQTT:debug 2022-07-22 19:34:30: Received MQTT message on 'testmqtt/bridge/extensions' with data '[]'
Zigbee2MQTT:info  2022-07-22 19:34:30: Started frontend on port 0.0.0.0:8080
Zigbee2MQTT:info  2022-07-22 19:35:37: Device '0x680ae2fffe7b5e49' joined
Zigbee2MQTT:info  2022-07-22 19:35:37: MQTT publish: topic 'testmqtt/bridge/event', payload '{"data":{"friendly_name":"0x680ae2fffe7b5e49","ieee_address":"0x680ae2fffe7b5e49"},"type":"device_joined"}'
Zigbee2MQTT:info  2022-07-22 19:35:37: Starting interview of '0x680ae2fffe7b5e49'
Zigbee2MQTT:info  2022-07-22 19:35:37: MQTT publish: topic 'testmqtt/bridge/event', payload '{"data":{"friendly_name":"0x680ae2fffe7b5e49","ieee_address":"0x680ae2fffe7b5e49","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:debug 2022-07-22 19:35:37: Device '0x680ae2fffe7b5e49' announced itself
Zigbee2MQTT:info  2022-07-22 19:35:37: MQTT publish: topic 'testmqtt/bridge/event', payload '{"data":{"friendly_name":"0x680ae2fffe7b5e49","ieee_address":"0x680ae2fffe7b5e49"},"type":"device_announce"}'
Zigbee2MQTT:debug 2022-07-22 19:35:38: Received Zigbee message from '0x680ae2fffe7b5e49', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"Home Control AS","modelId":"HC-SLM-1"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-07-22 19:35:38: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-07-22 19:35:38: Received Zigbee message from '0x680ae2fffe7b5e49', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"Home Control AS","modelId":"HC-SLM-1"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-07-22 19:35:38: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-07-22 19:35:51: Received Zigbee message from '0x680ae2fffe7b5e49', type 'readResponse', cluster 'genBasic', data '{"powerSource":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-07-22 19:35:51: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-07-22 19:35:54: Received Zigbee message from '0x680ae2fffe7b5e49', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-07-22 19:35:54: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-07-22 19:35:57: Received Zigbee message from '0x680ae2fffe7b5e49', type 'readResponse', cluster 'genBasic', data '{"appVersion":100}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-07-22 19:35:57: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-07-22 19:36:00: Received Zigbee message from '0x680ae2fffe7b5e49', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-07-22 19:36:00: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-07-22 19:36:02: Received Zigbee message from '0x680ae2fffe7b5e49', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-07-22 19:36:02: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-07-22 19:36:06: Received Zigbee message from '0x680ae2fffe7b5e49', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-07-22 19:36:06: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-07-22 19:36:10: Received Zigbee message from '0x680ae2fffe7b5e49', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-07-22 19:36:10: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2022-07-22 19:36:13: Received Zigbee message from '0x680ae2fffe7b5e49', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-07-22 19:36:13: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:info  2022-07-22 19:36:13: Successfully interviewed '0x680ae2fffe7b5e49', device has successfully been paired
Zigbee2MQTT:info  2022-07-22 19:36:13: Device '0x680ae2fffe7b5e49' is supported, identified as: Home Control AS Heimgard (Wattle) door lock pro (HC-SLM-1)
Zigbee2MQTT:info  2022-07-22 19:36:13: MQTT publish: topic 'testmqtt/bridge/event', payload '{"data":{"definition":{"description":"Heimgard (Wattle) door lock pro","exposes":[{"features":[{"access":7,"description":"State of the lock","name":"state","property":"state","type":"binary","value_off":"UNLOCK","value_on":"LOCK"},{"access":1,"description":"Actual state of the lock","name":"lock_state","property":"lock_state","type":"enum","values":["not_fully_locked","locked","unlocked"]}],"type":"lock"},{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"HC-SLM-1","options":[],"supports_ota":false,"vendor": [... cut ...] ":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240))
    at ZStackAdapter.sendZclFrameToEndpointInternal (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:414:23)
    at Queue.executeNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
Zigbee2MQTT:debug 2022-07-22 19:39:18: Saving state to file /opt/zigbee2mqtt/data/state.json
```

So for now no luck

nagydavid, 2022-07-22:
I added the PR for the ZigbeeOTA repo regarding the firmware published by the vendor

closed 2022-07-22 by nagydavid

TheStigh, 2022-07-22:
@nagydavid Why did you close this? Did you get it to work?

nagydavid, 2022-07-22:
Tusind tak, i lige måde :)