Koenkk/zigbee2mqtt #13078: Container can't start if environment variable ZIGBEE2MQTT_CONFIG_FRONTEND_PORT is set in docker-compose

Cyneath, 2022-07-08:
### What happened?

The container fails to start if the environment variable ZIGBEE2MQTT_CONFIG_FRONTEND_PORT is set in docker-compose:

Using '/app/data' as data directory
TypeError: Cannot create property 'port' on boolean 'true'
    at /app/lib/util/settings.ts:391:41
    at Array.forEach (<anonymous>)
    at iterate (/app/lib/util/settings.ts:369:26)
    at /app/lib/util/settings.ts:408:21
    at Array.forEach (<anonymous>)
    at iterate (/app/lib/util/settings.ts:369:26)
    at /app/lib/util/settings.ts:408:21
    at Array.forEach (<anonymous>)
    at iterate (/app/lib/util/settings.ts:369:26)
    at /app/lib/util/settings.ts:408:21
Container stopped

### What did you expect to happen?

The docker container can start successfully

### How to reproduce it (minimal and precise)

Configure the frontend port as env variable in docker-compose:

`environment:`
`- ZIGBEE2MQTT_CONFIG_FRONTEND_PORT=8080` 

### Zigbee2MQTT version

1.26.0-dev commit: 8af3ad5

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

Koenkk, 2022-07-09:
How does your `configuration.yaml` look like? Does it contain `frontend: true`? If yes this should be replaced with `frontend: {}`.

Cyneath, 2022-07-09:
> How does your `configuration.yaml` look like? Does it contain `frontend: true`? If yes this should be replaced with `frontend: {}`.

Yeah, it had  `frontend: true`. After I replaced it with `frontend: {}`, it works again.

My docker-compose contained the following:
  `  - ZIGBEE2MQTT_CONFIG_FRONTEND=true`
  `  - ZIGBEE2MQTT_CONFIG_FRONTEND_PORT=8080`

That did work until  ~1 or 2 weeks ago. Maybe something changed in versions which are newer than that.

closed 2022-07-09 by Cyneath

NathanSweet, 2022-12-26:
I got bit by this one too. Documentation still says `frontend: true` is OK:
https://www.zigbee2mqtt.io/guide/configuration/frontend.html
I'm using the ENV variable, I'd think `true` should be OK there.

NathanSweet, 2022-12-26:
To be a little more clear, I set:
```
ZIGBEE2MQTT_CONFIG_FRONTEND=true
ZIGBEE2MQTT_CONFIG_FRONTEND_PORT=4321
```
Setting the port environment variable this way crashes (`Cannot create property 'port' on boolean 'true'`).

There appears to be a real bug here, maybe we can reopen this issue?

dmfrey, 2023-10-31:
I see this is closed, but was it ever resolved? I still see it happening with 1.33.1

nlg35, 2023-11-04:
Hi, I have exactly the same error with the environment variable `ZIGBEE2MQTT_CONFIG_FRONTEND_PORT` and version 1.33.2.
As my host already use port 8080 and I want to have a host-network container Z2M, i need to change fontend port.

I did these two experiences :

First I set the frontend port in the environment variables of my compose file:
And, I got the following error: 
```
TypeError: Cannot create property 'port' on boolean 'true'
    at /app/lib/util/settings.ts:411:41
```

... and the `configuration.yaml` was the default file, all my other environment variables were ignored.
For example, the serial was `port: /dev/ttyACM0`

Secondly, I tried to install Z2M without the frontend port (but with the other environment variables).
I did not have any error and the `configuration.yaml` mentionned well the informations given by the environment.
The serial port was `/dev/ttyUSB0`

Here is my compose file:

```
version: '3'
services:

  homeassistant:
    container_name: home-assistant
    image: homeassistant/home-assistant:latest
    volumes:
      - /share/Container/mes-app-data/home-assistant:/config
      - /etc/localtime:/etc/localtime:ro
    restart: unless-stopped
    network_mode: host
    depends_on:
      - zigbee2mqtt
      - eclipse-mosquitto

  eclipse-mosquitto:
    container_name: mosquitto
    image: eclipse-mosquitto:latest
    restart: unless-stopped
    network_mode: host
    volumes:
      - /share/Container/mes-app-data/mosquitto/data:/mosquitto/data
      - /share/Container/mes-app-data/mosquitto/log:/mosquitto/log
      - /share/Container/mes-app-data/mosquitto/config:/mosquitto/config

  zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt:latest
    restart: unless-stopped
    network_mode: host
    environment:
      TZ: Europe/Paris
      ZIGBEE2MQTT_CONFIG_ADVANCED_CHANNEL: 25     
      ZIGBEE2MQTT_CONFIG_MQTT_BASE_TOPIC: zigbee2mqtt
      ZIGBEE2MQTT_CONFIG_HOMEASSISTANT: true
      ZIGBEE2MQTT_CONFIG_PERMIT_JOIN: true
      ZIGBEE2MQTT_CONFIG_SERIAL_PORT: /dev/ttyUSB0
      ZIGBEE2MQTT_CONFIG_FRONTEND_PORT: 8888  # Problem
      ZIGBEE2MQTT_CONFIG_ADVANCED_NETWORK_KEY: GENERATE
    volumes:
      - /run/udev:/run/udev:ro
      - /share/Container/mes-app-data/zigbee2mqtt:/app/data
    devices:
      - /dev/ttyUSB0:/dev/ttyUSB0
    privileged: true
    depends_on:
      - eclipse-mosquitto
```
Thanks for any help.

eloo, 2023-11-13:
Hi,

yes it seems that this bug is still not solved.

I'm not sure but it looks like its related to the very first run. 
It seems that if no configuration.yaml could be found a default one would be created, nevermind if all would be configured using env variables. 

So @nlg35 you can simply create a configuration.yaml once manually in your "data" directory and then use env variables. 
This should work.

But yes it looks like env variable configuration seems to be a bit broken. 

I have also not figured out how the env config is applied. At least its not written back to the configuration.yaml.
I guess its just loaded in the internal state, but together with the other "write back config" from the GUI into the configuration.yaml file it looks a bit strange and its hard to determine which setting has presedence.

LeVraiRoiDHyrule, 2023-12-24:
Hi, I'm struggling with this too. It was working a few months ago.
I've only set the env variable `ZIGBEE2MQTT_CONFIG_FRONTEND_URL="https://zigbee.mydomain.com"` to enable the frontend but it creates this error. Is the config file the only way then ?