Koenkk/zigbee2mqtt #13080: Problem after update to home assistant 2022.7

Koterak, 2022-07-09:
### What happened?

When I updated the home assistant to 2022.7 I see error in logs:

### Zigbee2MQTT version

1.26.0

### Adapter firmware version

20220507

### Adapter

zStack3x0

### Debug log

Logger: homeassistant.util.logging
Source: util/logging.py:116
First occurred: 09:46:41 (14 occurrences)
Last logged: 09:46:41

Exception in async_discover when dispatching 'mqtt_discovery_new_number_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}, {'topic': 'zigbee2mqtt/Salon - Czujniki/availability'}], 'availability_mode': 'all', 'command_topic': 'zigbee2mqtt/Salon - Czujniki/set/alarm_humidity_min', 'device': {'identifiers': ['zigbee2mqtt_0xcc86ecfffed10fd5'], 'manufacturer': 'TuYa', 'model': 'Temperature & humidity & illuminance sensor with display (LCZ030)', 'name': 'Salon - Czujniki'}, 'device_class': 'humidity', 'icon': 'mdi:water-minus', 'json_attributes_topic': 'zigbee2mqtt/Salon - Czujniki', 'max': 100, 'min': 0, 'name': 'Salon - Czujniki alarm humidity min', 'state_topic': 'zigbee2mqtt/Salon - Czujniki', 'unique_id': '0xcc86ecfffed10fd5_alarm_humidity_min_zigbee2mqtt', 'unit_of_measurement': '%', 'value_template': '{{ value_json.alarm_humidity_min }}', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 309, in async_discover config = schema(discovery_payload) File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 232, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 355, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 351, in _exec v = func(v) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 818, in validate_callable return schema(data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 595, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 433, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: expected NumberDeviceClass or one of 'temperature' for dictionary value @ data['device_class']
Exception in async_discover when dispatching 'mqtt_discovery_new_number_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}, {'topic': 'zigbee2mqtt/Rodzice - Czujniki/availability'}], 'availability_mode': 'all', 'command_topic': 'zigbee2mqtt/Rodzice - Czujniki/set/alarm_humidity_max', 'device': {'identifiers': ['zigbee2mqtt_0xb4e3f9fffe07b467'], 'manufacturer': 'TuYa', 'model': 'Temperature & humidity & illuminance sensor with display (LCZ030)', 'name': 'Rodzice - Czujniki'}, 'device_class': 'humidity', 'icon': 'mdi:water-plus', 'json_attributes_topic': 'zigbee2mqtt/Rodzice - Czujniki', 'max': 100, 'min': 0, 'name': 'Rodzice - Czujniki alarm humidity max', 'state_topic': 'zigbee2mqtt/Rodzice - Czujniki', 'unique_id': '0xb4e3f9fffe07b467_alarm_humidity_max_zigbee2mqtt', 'unit_of_measurement': '%', 'value_template': '{{ value_json.alarm_humidity_max }}', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 309, in async_discover config = schema(discovery_payload) File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 232, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 355, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 351, in _exec v = func(v) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 818, in validate_callable return schema(data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 595, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 433, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: expected NumberDeviceClass or one of 'temperature' for dictionary value @ data['device_class']
Exception in async_discover when dispatching 'mqtt_discovery_new_number_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}, {'topic': 'zigbee2mqtt/Rodzice - Czujniki/availability'}], 'availability_mode': 'all', 'command_topic': 'zigbee2mqtt/Rodzice - Czujniki/set/alarm_humidity_min', 'device': {'identifiers': ['zigbee2mqtt_0xb4e3f9fffe07b467'], 'manufacturer': 'TuYa', 'model': 'Temperature & humidity & illuminance sensor with display (LCZ030)', 'name': 'Rodzice - Czujniki'}, 'device_class': 'humidity', 'icon': 'mdi:water-minus', 'json_attributes_topic': 'zigbee2mqtt/Rodzice - Czujniki', 'max': 100, 'min': 0, 'name': 'Rodzice - Czujniki alarm humidity min', 'state_topic': 'zigbee2mqtt/Rodzice - Czujniki', 'unique_id': '0xb4e3f9fffe07b467_alarm_humidity_min_zigbee2mqtt', 'unit_of_measurement': '%', 'value_template': '{{ value_json.alarm_humidity_min }}', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 309, in async_discover config = schema(discovery_payload) File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 232, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 355, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 351, in _exec v = func(v) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 818, in validate_callable return schema(data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 595, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 433, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: expected NumberDeviceClass or one of 'temperature' for dictionary value @ data['device_class']
Exception in async_discover when dispatching 'mqtt_discovery_new_number_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}, {'topic': 'zigbee2mqtt/Łazienka - Czujniki/availability'}], 'availability_mode': 'all', 'command_topic': 'zigbee2mqtt/Łazienka - Czujniki/set/alarm_humidity_max', 'device': {'identifiers': ['zigbee2mqtt_0xcc86ecfffed110f2'], 'manufacturer': 'TuYa', 'model': 'Temperature & humidity & illuminance sensor with display (LCZ030)', 'name': 'Łazienka - Czujniki'}, 'device_class': 'humidity', 'icon': 'mdi:water-plus', 'json_attributes_topic': 'zigbee2mqtt/Łazienka - Czujniki', 'max': 100, 'min': 0, 'name': 'Łazienka - Czujniki alarm humidity max', 'state_topic': 'zigbee2mqtt/Łazienka - Czujniki', 'unique_id': '0xcc86ecfffed110f2_alarm_humidity_max_zigbee2mqtt', 'unit_of_measurement': '%', 'value_template': '{{ value_json.alarm_humidity_max }}', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 309, in async_discover config = schema(discovery_payload) File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 232, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.10/site-packages/voluptuou [... cut ...] uous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 595, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 433, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: expected NumberDeviceClass or one of 'temperature' for dictionary value @ data['device_class']

ejohb, 2022-07-10:
Seeing same issue:

Announce message:

```
2022-07-10 12:07:57 INFO  amniotic.tools   : Queue: publish:homeassistant/number/development_amniotic_master_volume/config>
```
```json
{
   "name":"Development Amniotic Master Volume",
   "unique_id":"development_amniotic_master_volume",
   "object_id":"development_amniotic_master_volume",
   "device":{
      "connections":[
         [
            "mac",
            "development-amniotic-00155d216ca5"
         ]
      ],
      "sw_version":"0.3.9",
      "name":"Development Amniotic",
      "model":"Amniotic",
      "manufacturer":"Frontmatter",
      "identifiers":"development-amniotic-00155d216ca5"
   },
   "device_class":"number",
   "force_update":true,
   "payload_available":"Online",
   "payload_not_available":"Offline",
   "availability_topic":"tele/development/amniotic/LWT",
   "state_topic":"stat/development/amniotic/master/volume/state",
   "command_topic":"stat/development/amniotic/master/volume/command",
   "icon":"mdi:volume-high",
   "min":0,
   "max":100
}
```

Error in log:

> ```Exception in async_discover when dispatching 'mqtt_discovery_new_number_mqtt': ({'name': 'Bedroom Amniotic Master Volume', 'unique_id': 'bedroom_amniotic_master_volume', 'object_id': 'bedroom_amniotic_master_volume', 'device': {'connections': [['mac', 'bedroom-amniotic-dca632f63093']], 'sw_version': '0.3.9', 'name': 'Bedroom Amniotic', 'model': 'Amniotic', 'manufacturer': 'Frontmatter', 'identifiers': 'bedroom-amniotic-dca632f63093'}, 'device_class': 'number', 'force_update': True, 'payload_available': 'Online', 'payload_not_available': 'Offline', 'availability_topic': 'tele/bedroom/amniotic/LWT', 'state_topic': 'stat/bedroom/amniotic/master/volume/state', 'command_topic': 'stat/bedroom/amniotic/master/volume/command', 'icon': 'mdi:volume-high', 'min': 0, 'max': 100, 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 309, in async_discover config = schema(discovery_payload) File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 232, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 355, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 351, in _exec v = func(v) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 818, in validate_callable return schema(data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 595, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 433, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: expected NumberDeviceClass or one of 'temperature' for dictionary value @ data['device_class']```

El-mou33, 2022-07-11:
I had the same issue after upgrading to 2022.7.3

ejohb, 2022-07-11:
> I had the same issue after upgrading to 2022.7.3

FYI I'm having a similar problem with my own project and have raised it here: https://github.com/home-assistant/core/issues/74879

Koenkk, 2022-07-11:
Pushed a fix, could you check if the error is gone?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Koterak, 2022-07-12:
Installed edge version. For now there are no errors about this. But I see something suspicious about thermostats. Maybe it was because of some change in home assistant and now there won't be more errors. But maybe log will help:

> 2022-07-12 15:26:53 ERROR (MainThread) [homeassistant.util.logging] Exception in async_discover when dispatching 'mqtt_discovery_new_number_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}, {'topic': 'zigbee2mqtt/Kuba - Termostat/availability'}], 'availability_mode': 'all', 'command_topic': 'zigbee2mqtt/Kuba - Termostat/set/min_temperature', 'device': {'identifiers': ['zigbee2mqtt_0x040d84fffe5ccc56'], 'manufacturer': 'Moes', 'model': 'Thermostatic radiator valve (BRT-100-TRV)', 'name': 'Kuba - Termostat'}, 'device_class': None, 'entity_category': 'config', 'icon': 'mdi:thermometer', 'json_attributes_topic': 'zigbee2mqtt/Kuba - Termostat', 'max': 15, 'min': 1, 'name': 'Kuba - Termostat min temperature', 'state_topic': 'zigbee2mqtt/Kuba - Termostat', 'unique_id': '0x040d84fffe5ccc56_min_temperature_zigbee2mqtt', 'unit_of_measurement': '°C', 'value_template': '{{ value_json.min_temperature }}', 'platform': 'mqtt'},)
> Traceback (most recent call last):
> File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 309, in async_discover
> config = schema(discovery_payload)
> File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 232, in __call__
> return self._exec((Schema(val) for val in self.validators), v)
> File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 355, in _exec
> raise e if self.msg is None else AllInvalid(self.msg, path=path)
> File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 351, in _exec
> v = func(v)
> File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__
> return self._compiled([], data)
> File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 818, in validate_callable
> return schema(data)
> File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__
> return self._compiled([], data)
> File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 595, in validate_dict
> return base_validate(path, iteritems(data), out)
> File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 433, in validate_mapping
> raise er.MultipleInvalid(errors)
> voluptuous.error.MultipleInvalid: expected NumberDeviceClass or one of 'temperature' for dictionary value @ data['device_class']
> 2022-07-12 15:26:53 ERROR (MainThread) [homeassistant.util.logging] Exception in async_discover when dispatching 'mqtt_discovery_new_number_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}, {'topic': 'zigbee2mqtt/Kuba - Termostat/availability'}], 'availability_mode': 'all', 'command_topic': 'zigbee2mqtt/Kuba - Termostat/set/boost_heating_countdown_time_set', 'device': {'identifiers': ['zigbee2mqtt_0x040d84fffe5ccc56'], 'manufacturer': 'Moes', 'model': 'Thermostatic radiator valve (BRT-100-TRV)', 'name': 'Kuba - Termostat'}, 'device_class': None, 'json_attributes_topic': 'zigbee2mqtt/Kuba - Termostat', 'max': 900, 'min': 100, 'name': 'Kuba - Termostat boost heating countdown time set', 'state_topic': 'zigbee2mqtt/Kuba - Termostat', 'unique_id': '0x040d84fffe5ccc56_boost_heating_countdown_time_set_zigbee2mqtt', 'unit_of_measurement': 'second', 'value_template': '{{ value_json.boost_heating_countdown_time_set }}', 'platform': 'mqtt'},)
> Traceback (most recent call last):
> File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 309, in async_discover
> config = schema(discovery_payload)
> File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 232, in __call__
> return self._exec((Schema(val) for val in self.validators), v)
> File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 355, in _exec
> raise e if self.msg is None else AllInvalid(self.msg, path=path)
> File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 351, in _exec
> v = func(v)
> File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__
> return self._compiled([], data)
> File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 818, in validate_callable
> return schema(data)
> File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__
> return self._compiled([], data)
> File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 595, in validate_dict
> return base_validate(path, iteritems(data), out)
> File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 433, in validate_mapping
> raise er.MultipleInvalid(errors)
> voluptuous.error.MultipleInvalid: expected NumberDeviceClass or one of 'temperature' for dictionary value @ data['device_class']
> 2022-07-12 15:26:53 ERROR (MainThread) [homeassistant.util.logging] Exception in async_discover when dispatching 'mqtt_discovery_new_number_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}, {'topic': 'zigbee2mqtt/Salon - Termostat/availability'}], 'availability_mode': 'all', 'command_topic': 'zigbee2mqtt/Salon - Termostat/set/eco_temperature', 'device': {'identifiers': ['zigbee2mqtt_0x040d84fffe5ccd78'], 'manufacturer': 'Moes', 'model': 'Thermostatic radiator  [... cut ...]
> File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__
> return self._compiled([], data)
> File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 595, in validate_dict
> return base_validate(path, iteritems(data), out)
> File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 433, in validate_mapping
> raise er.MultipleInvalid(errors)
> voluptuous.error.MultipleInvalid: expected NumberDeviceClass or one of 'temperature' for dictionary value @ data['device_class']
> 
>

Koenkk, 2022-07-12:
@Koterak I'm confused, you say there are no more errors anymore after update to edge but the log you provided is full of the `expected NumberDeviceClass or one of 'temperature' for dictionary value @ data['device_class']` errors?

Koterak, 2022-07-12:
As I can see it happened only once after restart now. Now the error is
about another device. First one I sent was about
https://www.zigbee2mqtt.io/devices/LCZ030.html#tuya-lcz030 now there's no
error from this device type. Now It showed from
https://www.zigbee2mqtt.io/devices/BRT-100-TRV.html#moes-brt-100-trv .
Maybe it's something device related?

I confirmed: it only happens after restart.

nukusinji, 2022-07-13:
Hi.
It is still.

model : SiHAS multipurpose sensor (CSM-300ZB), entity : people
model': 'Temperature & humidity & illuminance sensor with display (LCZ030), entity : alarm_humidity_max, alarm_humidity_min

`Exception in async_discover when dispatching 'mqtt_discovery_new_number_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}], 'command_topic': 'zigbee2mqtt/bathroom_counter/set/people', 'device': {'configuration_url': '//#/device/0x00158d0003613007/info', 'identifiers': ['zigbee2mqtt_0x00158d0003613007'], 'manufacturer': 'ShinaSystem', 'model': 'SiHAS multipurpose sensor (CSM-300ZB)', 'name': 'bathroom_counter', 'sw_version': '4000-0001'}, 'device_class': None, 'icon': 'mdi:account-multiple', 'json_attributes_topic': 'zigbee2mqtt/bathroom_counter', 'max': 50, 'min': 0, 'name': 'bathroom_counter_people', 'state_class': 'measurement', 'state_topic': 'zigbee2mqtt/bathroom_counter', 'unique_id': '0x00158d0003613007_people_zigbee2mqtt', 'value_template': '{{ value_json.people }}', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 309, in async_discover config = schema(discovery_payload) File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 232, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 355, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 351, in _exec v = func(v) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 818, in validate_callable return schema(data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 595, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 433, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: expected NumberDeviceClass or one of 'temperature' for dictionary value @ data['device_class']

Exception in async_discover when dispatching 'mqtt_discovery_new_number_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}], 'command_topic': 'zigbee2mqtt/mainroom_multipurpose_sensor/set/alarm_humidity_max', 'device': {'configuration_url': '//#/device/0xb4e3f9fffe0c464d/info', 'identifiers': ['zigbee2mqtt_0xb4e3f9fffe0c464d'], 'manufacturer': 'TuYa', 'model': 'Temperature & humidity & illuminance sensor with display (LCZ030)', 'name': 'mainroom_multipurpose_sensor'}, 'device_class': 'humidity', 'icon': 'mdi:water-plus', 'json_attributes_topic': 'zigbee2mqtt/mainroom_multipurpose_sensor', 'max': 100, 'min': 0, 'name': 'mainroom_multipurpose_sensor_alarm_humidity_max', 'state_topic': 'zigbee2mqtt/mainroom_multipurpose_sensor', 'unique_id': '0xb4e3f9fffe0c464d_alarm_humidity_max_zigbee2mqtt', 'unit_of_measurement': '%', 'value_template': '{{ value_json.alarm_humidity_max }}', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 309, in async_discover config = schema(discovery_payload) File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 232, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 355, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 351, in _exec v = func(v) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 818, in validate_callable return schema(data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 595, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 433, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: expected NumberDeviceClass or one of 'temperature' for dictionary value @ data['device_class']

Exception in async_discover when dispatching 'mqtt_discovery_new_number_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}], 'command_topic': 'zigbee2mqtt/mainroom_multipurpose_sensor/set/alarm_humidity_min', 'device': {'configuration_url': '//#/device/0xb4e3f9fffe0c464d/info', 'identifiers': ['zigbee2mqtt_0xb4e3f9fffe0c464d'], 'manufacturer': 'TuYa', 'model': 'Temperature & humidity & illuminance sensor with display (LCZ030)', 'name': 'mainroom_multipurpose_sensor'}, 'device_class': 'humidity', 'icon': 'mdi:water-minus', 'json_attributes_topic': 'zigbee2mqtt/mainroom_multipurpose_sensor', 'max': 100, 'min': 0, 'name': 'mainroom_multipurpose_sensor_alarm_humidity_min', 'state_topic': 'zigbee2mqtt/mainroom_multipurpose_sensor' [... cut ...] ous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 595, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 433, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: expected NumberDeviceClass or one of 'temperature' for dictionary value @ data['device_class']`

Koterak, 2022-07-13:
Can confirm too. Still have same error which I posted before

Koterak, 2022-07-13:
Can confirm that the error still exists

Koenkk, 2022-07-13:
I think I fixed it now

@emontnemery I believe `device_class` should be set to `temperature` OR not be present in the discovery payload at all (`None` gives me errors)

Koterak, 2022-07-13:
As I can see from newest update no more errors

closed 2022-07-13 by Koterak