Koenkk/zigbee2mqtt #13086: Home Assistant 2022.7.2 The 'value_template' option has been removed

stickpin, 2022-07-09:
### What happened?

Hi,

After upgrading to Home Assistant 2022.7.1+, when I restart HA, all my bulbs have shown as ON.
When I check the HA logs I see the following error:
```
2022-07-09 16:07:28 ERROR (MainThread) [homeassistant.util.logging] Exception in async_discover when dispatching 'mqtt_discovery_new_light_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state', 'value_template': '{{ value_json.state }}'}, {'topic': 'zigbee2mqtt/Kitchen Left Spots/availability', 'value_template': '{{ value_json.state }}'}], 'availability_mode': 'all', 'command_topic': 'zigbee2mqtt/Kitchen Left Spots/l2/set', 'device': {'identifiers': ['zigbee2mqtt_0x00158d000444d89c'], 'manufacturer': 'Xiaomi', 'model': 'Aqara wireless relay controller (LLKZMK11LM)', 'name': 'Kitchen Left Spots'}, 'name': 'Kitchen Left Spots l2', 'payload_off': 'OFF', 'payload_on': 'ON', 'state_topic': 'zigbee2mqtt/Kitchen Left Spots', 'unique_id': '0x00158d000444d89c_light_l2_zigbee2mqtt', 'value_template': '{{ value_json.state_l2 }}', 'platform': 'mqtt'},)
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 818, in validate_callable
    return schema(data)
  File "/usr/src/homeassistant/homeassistant/helpers/config_validation.py", line 829, in validator
    raise vol.Invalid(warning % arguments)
voluptuous.error.Invalid: The 'value_template' option has been removed, please remove it from your configuration

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 309, in async_discover
    config = schema(discovery_payload)
  File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 232, in __call__
    return self._exec((Schema(val) for val in self.validators), v)
  File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 355, in _exec
    raise e if self.msg is None else AllInvalid(self.msg, path=path)
  File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 351, in _exec
    v = func(v)
  File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 818, in validate_callable
    return schema(data)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/light/__init__.py", line 49, in validate_mqtt_light_discovery
    return schemas[value[CONF_SCHEMA]](value)
  File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 232, in __call__
    return self._exec((Schema(val) for val in self.validators), v)
  File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 355, in _exec
    raise e if self.msg is None else AllInvalid(self.msg, path=path)
  File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 351, in _exec
    v = func(v)
  File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 276, in __call__
    raise er.MultipleInvalid([e])
voluptuous.error.MultipleInvalid: The 'value_template' option has been removed, please remove it from your configuration

```
Does anyone have a similar issue?

Thanks a lot in advance!

### Zigbee2MQTT version

1.26.0

### Adapter firmware version

20220219

### Adapter

Slaesh's CC2652RB

Koenkk, 2022-07-11:
Do you fully rely on the MQTT discovery or do you have any manual mqtt light entries in the Home Assistant `configuration.yaml` file?

stickpin, 2022-07-11:
> Do you fully rely on the MQTT discovery or do you have any manual mqtt light entries in the Home Assistant `configuration.yaml` file?

No, I don't have any manual lights configured.
Strangely, it seems to be affecting only a specific device: LLKZMK11LM
I have 86 devices in the network and only this throwing an error.

Koenkk, 2022-07-11:
Are you overriding the home assistant discovery payload for this device in the `configuration.yaml`?

stickpin, 2022-07-11:
@Koenkk 

Thanks for pointing it out!
I am dropping/modifying some exposes in Z2M devices.xml towards HA for this device:

```
"0x00158d000444ec9a":
  friendly_name: Kitchen Right Spots
  homeassistant:
    linkquality: null
    update_state: null
    update_available: null
    tamper: null
    voltage: null
    power_on_behavior: null
    interlock: null
    device_temperature: null
    energy: null
    power: null
    power_outage_count: null
    power_outage_memory: null
    switch_l1:
      type: light
      object_id: light_l1
    switch_l2:
      type: light
      object_id: light_l2
```

Seems like this one is no longer working:
```
    switch_l1:
      type: light
      object_id: light_l1
    switch_l2:
      type: light
      object_id: light_l2
```

Once I've removed it, the error is gone.