Koenkk/zigbee2mqtt #13087: [New device support]:

rafaelreis-r, 2022-07-09:
### Link

https://www.aqara.com/en/product/smart-door-lock-a100-zigbee

### Database entry

"definition":null,"friendly_name":"0x54ef4410004c5112","ieee_address":"0x54ef4410004c5112","status":"successful","supported":false},"type":"device_interview"

### Comments

Hey all. I just got my hands on a **Aqara A100 Pro** smart lock. Although I'm a competent sysadmin, I've never worked with zigbee devices. Wondering if I could provide info / be guided to dump information in order to add support to this family of door locks.

### Hardware I own:

- Conbee II stick on the latest firmware
- Pi4 running HASSIO / Zigbee2mqtt
- Aqara G3 Camera Hub (unfortunately[ non rooted](https://github.com/Wh1terat/aQRootG3#warningwarning-warning))

I'm willing to purchase a CC2531 sniffer if necessary.

### What have I done so far:

- read the [guide](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#instructions)
- Checked out `lumi.lock `entries on [Xiaomi.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/xiaomi.js)
- Figured out a Pairing process explained below.

### Pairing Process

The A100 series do not have a physical Zigbee pairing button / process. You *need* a compatible Aqara Hub to trigger the pairing on lock. I figured out a way to do it:

1. Add the lock to the Aqara app (Bluetooth)
2. Add your zigbee hub to the Aqara app (wifi)
3. Permit Join All on Zigbee2mqtt
4. Still in the Aqara app, go to the lock settings and follow the prompts to bind it to your Aqara Zigbee Hub (in my case the G3)
5. As soon as you hear the audio / led prompt from the Aqara hub, cut it's power. 
6. If you get the timing right, (took me 3 attempts) the lock will complete pairing process to Zigbee2Mqtt

### A100 Pro vs A100 Zigbee

I read that both have Zigbee functionality, Pro version lacking remote unlock and remote passcode setup as compared to the Zigbee version. I couldn't find the Zigbee version at the time of purchase. Details [here](https://homekitnews.com/2022/03/24/aqara-a100-zigbee-smart-lock-review/)

### Logs I collected over the pairing procedure

`info  2022-07-06 18:29:43: Device '0x54ef4410004c5112' joined
info  2022-07-06 18:29:44: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef4410004c5112","ieee_address":"0x54ef4410004c5112"},"type":"device_joined"}'
info  2022-07-06 18:29:44: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x54ef4410004c5112"},"type":"device_connected"}'
info  2022-07-06 18:29:44: Starting interview of '0x54ef4410004c5112'
info  2022-07-06 18:29:44: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef4410004c5112","ieee_address":"0x54ef4410004c5112","status":"started"},"type":"device_interview"}'
info  2022-07-06 18:29:44: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x54ef4410004c5112"},"type":"pairing"}'
error 2022-07-06 18:32:10: Publish 'set' 'read' to '0x54ef4410004c5112' failed: 'Error: Read 0x54ef4410004c5112/1 genIdentify(["identifyTime"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
info  2022-07-06 18:32:10: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'read' to '0x54ef4410004c5112' failed: 'Error: Read 0x54ef4410004c5112/1 genIdentify([\"identifyTime\"], {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received)'","meta":{"friendly_name":"0x54ef4410004c5112"},"type":"zigbee_publish_error"}'
error 2022-07-06 18:32:17: Publish 'set' 'read' to '0x54ef4410004c5112' failed: 'Error: Read 0x54ef4410004c5112/1 genIdentify(["identifyTime","identifyCommissionState"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
info  2022-07-06 18:32:17: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'read' to '0x54ef4410004c5112' failed: 'Error: Read 0x54ef4410004c5112/1 genIdentify([\"identifyTime\",\"identifyCommissionState\"], {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received)'","meta":{"friendly_name":"0x54ef4410004c5112"},"type":"zigbee_publish_error"}'
error 2022-07-06 18:32:23: Publish 'set' 'read' to '0x54ef4410004c5112' failed: 'Error: Read 0x54ef4410004c5112/1 genGroups(["nameSupport"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
info  2022-07-06 18:32:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'read' to '0x54ef4410004c5112' failed: 'Error: Read 0x54ef4410004c5112/1 genGroups([\"nameSupport\"], {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received)'","meta":{"friendly_name":"0x54ef4410004c5112"},"type":"zigbee_publish_error"}'
error 2022-07-06 18:32:49: Publish 'set' 'read' to '0x54ef4410004c5112' failed: 'Error: Read 0x54ef4410004c5112/1 aqaraOpple(["mode"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
info  2022-07-06 18:32:49: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'read' to '0x54ef4410004c5112' failed: 'Error: Read 0x54ef4410004c5112/1 aqaraOpple([\"mode\"], {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":4447,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received)'","meta":{"friendly_name":"0x54ef4410004c5112"},"type":"zigbee_publish_error"}'
error 2022-07-06 18:32:53: Publish 'set' 'read' to '0x54ef4410004c5112' failed: 'Error: Read 0x54ef4410004c5112/1 aqaraOpple(["mode","illuminance"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
info  2022-07-06 18:32:53: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'read' to '0x54ef4410004c5112' failed: 'Error: Read 0x54ef4410004c5112/1 aqaraOpple([\"mode\",\"illuminance\"], {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":4447,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received)'","meta":{"friendly_name":"0x54ef4410004c5112"},"type":"zigbee_publish_error"}'
info  2022-07-06 18:41:04: Successfully interviewed '0x54ef4410004c5112', device has successfully been paired
warn  2022-07-06 18:41:04: Device '0x54ef4410004c5112' with Zigbee model 'undefined' and manufacturer name 'undefined' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
[... cut ...]
info  2022-07-06 18:41:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0x54ef4410004c5112","supported":false},"type":"pairing"}'
info  2022-07-07 14:56:39: Removing device '0x54ef4410004c5112' (block: false, force: true)
info  2022-07-07 14:56:39: MQTT publish: topic 'zigbee2mqtt/0x54ef4410004c5112', payload ''
info  2022-07-07 14:56:39: Successfully removed device '0x54ef4410004c5112' (block: false, force: true)
info  2022-07-07 14:56:39: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/remove', payload '{"data":{"block":false,"force":true,"id":"0x54ef4410004c5112"},"status":"ok","transaction":"tpf8h-1"}'`

idomp, 2022-07-31:
Would really like to see support for this lock!

rafaelreis-r, 2022-08-02:
I'm from Brazil and BRL to USD exchange rates are atrocious, but I'm able to provide a 40 USD bounty to get this working.

@Koenkk maybe you would be able to take a look at this or point me out on a direction?

Koenkk, 2022-08-02:
@rafaelreis-r could you make a sniff when pairing the device to the original hub? https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

rafaelreis-r, 2022-08-04:
> @rafaelreis-r could you make a sniff when pairing the device to the original hub? https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

Just received the CC2531 today. Flashed it and compiled whsniff. Files are attached below.
I've also added the pairing process to a known device ([Aqara Mini Switch - WXKG12LM)](https://www.zigbee2mqtt.io/devices/WXKG12LM.html) for reference. 

**Attachements:**

- [A100 repairing + actions.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/9263582/A100.repairing.%2B.actions.pcapng.gz)

- [button+transportkey.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/9263584/button%2Btransportkey.pcapng.gz)

Notes:

- Transport Key is in the Device Announcement packet in the capture. 
- This was more challenging than I anticipated. But I suppose everything was sniffed properly.

Koenkk, 2022-08-06:
@rafaelreis-r

- Could you also provide me a sniff when pairing it to z2m?
- What command is send when the lock is locked/unlocked? (can you indicate the `No.` in wireshark?)

rafaelreis-r, 2022-08-08:
@Koenkk 

> Could you also provide me a sniff when pairing it to z2m?

- [A100 Z2M Pairing.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/9284568/A100.Z2M.Pairing.pcapng.gz)
- [log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9284569/log.txt) - Zigbee Herdsman debug log. 

I tried using a converter based on [lumi.lock.acn02](https://github.com/Koenkk/zigbee-herdsman-converters/blob/73b58ef67edb9c8c19929c05cd4e163cadf029fa/devices/xiaomi.js#L1508), since the A100 reports itself as `lumi.lock.acn01`, but didn't seem to work (I don't discard having screwed up somewhere).

I had to spin up a separate Z2M instance since my network is very busy atm, and I needed to be close to the lock. 
All I had was a windows laptop, so the logs will reflect that. 

> What command is send when the lock is locked/unlocked? (can you indicate the No. in wireshark?)

This is the sequence of events reported in the door lock log. Note that it is from new to old, so reverse of what wireshark logged. `"Me"` refers to me. It reports the user which did the action. If it was my wife for instance, it would say her `Name` instead of `"Me"`

Unfortunately I'm unable to align `No.` with `time`. I could do a new dump paying attention to the `No.` if you need.

- [lock_log.jpg](https://user-images.githubusercontent.com/57492577/183503088-2c2cade7-d784-4500-992c-577134bbb7fa.jpeg)

Koenkk, 2022-08-08:
So the first issue that needs to be tackled is that the device leaves almost immediately after pairing it to z2m (packet 71 in `A100 Z2M Pairing.pcapng.gz`). I've compared the z2m vs the xiaomi gateway sniff and everything is the same until the `Node Descriptor Response` (packet 117/67). In the z2m sniff the device leaves 2 seconds after this. In the Xiaomi sniff some encrypted commands are send after this (it seems Xiaomi uses a custom encryption key for this). My first suspicion is that the small differences in the node descriptor response may cause this but unfortunately I cannot change the adapter the node descriptor response because the Conbee firmware is not opensource. Do you also have access to a Texas Instruments based adapter?

rafaelreis-r, 2022-08-08:
@Koenkk 

I only own the Conbee-II and the CC2531 flashed for sniffing. Electronics are highly tariffed and expensive in Brazil. I usually import them directly from China (which is what I did with the door lock) but it takes 45 to 60 days for customs clearance. 

I could purchase another TI device. The only options I could find in stock atm are:

- Another CC2531
- SONOFF Zigbee 3.0 USB Dongle Plus  (CC2652P)

What would you technically recommend?

Koenkk, 2022-08-09:
The SONOFF Zigbee 3.0 USB Dongle Plus (CC2652P)

rafaelreis-r, 2022-08-15:
Hey @Koenkk 

Device arrived. It came preloaded with `Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210708,"transportrev":2},"type":"zStack3x0"}'`

Let me know if it requires updating (I still have to learn the update process). 

I took a sniff and debug logs from the pairing process to this device:

- [A100 pairing to CC2652P .pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/9338131/A100.pairing.to.CC2652P.pcapng.gz)
- [log.txt.zip](https://github.com/Koenkk/zigbee2mqtt/files/9338132/log.txt.zip)

Interesting to note this time the lock looped trying to pair to the coordinator until it gave up after 4 or 5 attempts, as you will be able to see in the dump.

Let me know what else you need me to do!

Koenkk, 2022-08-16:
So it seems the device stays connected for a bit longer and z2m is able to retrieve some data from it. It is important to use the latest firmware on your adapter; otherwise the node descriptor returns an incorrect manufacturer code for Xiaomi devices. There are some instructions on how to flash the fw here: https://www.zigbee2mqtt.io/guide/adapters/#notes (on this page you can also find the correct coordinator firmware).

What worries me a bit is that the device now leaves after it doesn't receive a reply to the encrypted magic command. With the Xiaomi gateway you see the the gateway replies:

[image]

With z2m it doesn't (which of course we cannot do since we cannot encrypt this message):

[image]

I think this causes the device to leave. Seems Xiaomi doesn't want this to be device to be used with other gateways. If the coordinator firmware update doesn't solve the issue I'm quite certain this is the reason the device leaves. So in order to be able to continue we need the encryption key. I see 2 scenarios to obtain this:
- Xiaomi provides us with the encryption key
- We do a brute-force; however this doesn't seem to be a realistic option. The key is 128 bits which results in the following computation time ([source](http://kth.diva-portal.org/smash/get/diva2:1460818/FULLTEXT01.pdf)):
[image]

rafaelreis-r, 2022-08-16:
Hey @Koenkk,

Updated the firmware to `Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20220219,"transportrev":2},"type":"zStack3x0"}'` and redid the sniffs / logs. For some reason, z2m and zigbee-herdsman debug logs were very verbose this time.

- [A100 pairing (latest coord fw).pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/9354396/A100.pairing.latest.coord.fw.pcapng.gz)
- [log.txt.zip](https://github.com/Koenkk/zigbee2mqtt/files/9354391/log.txt.zip)

I hope we aren't locked out due to the encryption key. There is a 3rd way to get them, similar to what this engineer did recently with the embedded linux embedded os fw in his car. We might be able to find the key inside the source files / update files to Aqara's hubs. 

[It is a pretty interesting read, despite offtopic.](https://programmingwithstyle.com/posts/howihackedmycar/)

Koenkk, 2022-08-17:
Strangely enough I don't see the magic commands anymore in this sniff but the device still leaves. What I find interesting is that the `Transport key` message when paired with the Xiaomi gateway uses the non-standard TC link key (at least I guess this is the `transport key` because it comes right after the `assocation response` and there are not any other `transport key` messages).

[image]

I guess that we need the Xiaomi encryption key to continue further with this.

wifispray, 2022-08-23:
Would love to see support for this device

NdR91, 2022-08-28:
Would love to use this device with z2m...

IamMattM, 2022-10-02:
> Would love to use this device with z2m...

Same for me.

waterdrop01, 2022-10-03:
I encourage anyone interested into this issue to report it on the Aqara support page => https://static-resource.aqara.com/html/feedback.html, 

Asking them **to share the encryption key**. Here is a message example:

[image]

```
Dear Aqara team,

Looking at the comments in this youtube video (https://www.youtube.com/watch?v=LBAfbBE9-Bo), you can see that A LOT of people WILL NOT BUY the Aqara A100 because it is NOT SUPPORTED by zigbee2mqtt.

The issue is that Aqara A100 firmware is using en encryption key to communicate over zigbee (see details about the issue here => https://github.com/Koenkk/zigbee2mqtt/issues/13087) and WITHOUT this encryption key, it's nearly impossible to support the A100 into zigbee2mqtt.

Could you please consider working out a solution to add support for the A100 into zigbee2mqtt? (like sharing the key with z2m devs)

Having an open system will definitely boost your sales ;)

Thanks!
```

NdR91, 2022-10-03:
Thank you @waterdrop01, I just sent the ticket. 

I have another questione for who already own the Aqara A100: what about the integration in Home Assistant over HomeKit Controller? Is that possible?

rafaelreis-r, 2022-10-03:
> Thank you @waterdrop01, I just sent the ticket. 
> 
> 
> 
> I have another questione for who already own the Aqara A100: what about the integration in Home Assistant over HomeKit Controller? Is that possible?

It is VERY limited in HomeKit. Not a viable solution. Best path so far has been AqaraGateway + rooted hub.

NdR91, 2022-10-03:
> It is VERY limited in HomeKit. Not a viable solution. Best path so far has been AqaraGateway + rooted hub.

I'll probably go with that solution.

What's the benefits using the Rooted gateway instead of the stock one?

IamMattM, 2022-10-03:
I think that it allows custom HomeAssistant AqaraGateway integration to take control over Aqara gateway connected Zigbee devices ?
https://github.com/niceboygithub/AqaraGateway

I got the AqaraGateway M1S-CN myself for other purposes (to register some Mi BLE sensors and get BLE encryption key once registered to use with passive BLE monitor integration)

The M1S-CN is listed as compatible so I think it would possibly allow A100 Zigbee control ...?

rafaelreis-r, 2022-10-03:
@IamMattM it should be. However, if you see https://github.com/AlexxIT/XiaomiGateway3/issues/808 we still have to map out actions to create a handler for the A100 in the integration code. Logs are there but it is quite a bit of work to separate each action and map it to its id. I haven't had the time yet. 

So, currently, lock shows up in HA, but I can only get locked / unlocked status.

rafaelreis-r, 2022-10-03:
@Koenkk 

You should take a look at https://github.com/AlexxIT/XiaomiGateway3/issues/808#issuecomment-1246169859
and the logs attached here: https://github.com/AlexxIT/XiaomiGateway3/issues/808#issuecomment-1248235091

Since the Gateway is rooted, they get Zigbee from within the gateway itself, past the encryption keys. You might be able to work backwards, since we have: Sniffs, network keys from within the gateway, and that action map which I believe refers to the encrypted payload we see in the sniffs.

rafaelreis-r, 2022-10-04:
@Koenkk to simplify things, I collected, grouped and cleaned up the information for you. 

I did a new sniff of G3 hub (rooted) to A100 lock. Unpaired, Paired, did some door lock / unlocking actions. **The difference is that now we have a debug log from mqtt installed from the root in the G3.**

[a100-g3hub.pcapng.zip](https://github.com/Koenkk/zigbee2mqtt/files/9709971/a100-g3hub.pcapng.zip)

[g3-rooted-log.zip](https://github.com/Koenkk/zigbee2mqtt/files/9709975/g3-rooted-log.zip)

Check out this section in particular:

```scala
 2022-10-04 15:47:46  DEBUG  gateway  192.168.1.144: MQTT on_message: zigbee/recv {"cmd":"write","did":"lumi.0","id":73,"params":[{"res_name":"8.0.2085","value":"efbebc24a46e03d18f51fd7bf12f15daCC45"}]}
 2022-10-04 15:47:46  DEBUG  gateway  192.168.1.144: MQTT on_message: zigbee/recv {"cmd":"write","did":"lumi.0","id":74,"params":[{"res_name":"8.0.2109","value":60}]}
 2022-10-04 15:47:46  DEBUG  gateway  192.168.1.144: MQTT on_message: zigbee/send {"cmd":"write_rsp","id":73,"time":1664909266756,"did":"lumi.0","zseq":0,"results":[{"res_name":"8.0.2085","value":"efbebc24a46e03d18f51fd7bf12f15daCC45","error_code":0}]}
 2022-10-04 15:47:46  DEBUG  gateway  192.168.1.144: gateway <= [{'res_name': '8.0.2085', 'value': 'efbebc24a46e03d18f51fd7bf12f15daCC45', 'error_code': 0}]
 2022-10-04 15:47:46  DEBUG  gateway  192.168.1.144: MQTT on_message: gw/54EF44100025A4E4/commands {"commands":[{"command":"lumi open-with-install-code 0x3c {efbebc24a46e03d18f51fd7bf12f15dacc45} {ffffffffffffffff}","postDelayMs":0}]}
 2022-10-04 15:47:46  DEBUG  gateway  192.168.1.144: MQTT on_message: zigbee/send {"cmd":"write_rsp","id":74,"time":1664909266757,"did":"lumi.0","zseq":0,"results":[{"res_name":"8.0.2109","value":60,"error_code":0}]}
 2022-10-04 15:47:46  DEBUG  gateway  192.168.1.144: MQTT on_message: gw/54EF44100025A4E4/executed {"command":"lumi open-with-install-code 0x3c {efbebc24a46e03d18f51fd7bf12f15dacc45} {ffffffffffffffff}"}
 2022-10-04 15:47:51  DEBUG  gateway  192.168.1.144: MQTT on_message: gw/54EF44100025A4E4/devicejoined {"nodeId":"0x63F4","deviceState":16,"deviceType":"0x00FF","timeSinceLastMessage":0,"deviceEndpoint":{"eui64":"0x54EF4410004C5112","endpoint":0,"clusterInfo":[]},"firstjoined":1}
 2022-10-04 15:47:52  DEBUG  gateway  192.168.1.144: MQTT on_message: gw/54EF44100025A4E4/MessageReceived {"sourceAddress":"0x63F4","eui64":"0x54EF4410004C5112","destinationEndpoint":"0x01","clusterId":"0x0000","profileId":"0x0104","sourceEndpoint":"0x01","APSCounter":"0x9E","APSPlayload":"0x184B0A01002001","rssi":-74,"linkQuality":104}
 2022-10-04 15:47:52  DEBUG  gateway  192.168.1.144: MQTT on_message: gw/54EF44100025A4E4/commands {"commands":[{"commandcli":"zcl mfg-code 0x0000"},{"commandcli":"zcl global read 0x0000 0x0001"},{"commandcli":"send 0x63f4 1 1","postDelayMs":0}]}
 2022-10-04 15:47:52  DEBUG  gateway  192.168.1.144: MQTT on_message: gw/54EF44100025A4E4/executed {"command":"zcl mfg-code 0x0000"}
 2022-10-04 15:47:52  DEBUG  gateway  192.168.1.144: MQTT on_message: gw/54EF44100025A4E4/executed {"command":"zcl global read 0x0000 0x0001"}
 2022-10-04 15:47:52  DEBUG  gateway  192.168.1.144: MQTT on_message: gw/54EF44100025A4E4/MessagePreSentCallback {"eui64":"0x54EF4410004C5112","destinationEndpoint":"0x01","clusterId":"0x0000","profileId":"0x0104","sourceEndpoint":"0x01","APSCounter":"0x79","APSPlayload":"0x1054000100"}
 2022-10-04 15:47:52  DEBUG  gateway  192.168.1.144: MQTT on_message: gw/54EF44100025A4E4/executed {"command":"send 0x63f4 1 1"}
 2022-10-04 15:47:52  DEBUG  gateway  192.168.1.144: MQTT on_message: gw/54EF44100025A4E4/MessageReceived {"sourceAddress":"0x63F4","eui64":"0x54EF4410004C5112","destinationEndpoint":"0x01","clusterId":"0x0000","profileId":"0x0104","sourceEndpoint":"0x01","APSCounter":"0x9F","APSPlayload":"0x184C0A0500421161716172612E6C6F636B2E61636E303031","rssi":-74,"linkQuality":104}
 2022-10-04 15:47:52  DEBUG  gateway  192.168.1.144: MQTT on_message: gw/54EF44100025A4E4/MessageReceived {"sourceAddress":"0x63F4","eui64":"0x54EF4410004C5112","destinationEndpoint":"0x01","clusterId":"0x0000","profileId":"0x0104","sourceEndpoint":"0x01","APSCounter":"0xA0","APSPlayload":"0x1854010100002001","rssi":-74,"linkQuality":104}
 2022-10-04 15:47:52  DEBUG  gateway  192.168.1.144: MQTT on_message: zigbee/send {"cmd":"report","id":2000003389,"did":"lumi.0","dev_src":"0","zseq":0,"params":[{"res_name":"8.0.2084","value":{"did":"lumi.54ef4410004c5112","mac":"54ef4410004c5112","model":"aqara.lock.acn001","version":"0.0.0_0001","zb_ver":"3.0","joined_type":1}}]}
 2022-10-04 15:47:52  DEBUG  gateway  192.168.1.144: MQTT on_message: gw/54EF44100025A4E4/MessageReceived {"sourceAddress":"0x63F4","eui64":"0x54EF4410004C5112","destinationEndpoint":"0x01","clusterId":"0xFCC0","profileId":"0x0104","sourceEndpoint":"0x01","APSCounter":"0xA1","APSPlayload":"0x1C5F114D0AF3FF41413E86A6879DAB55D293E7947CD188A350F68A3B104B68D0A879FE0409846D6BEDF20D95DAA19921C3905C3352A7D668D103D0CA4341295E8585645345EDFE693546","rssi":-73,"linkQuality":108}
 2022-10-04 15:47:53  DEBUG  gateway  192.168.1.144: MQTT on_message: zigbee/send {"cmd":"report","id":2000003390,"did":"lumi.54ef4410004c5112","time":1664909272956,"rssi":-73,"zseq":77,"params":[{"res_name":"20.8.85","value":"{\"encry\":\"1eeebfe53ee5438344ca1242692c6aff827d1105288ff1bee180a0e653ddf8e6540c5548fbf9a5b3531ea6469e226d450c18d03e0ea0a8db06c45f2184d14b89\",\"devcry\":\"86a6879dab55d293e7947cd188a350f68a3b104b68d0a879fe0409846d6be [... cut ...]
 2022-10-04 15:47:53  DEBUG  gateway  192.168.1.144: MQTT on_message: gw/54EF44100025A4E4/executed {"command":"plugin network-creator-security close-network"}
 2022-10-04 15:47:53  DEBUG  gateway  192.168.1.144: MQTT on_message: zigbee/send {"cmd":"write_rsp","id":75,"time":1664909272990,"did":"lumi.0","zseq":0,"results":[{"res_name":"8.0.2109","value":0,"error_code":0}]}
```

This is he APS handshake during pairing. It is way beyond my knowledge, but I was wondering there is  any useful info in there? Maybe the APS Key?

rafaelreis-r, 2022-10-04:
@Koenkk more discovery:

It is clear to me that the devices are communicating over [ZigBee Smart Energy Standard (pdf)](https://zigbeealliance.org/wp-content/uploads/2019/11/docs-07-5356-19-0zse-zigbee-smart-energy-profile-specification.pdf). 

According to the standard:

[image]

> Smart Energy profile uses the same network layer security mechanism as HA, but has further APS layer security which uses a key transferred with certicom certificates which are very secure (as it is typically utility companies using this for metering and do not want the meter reading tampered with).

The encryption key should maybe be these certificates, then?

ChipWolf, 2022-12-26:
@Koenkk can we destale this?

Hobbit44, 2023-02-08:
Could this be reopened and pinned please?

Hobbit44, 2023-05-22:
@Koenkk

TerryRhodes, 2023-08-01:
I assume there's nothing to update on this yet, eh? :(

rafaelreis-r, 2023-08-01:
Zigbee Smart Energy is an entirely different standard as compared to standard Zigbee protocol. It would take:

1. Adding support to ZSE in Z2M
2. Find a workaround to obtain root certificates from manufacturer, be it shared by them or obtained by reverse engineering. Without those certificates, there is no way to derive the transport key for the device.

TerryRhodes, 2023-08-08:
> Zigbee Smart Energy is an entirely different standard as compared to standard Zigbee protocol. It would take:
> 
> 1. Adding support to ZSE in Z2M
> 2. Find a workaround to obtain root certificates from manufacturer, be it shared by them or obtained by reverse engineering. Without those certificates, there is no way to derive the link key for the device.
> 
> Another way would be to have the community develop a custom firmware for the A100 (N100 and other locks in that family) that supports standard zigbee.

And I assume the custom firmware path is not currently being worked on by anyone? Shame if not.

alectogeek, 2024-01-11:
The lack of support for HA stops me from buying their product, LOL. 
Shame on you, Aqara!

NdR91, 2024-01-11:
> The lack of support for HA stops me from buying their product, LOL. 
> 
> Shame on you, Aqara! 

At the end of all I went for a very classical electromechanical lock. I'll connect it to a Shelly 1 soon and to an UPS.
Afrer that with an ESP i will manage the fingerprint sensor.

All managed by HA.

Yes, this isn't the same prpduct but imho it will work even better.

vantuyen3103, 2024-03-22:
Is there any update on this?

sulliwane, 2024-04-01:
> A100 Bluetooth integration on HA seems to be stable now. Requires Bluetooth range.
> 
> Pairing it to HomeKit controller also works.
> 
> Over zigbee the only way is through AqaraGateway on supported gateways.

May I ask what type of sensors do you get with Bluethooth connection? Here is what I have using Aqara Gateway:
[image]

Also, what is the pairing method if using bluetooth? (My HA supports bluetooth so it may be more reliable than the Aqara Gateway integration that sometimes does not start properly).

Thanks :)

raarts, 2024-04-15:
> A100 Bluetooth integration on HA seems to be stable now. Requires Bluetooth range.

Does that mean I can now use the lock with all functionality on my HA? Any more info on how to set this up, or is it automatic?

JustAnotherPU, 2024-06-19:
same question here guys, can we integrate the doorlock to HA now ? I have a D200I what about this one ?

mainmind83, 2024-10-08:
https://www.home-assistant.io/blog/2024/09/03/aqara-joins-works-with-home-assistant/

same situation...

SemonCat, 2024-11-06:
> A100 Bluetooth integration on HA seems to be stable now. Requires Bluetooth range.
> 
> Pairing it to HomeKit controller also works.
> 
> Over zigbee the only way is through AqaraGateway on supported gateways.

which Bluetooth integration can integrate a100?

kevin-olbrich, 2024-11-06:
> > A100 Bluetooth integration on HA seems to be stable now. Requires Bluetooth range.
> > Pairing it to HomeKit controller also works.
> > Over zigbee the only way is through AqaraGateway on supported gateways.
> 
> which Bluetooth integration can integrate a100?

https://esphome.io/components/bluetooth_proxy.html

almirus, 2024-11-06:
@JustAnotherPU @mainmind83 
https://github.com/niceboygithub/AqaraGateway
[image]

GoshaGinzburg, 2025-01-01:
> > > A100 Bluetooth integration on HA seems to be stable now. Requires Bluetooth range.
> > 
> > which Bluetooth integration can integrate a100?
> 
> https://esphome.io/components/bluetooth_proxy.html

Hi! Can you explain how to connect aqara A100 with esphome bluetooth proxy? Thanks.

mainmind83, 2025-01-01:
> @JustAnotherPU @mainmind83 
> https://github.com/niceboygithub/AqaraGateway
> [image]
> 

In mi case problems with IPv6, QNAP and HA:
 https://www.mainmind.com/blog/ipv6-nas-qnap/

sprint1849, 2025-01-24:
Hi, I have been following this thread.

I have successfuly set up Aqara A100 Pro with E1 Hub using niceboy's Aqara Gateway integration.

It's all fun and game until I block internet to the gateway and you restart the gateway or there is power interruption.

The setup is perfect for me, however, this is my only problem. I need my setup to be offline.

Is there any way how to do this?

makleso6, 2026-10-04:
For my **Aqara A100 Pro (`aqara.lock.acn001`, `Lumi`)**, an experimental Zigbee2MQTT integration maintained the connection, decrypted reports and physically unlocked the door. Findings come from BLE/Zigbee captures, Aqara Home and firmware analysis (**0043**). They do not establish complete upstream support or compatibility with other variants.

## 1. Joining a Zigbee network

An **authenticated Aqara BLE connection** configures the target Zigbee **PAN ID and channel** and starts joining: `01 3F 01 <PAN_ID_LE16> <channel_u8>` before BLE encryption/framing. Operation `02` unbinds Zigbee. The coordinator needs permit-join enabled.

The **install code can be requested over authenticated BLE (`03/19`)** and provisioned on the coordinator for the lock's IEEE address. **I also verified joining without preloading this code on the coordinator.** It is therefore not universally mandatory in the tested setups. The lock still receives a Zigbee network key.

Initially, the lock joined and completed its interview, repeated FFF3 challenges, then left after approximately **52 seconds**. Completing the proprietary FFF3 exchange subsequently allowed it to remain connected.

## 2. Different keys have different purposes

| Material | Purpose |
| --- | --- |
| Zigbee install code / derived link key | Zigbee commissioning; not the factory application secret. |
| Zigbee network key, 16 bytes | Normal Zigbee network-layer protection. |
| Factory secret `K`, 16 bytes | AES-128-ECB wrapping of public P-256 points during FFF3 session establishment. |
| FFF3 session key, 32 bytes | Locally calculated ECDH result; AES-256-ECB protection of `0x93` messages. |
| Remote-unlock key, 32 bytes, plus key ID | Additional AES-256-CBC protection of the inner unlock command. Registered in the lock separately. |
| Aqara BLE / HomeKit credentials | Separate authentication and transport keys; they are not `K` or the FFF3 session key. |

The firmware loads `K` from a factory block in **nRF52840 flash**, outside the application image. Its value was absent from the analysed OTA; no normal BLE/Zigbee export was found. **APPROTECT blocked ordinary SWD memory reads on the tested hardware.** The factory secret remains unknown.

## 3. Where Aqara cloud participates

The working client uses M3-style **CoAP-over-TLS**, endpoint `/v1.0/lumi/dev/assist/encryption`, with valid hub credentials. An Aqara Home account token alone is insufficient for this implementation.

For a new FFF3 handshake:

1. The lock sends `3E || wrapped_public_point_64`.
2. The integration generates a P-256 pair and sends `{did, encryStr, decryStr}`: its public point and the lock's wrapped point.
3. Cloud returns `encryValue` (wrapped integration point) and `decryValue` (unwrapped lock point). The integration sends `3E || encryValue` to the lock.
4. Both sides derive the **32-byte session key locally through ECDH**. The integration's private key is not uploaded; cloud returns neither the session key nor `K`.

**Hypothesis:** Aqara selects the factory secret by the supplied lock DID. A DID-indexed database or device-specific cryptographic service is plausible; the actual server storage/derivation mechanism is unverified.

**A new FFF3 session currently needs cloud assistance; individual commands need no new assist request.** Established-session encryption is local. Early Z2M code still fetched cloud time for unlocking; the later Homey implementation used local timestamps. Persisted session state/counters survived process restart without another unlock; a new handshake still requires assistance.

## 4. Unlocking over Zigbee

Transport: **EP1 / manufacturer `0x115F` / cluster `0xFCC0` / attribute `0xFFF3`**, ZCL Write Attributes / octet string. `0x3E` carries key exchange; `0x93` carries encrypted messages.

Opcode `01 74 01`, timestamp fields and CRC are encrypted with the remote-unlock key; key ID and ciphertext are then protected by FFF3 session encryption. FFF3 sequence counters and the inner packet ID are maintained separately. Standard `0x0101` Door Lock unlock attempts did not actuate my lock; this path produced encrypted `81 74 00` **and confirmed physical unlocking**.

A remote-unlock key was also registered locally through **authorised HomeKit → Aqara BLE → `01/8A`**, without Aqara Home/cloud for that registration. This does not replace FFF3 cloud assistance. The key/key ID must match the current registration. Remote operation was enabled in the successful test; its necessity was not isolated. Clock tolerance is unmeasured.

M3 also uses `FCC0/00FF` for a separate authentication exchange; the successful experimental FFF3 client did not implement it.

## 5. Sequence overview

```mermaid
sequenceDiagram
    participant B as Authenticated BLE client
    participant L as A100 Pro
    participant Z as Zigbee coordinator + integration
    participant C as Aqara cloud
    opt Optional coordinator install-code provisioning
        B->>L: Request install code - 03/19
        L-->>B: Install code
        B->>Z: Register code for lock IEEE
    end
    B->>L: Start Zigbee join with PAN ID and channel
    L->>Z: Join and obtain network key
    L->>Z: FFF3 / 3E - wrapped public point
    Z->>C: DID + lock wrapped point + integration public point
    C-->>Z: Unwrapped lock point + wrapped integration point
    Z->>L: FFF3 / 3E - wrapped integration point
    Note over L,Z: Both calculate the session key locally using ECDH
    opt Separate remote-unlock key registration
        B->>L: Aqara BLE 01/8A - register key
        L-->>B: Status and key ID
    end
    Z->>L: FFF3 / 93 - locally encrypted unlock
    L-->>Z: FFF3 / 93 - encrypted result or event
```