Koenkk/zigbee2mqtt #13094: [New device support]: HC-020 ZigBee thermostat

KGS501, 2022-07-10:
### Link

https://www.aliexpress.com/item/1005003472952492.html

### Database entry

'{"data":{"definition":{"description":"Moes BHT series Thermostat","exposes": [{"features":[{"access":3,"description":"Enables/disables physical input on the device","name":"stat e","property":"child_lock","type":"binary","value_off":"UNLOCK","value_on":"LOCK"}],"type":"lock"},{ "access":3,"description":"The delta between local_temperature and current_heating_setpoint to trigge r Heat. 1-5","name":"deadzone_temperature","property":"deadzone_temperature","type":"numeric","unit" :"°C"},{"access":3,"description":"Maximum temperature limit","name":"max_temperature_limit","propert y":"max_temperature_limit","type":"numeric","unit":"°C"},{"features":[{"access":3,"description":"Tem perature setpoint","name":"current_heating_setpoint","property":"current_heating_setpoint","type":"n umeric","unit":"°C","value_max":30,"value_min":5,"value_step":1},{"access":1,"description":"Current temperature measured on the device","name":"local_temperature","property":"local_temperature","type" :"numeric","unit":"°C"},{"access":3,"description":"Offset to be used in the local_temperature","name ":"local_temperature_calibration","property":"local_temperature_calibration","type":"numeric","unit" :"°C","value_max":20,"value_min":-20,"value_step":1},{"access":3,"description":"Mode of this device" ,"name":"system_mode","property":"system_mode","type":"enum","values":["off","heat"]},{"access":1,"d escription":"The current running state","name":"running_state","property":"running_state","type":"en um","values":["idle","heat","cool"]},{"access":3,"description":"Mode of this device (similar to syst em_mode)","name":"preset","property":"preset","type":"enum","values":["hold","program"]},{"access":3 ,"description":"Select temperature sensor to use","name":"sensor","property":"sensor","type":"enum", "values":["IN","AL","OU"]}],"type":"climate"},{"access":1,"description":"Link quality (signal streng th)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"va lue_min":0}],"model":"BHT-002-GCLZB","options":[],"supports_ota":false,"vendor":"Moes"},"friendly_na me":"0x60a423fffe650bb1","ieee_address":"0x60a423fffe650bb1","status":"successful","supported":true} ,"type":"device_interview"}'

### Comments

by default, the device picks up the converter for Moes

{modelID: 'TS0601', manufacturerName: '_TZE200_ztvwu4nk'}

but is missing program values and eco mode available in the thermostat settings. You can also force the unit change and min setpoint value.

The device has a similar way of reporting weekly schedules to Moes TRV {modelID: 'TS0601', manufacturerName: '_TZE200_b6wax7g0'} but manipulating converters got me nowhere so I explored tuya account instead.

I recorded these from my Tuya account:

eco
{
    "startRowId": "",
    "pageNo": 1,
    "pageSize": 10,
    "code": "4",
    "startTime": 1640728857333,
    "endTime": 1640901657333,
    "projectCode": "p1614864188469ysacj7",
    "sourceId": "eu1562663373867mMCv3",
    "sourceType": "4",
    "deviceId": "bfab86deb42a3e8ba7ir66",
    "pageStartRow": "",
    "region": "EU"
}

weekly schedule
{
    "startRowId": "",
    "pageNo": 1,
    "pageSize": 10,
    "code": "30",
    "startTime": 1640729106944,
    "endTime": 1640901906944,
    "projectCode": "p1614864188469ysacj7",
    "sourceId": "eu1562663373867mMCv3",
    "sourceType": "4",
    "deviceId": "bfab86deb42a3e8ba7ir66",
    "pageStartRow": "",
    "region": "EU"
}

working mode  (eco mode is missing)

{
    "startRowId": "",
    "pageNo": 1,
    "pageSize": 10,
    "code": "2",
    "startTime": 1640729245324,
    "endTime": 1640902045325,
    "projectCode": "p1614864188469ysacj7",
    "sourceId": "eu1562663373867mMCv3",
    "sourceType": "4",
    "deviceId": "bfab86deb42a3e8ba7ir66",
    "pageStartRow": "",
    "region": "EU"
}

set lower limit
{
    "startRowId": "",
    "pageNo": 1,
    "pageSize": 10,
    "code": "20",
    "startTime": 1640729278965,
    "endTime": 1640902078966,
    "projectCode": "p1614864188469ysacj7",
    "sourceId": "eu1562663373867mMCv3",
    "sourceType": "4",
    "deviceId": "bfab86deb42a3e8ba7ir66",
    "pageStartRow": "",
    "region": "EU"
}

units  C and F

{
    "startRowId": "",
    "pageNo": 1,
    "pageSize": 10,
    "code": "23",
    "startTime": 1640729303238,
    "endTime": 1640902103238,
    "projectCode": "p1614864188469ysacj7",
    "sourceId": "eu1562663373867mMCv3",
    "sourceType": "4",
    "deviceId": "bfab86deb42a3e8ba7ir66",
    "pageStartRow": "",
    "region": "EU"
}

current temp (reported temp needs to be multiplied by 10)

{
    "startRowId": "",
    "pageNo": 1,
    "pageSize": 10,
    "code": "24",
    "startTime": 1640729325008,
    "endTime": 1640902125008,
    "projectCode": "p1614864188469ysacj7",
    "sourceId": "eu1562663373867mMCv3",
    "sourceType": "4",
    "deviceId": "bfab86deb42a3e8ba7ir66",
    "pageStartRow": "",
    "region": "EU"
}

switch relay 
{
    "startRowId": "",
    "pageNo": 1,
    "pageSize": 10,
    "code": "1",
    "startTime": 1640729356867,
    "endTime": 1640902156867,
    "projectCode": "p1614864188469ysacj7",
    "sourceId": "eu1562663373867mMCv3",
    "sourceType": "4",
    "deviceId": "bfab86deb42a3e8ba7ir66",
    "pageStartRow": "",
    "region": "EU"
}

KGS501, 2022-08-10:
noo

KGS501, 2022-09-12:
unstale it please

KGS501, 2022-10-13:
I give up

xworld21, 2022-11-07:
I have implemented this as external converter, I'll send a PR some time this month. The fingerprint clashes with an existing converter, which is quite annoying as I don't know how to solve the clash.

In the meanwhile, if anybody following this issue can test it, I'd be very grateful!
[HC-020-ZB.js.zip](https://github.com/Koenkk/zigbee2mqtt/files/9955007/HC-020-ZB.js.zip)

KGS501, 2022-11-13:
Thank you so much!

closed 2022-12-22 by github-actions[bot]