Koenkk/zigbee2mqtt #13097: [New device support]: Tuya Smart ZigBee Energy Meter 80A Din Rail

Marc--, 2022-07-11:
### Link

https://fr.aliexpress.com/item/1005004399475951.html

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0xa4c1381bb74f67ff","nwkAddr":64237,"manufId":4417,"manufName":"_TZE200_eaac7dkw","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u001b@^*f\u001b@^*\u0012","65506":48,"65508":0,"stackVersion":0,"dateCode":"","appVersion":68}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00158d0000ce1622","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":821693351},"lastSeen":1657497965729,"defaultSendRequestWhen":"immediate"}

### Comments

Hi there !

I followed the guide on adding new Tuya devices, but without success (must be a lack of knowledge on my side !).

On the side of the device it says "DAC2161C". I found this but not very useful: 
[https://expo.tuya.com/product/942088]

After writing the converter and using "read_tuya_dump.py" script, i got :

```
2022-07-11T01:58:04 0xa4c1381bb74f67ff (seq:  256, value: #0) [DP   6 unknown     ] => raw: [   9  59   0   0   0   0   0   0 ]
2022-07-11T01:58:04 0xa4c1381bb74f67ff (seq:  512, value: #0) [DP   1 unknown     ] => int: 1
2022-07-11T01:58:04 0xa4c1381bb74f67ff (seq:  768, value: #0) [DP   3 unknown     ] => raw: [   0   7   0   7   0   0   0   1 ]
2022-07-11T01:58:04 0xa4c1381bb74f67ff (seq: 1024, value: #0) [DP   4 unknown     ] => raw: [   7  11   7  11   0   0   0   0 ]
2022-07-11T01:58:04 0xa4c1381bb74f67ff (seq: 1280, value: #0) [DP  11 unknown     ] => raw: [   1   0 ]
2022-07-11T01:58:04 0xa4c1381bb74f67ff (seq: 1536, value: #0) [DP  17 unknown     ] => raw: [   1   1   0  80 ]
2022-07-11T01:58:04 0xa4c1381bb74f67ff (seq: 1792, value: #0) [DP  16 unknown     ] => bool: False
2022-07-11T01:58:04 0xa4c1381bb74f67ff (seq: 1792, value: #1) [DP  18 unknown     ] => string: 22062703
```

So "DP16" is Relay Switch, i can use it with MQTT payload "bool,16,0" (or 1)
After hours of research and tries, it was a joyfull moment.

Then i found issue #7155 for ""_TZE200_byzdayie" and it seam very similar. So i try to add my manufacturerName to the model "TS0601_din" (DDS238-2) in tuya.js. No success.

I also tried an external converter without success.

I'm quite lost, i don't know how to go farther. I don't understant point 5 of the guide.  Can someone help me please ?

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different data points
    // it's usually necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_eaac7dkw'
        },
    ],
    model: 'DAC2161C',
    vendor: 'Tuya',
    description: 'DIN Rail Smart Energy Meter',
    fromZigbee: [
        fz.ignore_basic_report, // Add this if you are getting no converter for 'genBasic'
        fz.tuya_data_point_dump, // This is a debug converter, it will be described in the next part
		fz.tuya_dinrail_switch,
    ],
    toZigbee: [
        tz.tuya_data_point_test, // Another debug converter
		tz.tuya_switch_state,
    ],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [
        // Here you should put all functionality that your device exposes
    ],
};

module.exports = definition;
```

bootuz-dinamon, 2022-07-20:
DP 6  -  Phase A
DP 1 - Forward Energy Total
DP 3 - Month Energy
DP 4 - Daily Energy
DP 11 - Frozen set
DP 16 - Switch
DP 17 - Alarm Set
DP 10 - Fault
DP 18 - Meter ID
DP 21, 22, 23, 24 - Forward energy

bootuz-dinamon, 2022-07-23:
```
2022-07-23T11:30:25 0xa4c1383a1774b16b (seq:  256, value: #0) [DP   6 Phase A] => raw: [   8 194   0   0   0   0   0   0 ]
2022-07-23T11:30:26 0xa4c1383a1774b16b (seq:  512, value: #0) [DP   1 Forward energy total  ] => int: 59
2022-07-23T11:30:26 0xa4c1383a1774b16b (seq:  768, value: #0) [DP   3 month energy] => raw: [  22   7  22   7   0   0   0   0 ]
2022-07-23T11:30:26 0xa4c1383a1774b16b (seq: 1024, value: #0) [DP   4 daily energy] => raw: [   7  23   7  23   0   0   0   0 ]
2022-07-23T11:30:26 0xa4c1383a1774b16b (seq: 1280, value: #0) [DP  11 Frozen set] => raw: [  22  15 ]
2022-07-23T11:30:26 0xa4c1383a1774b16b (seq: 1536, value: #0) [DP  17 Alarm set] => raw: [   3   1   1   5   1   1   0  40 ]
2022-07-23T11:30:26 0xa4c1383a1774b16b (seq: 1792, value: #0) [DP  16 switch   ] => bool: True
2022-07-23T11:30:26 0xa4c1383a1774b16b (seq: 1792, value: #1) [DP  18 Meter ID ] => string: 22062315
```

bootuz-dinamon, 2022-08-27:
Please, add this device.

europanda, 2022-08-28:
I'm also really looking forward to supporting this device.

Antunes43, 2022-09-15:
> Hello. Also trying to make a converter for this device. To date, the device can be turned on and off, information about the mains voltage, current, power and consumed total power is correctly transmitted. I haven't had much success yet.
> 
> Обзор устройства на русском языке и инструкцию по добавлению конвертера ищите тут - https://youtu.be/KwTz35OWmP4
> 
> 27/07/2022 - added Threshold and Alarm (not all)
> 
> ```
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const extend = require('zigbee-herdsman-converters/lib/extend');
> const e = exposes.presets;
> const ea = exposes.access;
> const tuya = require("zigbee-herdsman-converters/lib/tuya");
> 
> 
> 
> const fzLocal = {
>     tuya_xocadinrail_switch: {
>         cluster: 'manuSpecificTuya',
>         type: ['commandDataResponse', 'commandDataReport'],
>         convert: (model, msg, publish, options, meta) => {
>             for (const dpValue of msg.data.dpValues) {
>                 const value = tuya.getDataValue(dpValue);
>                 const dp = dpValue.dp;
>                 meta.logger.info(`RECEIVED DP #${dp} -- VALUE = ${value}`);
>                 
>                 switch (dp) {
>                     case 16: // DPID that we added to common
>                        return {   state: value ? 'ON' : 'OFF'};
>                        
>                       
>                     case 1:
>                         return {
>                             energy: value / 100
>                         };
>                     case 6:   
>                      return {
>                             current: ((value[4] | value[3] << 8) / 1000), voltage: ((value[1] | value[0] << 8) / 10), power: ((value[7] | value[6] << 8))
>                         };
>                         
>                     case 17:
>                       const lookup = {
>                        0: 'not set',
>                        1: 'Over current threshold',
>                        3: 'Over voltage threshold',
>                                       };
>                         return {
>                             threshold_1: lookup[value[0]],
>                             threshold1_value : (value[3] | value[2] << 8),
>                             threshold_2: lookup[value[4]],
>                             threshold2_value : (value[7] | value[6] << 8),                            
>                     };    
>                         
>                       case 10:{
>                           const lookup = {
>                           0: 'clear',
>                           1: 'over current threshold',
>                           2: 'over power threshold',
>                           4: 'over voltage threshold',
>                           8: 'wrong frequency threshold',
>                            };
>                            return { alarm : lookup[value]} ;
>                     }
>                     case 18:{
>                         meta.logger.warn(`zigbee-herdsman-converters:: NOT RECOGNIZED DP ` +
>                             `#${dp} with data ${JSON.stringify(msg.data)} VALUE = ${value}`);
>                             break;
>                     }
>                     default: {
>                         meta.logger.warn(`zigbee-herdsman-converters:: NOT RECOGNIZED DP ` +
>                             `#${dp} with data ${JSON.stringify(msg.data)} VALUE = ${value}`);
>                             break;
>                     }
>                 }
>             }
>         },
>     },
> };
> 
> 
> const tzLocal = {
>     state: {
>         key: ['state'],
>         convertSet: async (entity, key, value, meta) => {
>             await tuya.sendDataPointBool(entity, 16, value === 'ON');
>         },
>     },
> };
> 
> 
> const definition = {
>     fingerprint: [{
>         modelID: 'TS0601',
>         manufacturerName: '_TZE200_eaac7dkw'
>     }],
>     model: 'DAC2161C',
>     vendor: 'Tuya',
>     extend: extend.switch(),
>     description: 'DIN Rail Smart Energy Meter',
>     fromZigbee: [fzLocal.tuya_xocadinrail_switch, ],
>     toZigbee: [tzLocal.state],
>     configure: async (device, coordinatorEndpoint, logger) => {
>         const endpoint = device.getEndpoint(1);
>         await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
>     },
>     exposes: [e.switch().setAccess('state', ea.STATE_SET), 
>               e.voltage(), 
>               e.power(), 
>               e.current(), 
>               e.energy(),
>               exposes.text('threshold_1', ea.STATE),
>               exposes.text('threshold1_value', ea.STATE),
>               exposes.text('threshold_2', ea.STATE),
>               exposes.text('threshold2_value', ea.STATE),
>               exposes.text('alarm', ea.STATE),
> 
>               ] 
> 
> };
> 
> module.exports = definition;
> ```
hello there, i tried using it with my DDS238-2 and it kinda of works, but i dont get my voltage, or any of the other data, like frequency, reactive power and that kinda of stuff, can you enlighten us to how u found what each raw datapoint was? and why u shift them 8 bits?

bootuz-dinamon, 2022-09-20:
_TZE200_lsanae15  

Another ID for dac2161c

Koenkk, 2022-10-04:
`_TZE200_lsanae15` is supported now

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

europanda, 2022-10-04:
> `_TZE200_lsanae15` is supported now

My device is defined as _TZE200_eaac7dkw

Koenkk, 2022-10-05:
@europanda added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Elreinodemu, 2022-10-10:
Ive followed the instructions in the video for my _TZE200_Isanae15 using the post from 6 days ago to create the  dac2161.js file and it still shows as not supported. I dont know how to go from here. Please help

europanda, 2022-10-10:
> Ive followed the instructions in the video for my _TZE200_Isanae15 using the post from 6 days ago to create the dac2161.js file and it still shows as not supported. I dont know how to go from here. Please help

Try  _TZE200_eaac7dkw

Elreinodemu, 2022-10-10:
> _TZE200_Isanae15

please explain a bit more, im new at this

europanda, 2022-10-10:
> please explain a bit more, im new at this

set model _TZE200_Isanae15 or _TZE200_eaac7dkw in line **manufacturerName: '_TZE200_eaac7dkw'**

Elreinodemu, 2022-10-10:
Done it, it doesnt work

europanda, 2022-10-10:
> Done it, it doesnt work

See the correct name on the device properties page.

europanda, 2022-10-10:
@Koenkk converter not work properly. Version: 1.28.0-dev commit: 0a71e39
almost all readings are null or 0
`{
    "current": 0,
    "energy": 608.47,
    "linkquality": 72,
    "state": "ON",
    "voltage": 0,
    "ac_frequency": null,
    "power": null,
    "power_factor": null
}`

Koenkk, 2022-10-11:
Then this device probably uses different data points. To find them out see https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#requirements-and-caveats

europanda, 2022-10-11:
> Then this device probably uses different data points. To find them out see https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#requirements-and-caveats
@Koenkk I don 't have an account and a Tuya coordinator . The device is connected directly to z2m.

europanda, 2022-10-11:
> Hello. Also trying to make a converter for this device. To date, the device can be turned on and off, information about the mains voltage, current, power and consumed total power is correctly [...]

This converter works fine.

Koss2001, 2022-11-04:
Any thoughts on _TZE200_lsanae15? I can't get it into the Home Assistant.

prokudin07, 2022-11-12:
> 0_lsanae15? I ca

Create [PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/4938) with right exposes