Koenkk/zigbee2mqtt #13098: Third button on Aurora Lighting AU-A1ZBRC remote

enzodesena, 2022-07-11:
### What happened?

The device works well for the "on/off" button, and for the "brightness up" and "brightness down" buttons, but not for the third button. I previously used the same remote with a Smartthings hub, and it worked properly, both for a short and a long click. I believe the zigbee2mqtt device simply ignores the relevant messages, so this is perhaps more of a feature request? 

I would be happy to try to improve the device myself, but I can't see zigbee messages coming in when clicking that button, also with the "debug" toggle added.  

### Zigbee2MQTT version

1.26.0-1

### Adapter firmware version

latest

### Adapter

Sonoff Universal Zigbee 3.0 USB Dongle Plus

Koenkk, 2022-07-11:
Can you provide the data/database.db entry of this device?

enzodesena, 2022-07-12:
@Koenkk Thanks for the message. Here you go:

{"id":20,"type":"EndDevice","ieeeAddr":"0x00158d0002122960","nwkAddr":34631,"manufId":4456,"manufName":"Aurora","powerSource":"Battery","modelId":"Remote50AU","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":2096,"inClusterList":[0,3,4096],"outClusterList":[4,3,6,8,5,4096,25],"clusters":{"genBasic":{"attributes":{"modelId":"Remote50AU","manufacturerName":"Aurora","powerSource":3,"zclVersion":2,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20181106","swBuildId":"2.4"}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c8d0ad","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c8d0ad","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c8d0ad","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20181106","swBuildId":"2.4","zclVersion":2,"interviewCompleted":true,"meta":{"configured":3},"lastSeen":1657578639497,"defaultSendRequestWhen":"immediate"}

Koenkk, 2022-07-12:
Could you check if any `No converter available..` messages are produced with the following external converter?

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['Remote50AU'],
    model: 'AU-A1ZBRC',
    vendor: 'Aurora Lighting',
    description: 'AOne smart remote',
    fromZigbee: [fz.battery, fz.command_on, fz.command_off, fz.command_step],
    toZigbee: [],
    exposes: [e.battery(), e.action(['on', 'off', 'brightness_step_up', 'brightness_step_down'])],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl', 'genPowerCfg', 'genScenes']);
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m

Koenkk, 2022-07-14:
You are probably adding it to the home assistant `configuration.yaml`, it should be added to the zigbee2mqtt `configuration.yaml`.

enzodesena, 2022-07-14:
Of course! Silly me. Yes, I receive ``No converter available..`` messages: 

For a short press: 
```
2022-07-14 17:42:08Received Zigbee message from 'Bedroom remote', type 'commandRecall', cluster 'genScenes', data '{"groupid":0,"sceneid":1}' from endpoint 1 with groupID 0
2022-07-14 17:42:08No converter available for 'AU-A1ZBRC' with cluster 'genScenes' and type 'commandRecall' and data '{"groupid":0,"sceneid":1}'
```

For a long press:
```
2022-07-14 17:43:57Received Zigbee message from 'Bedroom remote (Pepo's side)', type 'commandStore', cluster 'genScenes', data '{"groupid":0,"sceneid":1}' from endpoint 1 with groupID 0
Debug 2022-07-14 17:43:57No converter available for 'AU-A1ZBRC' with cluster 'genScenes' and type 'commandStore' and data '{"groupid":0,"sceneid":1}'
```

Koenkk, 2022-07-15:
Cool, the following should expose the actions:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    command_store: {
        cluster: 'genScenes',
        type: 'commandStore',
        convert: (model, msg, publish, options, meta) => {
            if (hasAlreadyProcessedMessage(msg)) return;
            const payload = {action: postfixWithEndpointName(`store_${msg.data.sceneid}`, msg, model)};
            addActionGroup(payload, msg, model);
            return payload;
        },
    }
}

const definition = {
    zigbeeModel: ['Remote50AU'],
    model: 'AU-A1ZBRC',
    vendor: 'Aurora Lighting',
    description: 'AOne smart remote',
    fromZigbee: [fz.battery, fz.command_on, fz.command_off, fz.command_step, fz.command_recall, fzLocal.command_store],
    toZigbee: [],
    exposes: [e.battery(), e.action(['on', 'off', 'brightness_step_up', 'brightness_step_down', 'recall_1', 'store_1'])],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl', 'genPowerCfg', 'genScenes']);
    },
};

module.exports = definition;
```

enzodesena, 2022-07-15:
@Koenkk Thanks a lot! It is *almost* done. The ``recall_1`` action works fine, but the ``store_1`` (that is, the long press) returns this error: 

```
2022-07-15 16:08:32Received Zigbee message from 'Bedroom remote', type 'commandStore', cluster 'genScenes', data '{"groupid":0,"sceneid":1}' from endpoint 1 with groupID 0
Error 2022-07-15 16:08:32Exception while calling fromZigbee converter: hasAlreadyProcessedMessage is not defined}
```

Koenkk, 2022-07-15:
@enzodesena I've updated https://github.com/Koenkk/zigbee2mqtt/issues/13098#issuecomment-1185626171, should be fixed now.

enzodesena, 2022-07-15:
Almost perfect! @Koenkk You only need to remove the trailing space in ``require('zigbee-herdsman-converters/lib/utils ');``. Then it works like a charm.

 Will you update the device or would you like me to create a pull request?

Koenkk, 2022-07-17:
Updated the code!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-07-17 by Koenkk

enzodesena, 2022-09-15:
Hi @Koenkk It looks like in one of the latest updates the long press fails with error "Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'logger')}". Any advice?

enzodesena, 2022-09-15:
Hi @Koenkk. Thanks for the quick reply. Does this help?

Debug 2022-09-15 09:12:08Received Zigbee message from 'Bedroom remote', type 'commandStore', cluster 'genScenes', data '{"groupid":0,"sceneid":1}' from endpoint 1 with groupID 0
Error 2022-09-15 09:12:08Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'logger')}
Debug 2022-09-15 09:12:08TypeError: Cannot read properties of undefined (reading 'logger') at Object.postfixWithEndpointName (/app/node_modules/zigbee-herdsman-converters/lib/utils.js:118:14) at Object.convert (/app/dist/util/externally-loaded.js:18:44) at Receive.onDeviceMessage (/app/lib/extension/receive.ts:143:51) at EventEmitter.emit (node:events:539:35) at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22) at Controller.<anonymous> (/app/lib/zigbee.ts:106:27) at Controller.emit (node:events:527:28) at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:503:14) at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:714:18) at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:144:70)

Koenkk, 2022-09-16:
Can you remove your external converter and provide me the error log again?

enzodesena, 2022-09-16:
@Koenkk Sure can do. The logs look nearly identical to me (I removed "- ext_converter.js" from the configuration file and rebooted HA):

Debug 2022-09-16 10:51:35Received Zigbee message from 'Bedroom remote', type 'commandStore', cluster 'genScenes', data '{"groupid":0,"sceneid":1}' from endpoint 1 with groupID 0
Error 2022-09-16 10:51:35Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'logger')}
Debug 2022-09-16 10:51:35TypeError: Cannot read properties of undefined (reading 'logger') at Object.postfixWithEndpointName (/app/node_modules/zigbee-herdsman-converters/lib/utils.js:118:14) at Object.convert (/app/dist/util/externally-loaded.js:18:44) at Receive.onDeviceMessage (/app/lib/extension/receive.ts:143:51) at EventEmitter.emit (node:events:539:35) at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22) at Controller.<anonymous> (/app/lib/zigbee.ts:106:27) at Controller.emit (node:events:527:28) at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:503:14) at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:714:18) at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:144:70)

Koenkk, 2022-09-19:
`Object.convert (/app/dist/util/externally-loaded.js:18:44) at` are you using any external extensions? (check your `configuration.yaml`)

enzodesena, 2022-09-20:
Hi @Koenkk you are quite right, I did have some other external extensions. I removed all of them and the error is no longer there, which somewhat expected. What is strange though is that when I added them back in (I need them as a couple of devices are not yet covered) the error no longer appeared. Strange, anyways, it works now, so everyone happy! Thanks.