Koenkk/zigbee2mqtt #13105: Zigbee2MQTT fails to start

1HyperZ, 2022-07-11:
### What happened?

from some reason the owner closed it by saying that i should check the link in the error though i wrote i looked it up and it showed nothing so posting it again

when i try to npm start i get this error: pi@raspberrypi:/opt/zigbee2mqtt $ npm start

[zigbee2mqtt@1.26.0-dev](mailto:zigbee2mqtt@1.26.0-dev) start
node index.js

Zigbee2MQTT:debug 2022-07-11 16:24:37: Loaded state from file /opt/zigbee2mqtt/data/state.json
Zigbee2MQTT:info 2022-07-11 16:24:37: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2022-07-11.16-24-37' filename: log.txt
Zigbee2MQTT:debug 2022-07-11 16:24:37: Removing old log directory '/opt/zigbee2mqtt/data/log/2022-07-11.16-22-02'
Zigbee2MQTT:info 2022-07-11 16:24:37: Starting Zigbee2MQTT version 1.26.0-dev (commit #14216b74)
Zigbee2MQTT:info 2022-07-11 16:24:37: Starting zigbee-herdsman (0.14.43)
Zigbee2MQTT:debug 2022-07-11 16:24:37: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_8619b34b3f98ec118254a1d044d80d13-if00-port0"}}'
Zigbee2MQTT:error 2022-07-11 16:24:38: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-07-11 16:24:38: Failed to start zigbee
Zigbee2MQTT:error 2022-07-11 16:24:38: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-07-11 16:24:38: Exiting...
Zigbee2MQTT:error 2022-07-11 16:24:38: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
at SerialPort. (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:146:28)
at SerialPort._error (/opt/zigbee2mqtt/node_modules/@serialport/stream/lib/index.js:198:14)
at /opt/zigbee2mqtt/node_modules/@serialport/stream/lib/index.js:242:12

i tried checking if something else was using it but it showed nothing: pi@raspberrypi:/opt/zigbee2mqtt $ sudo ls -l /proc/[0-9]/fd/ |grep /dev/ttyUSB0
pi@raspberrypi:/opt/zigbee2mqtt $

tried flashing the stick to the latest version but it didnt help it stayed the same

### Zigbee2MQTT version

dont know how to find

### Adapter firmware version

dont know how to find

### Adapter

sonoff zigbee stick

narc792, 2022-07-19:
Had a similar issue back in June and didn’t receive any response. Interested in hearing if you get it functional again.

King-anderson, 2022-07-24:
similar issue, Koenkk/zigbee-herdsman#556 , Koenkk/zigbee2mqtt#12847.

closed 2022-09-01 by github-actions[bot]