Koenkk/zigbee2mqtt #13111: power outage memory for SONOFF ZBMINI

delta010, 2022-07-12:
### What happened?

after the power outage SONOFF ZBMINI does not come back to the last state (on/off), after the power outage it goes to the off state. every time the power goes off if the relay is on I need to turn the switch on again when the power gets back

### What did you expect to happen?

before power outage = ON
after power outage = ON

       or

before power outage = OFF
after power outage = OFF

### How to reproduce it (minimal and precise)

use  Sonoff ZBMINI with zigbee2mqtt

### Zigbee2MQTT version

1.26.0-1 

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Debug log

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/socat.sh
[13:24:53] INFO: Socat not enabled, marking service as down
cont-init: info: /etc/cont-init.d/socat.sh exited 0
cont-init: info: running /etc/cont-init.d/zigbee2mqtt.sh
cont-init: info: /etc/cont-init.d/zigbee2mqtt.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun socat (no readiness notification)
services-up: info: copying legacy longrun zigbee2mqtt (no readiness notification)
[13:24:55] INFO: Handing over control to Zigbee2MQTT Core ...
> zigbee2mqtt@1.26.0 start
> node index.js
Zigbee2MQTT:info  2022-07-12 11:17:57: Succesfully changed options
Zigbee2MQTT:info  2022-07-12 11:17:57: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"enzrh-1"}'
Zigbee2MQTT:info  2022-07-12 11:18:11: MQTT publish: topic 'zigbee2mqtt/first floor bedroom plug_1', payload '{"linkquality":105,"state":"ON"}'
Zigbee2MQTT:info  2022-07-12 11:18:12: MQTT publish: topic 'zigbee2mqtt/first floor bedroom plug_1', payload '{"linkquality":138,"state":"OFF"}'
Zigbee2MQTT:info  2022-07-12 11:18:21: MQTT publish: topic 'zigbee2mqtt/first floor bedroom plug_1', payload '{"linkquality":120,"state":"ON"}'
Zigbee2MQTT:info  2022-07-12 11:18:28: MQTT publish: topic 'zigbee2mqtt/first floor main light', payload '{"linkquality":69,"state":"OFF"}'
Zigbee2MQTT:info  2022-07-12 11:18:31: MQTT publish: topic 'zigbee2mqtt/first floor bedroom plug_1', payload '{"linkquality":138,"state":"ON"}'
Zigbee2MQTT:info  2022-07-12 11:18:37: MQTT publish: topic 'zigbee2mqtt/cc2530 router', payload '{"led":false,"linkquality":54}'

Otnow, 2022-07-12:
Apparently this feature is not supported by the device: https://github.com/Koenkk/zigbee2mqtt/issues/5901.

delta010, 2022-07-12:
is that a hardware limitation or software limitation? I think the ewelink app has the option, is there any way to retain the last state after a power outage

Otnow, 2022-07-12:
> is that a hardware limitation or software limitation?

It's hard to say, but I think it's a software limitation.

> I think the ewelink app has the option, is there any way to retain the last state after a power outage

I don't have this device, so I can't confirm or deny 🤷‍♂️.

delta010, 2022-07-13:
On the z2m supported devices page, there is no mention of power outage memory in  SONOFF BASICZBR3 But the device support that function

closed 2022-08-20 by github-actions[bot]