Koenkk/zigbee2mqtt #13122: [New device support]: Develco Panic Button - PBTZB-110

qvntra, 2022-07-13:
### Link

https://www.develcoproducts.com/products/sensors-and-alarms/panic-button/

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0x0015bc00390037cd","nwkAddr":19422,"manufId":4117,"manufName":"Develco Products A/S","powerSource":"Battery","modelId":"PBTZB-110","epList":[1,35],"endpoints":{"1":{"profId":49353,"epId":1,"devId":1,"inClusterList":[5,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"35":{"profId":260,"epId":35,"devId":1026,"inClusterList":[0,1,3,32,1280],"outClusterList":[3,10,25],"clusters":{"genBasic":{"attributes":{"modelId":"PBTZB-110","manufacturerName":"Develco Products A/S","powerSource":3,"zclVersion":3,"dateCode":"2022-04-01 11:03"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b001caa6daa","zoneState":1,"zoneId":23}}},"binds":[{"cluster":1280,"type":"endpoint","deviceIeeeAddress":"0x00124b001caa6daa","endpointID":242}],"configuredReportings":[],"meta":{}}},"dateCode":"2022-04-01 11:03","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1370898936},"lastSeen":1657678379772}

### Comments

Hello,

Long-time supporter of @Koenkk and help out quite a bit in the Discord. I've written converters myself but I have never had to:

1. Enroll End-Device in IAS Zone client
2. Zone Enroll Response is sent from Coordinator to End Client

In addition, to turn off the "panic" alarm, I need the Coordinator to send "Default Response on Zone Status Change Notification". Otherwise, I cannot turn off the panic.

Would greatly appreciate any help in implementing this coordinator. I will pay $50 for a working converter via your preferred payment method. 

Thank you for any consideration.

[image]
[image]

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['PBTZB-110'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'PBTZB-110', // Vendor model number, look on the device for a model number
    vendor: 'Develco Products A/S', // Vendor of the device (only used for documentation and startup logging)
    description: 'Panic Button', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.ias_sos_alarm_2, fz.command_status_change_notification_action, fz.ias_wd, fz.develco_fw, fz.ias_enroll], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.sos()],
    configure: async (device, coordinatorEndpoint, logger) => {
              const options = {manufacturerCode: 4117};
              const endpoint = device.getEndpoint(35);
              await reporting.bind(endpoint, coordinatorEndpoint, ['ssIasZone']);
              await endpoint.read('ssIasZone', ['iasCieAddr', 'zoneState', 'zoneId']);
            },
};

module.exports = definition;
```

qvntra, 2022-07-13:
```tsk@tsk-ThinkPad-X1-Carbon-5th:/opt/zigbee2mqtt$ npm start

> zigbee2mqtt@1.21.2 start /opt/zigbee2mqtt
> node index.js

Zigbee2MQTT:debug 2022-07-13T10:58:01: Loaded state from file /opt/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2022-07-13T10:58:01: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2022-07-13.10-58-01' filename: log.txt
Zigbee2MQTT:debug 2022-07-13T10:58:01: Removing old log directory '/opt/zigbee2mqtt/data/log/2022-07-12.21-19-52'
Zigbee2MQTT:info  2022-07-13T10:58:01: Starting Zigbee2MQTT version 1.21.2 (commit #44922120)
Zigbee2MQTT:info  2022-07-13T10:58:01: Starting zigbee-herdsman (0.13.149)
Zigbee2MQTT:debug 2022-07-13T10:58:01: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[25],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
Zigbee2MQTT:info  2022-07-13T10:58:05: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2022-07-13T10:58:05: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20191106,"transportrev":2},"type":"zStack3x0"}'
Zigbee2MQTT:debug 2022-07-13T10:58:05: Zigbee network parameters: {"channel":25,"extendedPanID":"0x00124b001caa6daa","panID":6754}
Zigbee2MQTT:info  2022-07-13T10:58:05: Currently 1 devices are joined:
Zigbee2MQTT:info  2022-07-13T10:58:05: 0x0015bc00390037cd (0x0015bc00390037cd): Not supported (EndDevice)
Zigbee2MQTT:warn  2022-07-13T10:58:05: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2022-07-13T10:58:05: Allowing new devices to join.
Zigbee2MQTT:warn  2022-07-13T10:58:05: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2022-07-13T10:58:05: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2022-07-13T10:58:05: Connecting to MQTT server at mqtt://localhost:1883
Zigbee2MQTT:debug 2022-07-13T10:58:05: Using MQTT anonymous login
Zigbee2MQTT:info  2022-07-13T10:58:05: Connected to MQTT server
Zigbee2MQTT:info  2022-07-13T10:58:05: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:info  2022-07-13T10:58:05: MQTT publish: topic 'zigbee2mqtt/0x0015bc00390037cd/availability', payload 'online'
Zigbee2MQTT:debug 2022-07-13T10:58:05: Received MQTT message on 'zigbee2mqtt/bridge/info' with data '{"commit":"44922120","config":{"advanced":{"adapter_concurrent":null,"adapter_delay":null,"availability_blacklist":[],"availability_blocklist":[],"availability_passlist":[],"availability_timeout":0,"availability_whitelist":[],"cache_state":true,"cache_state_persistent":true,"cache_state_send_on_startup":true,"channel":11,"elapsed":false,"ext_pan_id":[221,221,221,221,221,221,221,221],"homeassistant_discovery_topic":"homeassistant","homeassistant_legacy_entity_attributes":false,"homeassistant_legacy_triggers":true,"homeassistant_status_topic":"hass/status","last_seen":"disable","legacy_api":false,"log_directory":"/opt/zigbee2mqtt/data/log/%TIMESTAMP%","log_file":"log.txt","log_level":"info","log_output":["console","file"],"log_rotation":true,"log_symlink_current":false,"log_syslog":{},"pan_id":6754,"report":false,"soft_reset_timeout":0,"timestamp_format":"YYYY-MM-DD HH:mm:ss"},"ban":[],"blocklist":[],"device_options":{"legacy":false},"devices":{"0x0015bc0039003a2c":{"friendly_name":"0x0015bc0039003a2c"}},"experimental":{"output":"json"},"external_converters":[],"groups":{},"homeassistant":false,"log_level":"debug","map_options":{"graphviz":{"colors":{"fill":{"coordinator":"#e04e5d","enddevice":"#fff8ce","router":"#4ea3e0"},"font":{"coordinator":"#ffffff","enddevice":"#000000","router":"#ffffff"},"line":{"active":"#009900","inactive":"#994444"}}}},"mqtt":{"base_topic":"zigbee2mqtt","force_disable_retain":false,"include_device_information":false,"server":"mqtt://localhost:1883"},"ota":{"disable_automatic_update_check":false,"update_check_interval":1440},"passlist":[],"permit_join":true,"serial":{"disable_led":false,"port":"/dev/ttyACM0"},"whitelist":[]},"config_schema":{"definitions":{"device":{"properties":{"debounce":{"description":"Debounces messages of this device","title":"Debounce","type":"number"},"debounce_ignore":{"description":"Protects unique payload values of specified payload properties from overriding within debounce time","examples":["action"],"items":{"type":"string"},"title":"Ignore debounce","type":"array"},"filtered_attributes":{"description":"Filter attributes from publish payload.","examples":["temperature","battery","action"],"items":{"type":"string"},"title":"Filtered publish attributes","type":"array"},"filtered_optimistic":{"description":"Filter attributes from optimistic publish payload when calling /set. (This has no effect if optimistic is set to false).","examples":["color_mode","color_temp","color"],"items":{"type":"string"},"title":"Filtered optimistic attributes","type":"array"},"friendly_name":{"description":"Used in the MQTT topic of a device. By default this is the device ID","readOnly":true,"title":"Friendly name","type":"string"},"icon":{"description":"The user-defined device icon for the frontend. It can be a link to an image (not a path to a file) or base64 encoded data URL like: image/svg+xml;base64,PHN2ZyB3aW....R0aD","title":"Icon","type":"string [... cut ...]
Zigbee2MQTT:debug 2022-07-13T11:01:40: Saving state to file /opt/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2022-07-13T11:01:40: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
Zigbee2MQTT:info  2022-07-13T11:01:40: Disconnecting from MQTT server
Zigbee2MQTT:info  2022-07-13T11:01:40: Stopping zigbee-herdsman...
Zigbee2MQTT:info  2022-07-13T11:01:48: Stopped zigbee-herdsman
Zigbee2MQTT:info  2022-07-13T11:01:48: Stopped Zigbee2MQTT
```

https://user-images.githubusercontent.com/67597954/178779138-1b266ef6-f668-42fb-80f2-d485c5cc07ab.MOV

Koenkk, 2022-07-13:
I see two possible ways to integrate this device:
- Immediately respond the alarm message with a default response
- Allow clearing the alarm by publishing to e.g. `zigbee2mqtt/MY_DEVICE/set` `{"clear_alarm": ""}`

What makes the most sense?

qvntra, 2022-07-13:
Wow. So the push and hold feature does nothing but disables the LED. I really disagree with the firmware design here then. I will email Develco.

Can you help me understand how to perform option #1?
I am able to stand-up option #2 already.

Thank you again for your consideration. Sending $50 when I get back from client.

Koenkk, 2022-07-14:
For the first option you can go the `onEvent` route, an example can be found here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/23e8391b9e28906d1444197e7b87303a77257c29/devices/adeo.js#L21

qvntra, 2022-07-14:
You're the man. Saved our implementation for this client. Tip sent. Closing issue.

qvntra, 2022-07-17:
Re-opening because I thought this would be an easy fix but it is not. This issue exists for many outstanding bugs including #3012 #5489 #1809 #8321. Truly, every Develco sensor that operates on IAS (almost all of them) does not work completely without fixing this.

@SwoopX supposedly fixed it for Deconz. I spoke with him on Discord and he cited following Zigbee spec correctly for proper default response for IAS devices.

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3775
https://github.com/dresden-elektronik/deconz-rest-plugin/pull/3732/commits/ed80334d89ce3f90684cffe212f9818b950a5ec7

I'm willing to tip $100 to get this fixed. We've got an install on Thursday and I bet the farm on these PBTZB-110's. Would really like to not overnight Aqara buttons on Monday.

Koenkk, 2022-07-17:
Does it work correctly now with this converter? If yes please make a pull request.

qvntra, 2022-07-25:
[image]

Solved the issue with big thanks to @SwoopX. Device is expecting Direction: 0 (Client to Server) vs Direction: 1 (Server to Client). 

https://github.com/Koenkk/zigbee-herdsman/blob/master/src/controller/model/endpoint.ts#L621

For my device, I changed line 621 to `options = this.getOptionsWithDefaults(options, true, Zcl.Direction.CLIENT_TO_SERVER, null);` and device functions perfectly.

@SwoopX says typically, you want to "switch" the direction in the response. @Koenkk do you know how / where we should implement this?

Koenkk, 2022-07-25:
With: 

```js
          await data.endpoint.defaultResponse(0, 0, 1280, data.meta.zclTransactionSequenceNumber, {"disableDefaultResponse": false, direction: 0});
```

qvntra, 2022-07-26:
`bee2MQTT:debug 2022-07-25T20:29:08: Received MQTT message on 'zigbee2mqtt/bridge/groups' with data '[]'
Zigbee2MQTT:debug 2022-07-25T20:29:08: Received MQTT message on 'zigbee2mqtt/bridge/config' with data '{"commit":"6484fee","coordinator":{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20200925,"transportrev":2},"type":"zStack3x0"},"log_level":"info","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.17.1"}'
Zigbee2MQTT:debug 2022-07-25T20:29:08: Received MQTT message on 'zigbee2mqtt/bridge/extensions' with data '[]'
Zigbee2MQTT:debug 2022-07-25T20:29:08: Passive device '0x0015bc0039003a2c' was last seen '0.05' hours ago.
Zigbee2MQTT:info  2022-07-25T20:29:08: MQTT publish: topic 'zigbee2mqtt/0x0015bc0039003a2c/availability', payload '{"state":"online"}'
Zigbee2MQTT:info  2022-07-25T20:29:08: Started frontend on port 0.0.0.0:5000
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,29,68,129,0,0,0,5,105,251,35,11,0,211,0,197,43,227,0,0,9,9,97,0,50,0,0,23,0,0,105,251] +16s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,29,68,129,0,0,0,5,105,251,35,11,0,211,0,197,43,227,0,0,9,9,97,0,50,0,0,23,0,0,105,251] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [29,114] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,29,68,129,0,0,0,5,105,251,35,11,0,211,0,197,43,227,0,0,9,9,97,0,50,0,0,23,0,0,105,251,29,114] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 29 - 2 - 4 - 129 - [0,0,0,5,105,251,35,11,0,211,0,197,43,227,0,0,9,9,97,0,50,0,0,23,0,0,105,251,29] - 114 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":1280,"srcaddr":64361,"srcendpoint":35,"dstendpoint":11,"wasbroadcast":0,"linkquality":211,"securityuse":0,"timestamp":14887877,"transseqnumber":0,"len":9,"data":{"type":"Buffer","data":[9,97,0,50,0,0,23,0,0]}} +16s
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":97,"manufacturerCode":null,"commandIdentifier":0},"Payload":{"zonestatus":50,"extendedstatus":0},"Command":{"ID":0,"parameters":[{"name":"zonestatus","type":33},{"name":"extendedstatus","type":32}],"name":"statusChangeNotification"}},"address":64361,"endpoint":35,"linkquality":211,"groupID":0,"wasBroadcast":false,"destinationEndpoint":11}' +16s
Zigbee2MQTT:debug 2022-07-25T20:29:24: Received Zigbee message from '0x0015bc0039003a2c', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":50}' from endpoint 35 with groupID 0
  zigbee-herdsman:controller:endpoint DefaultResponse 0x0015bc0039003a2c/35 1280(0, {"sendWhen":"active","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +0ms
  zigbee-herdsman:controller:endpoint ZclFrame {
  zigbee-herdsman:controller:endpoint   Header: {
  zigbee-herdsman:controller:endpoint     frameControl: {
  zigbee-herdsman:controller:endpoint       reservedBits: 0,
  zigbee-herdsman:controller:endpoint       frameType: 0,
  zigbee-herdsman:controller:endpoint       direction: 0,
  zigbee-herdsman:controller:endpoint       disableDefaultResponse: false,
  zigbee-herdsman:controller:endpoint       manufacturerSpecific: false
  zigbee-herdsman:controller:endpoint     },
  zigbee-herdsman:controller:endpoint     transactionSequenceNumber: 97,
  zigbee-herdsman:controller:endpoint     manufacturerCode: null,
  zigbee-herdsman:controller:endpoint     commandIdentifier: 11
  zigbee-herdsman:controller:endpoint   },
  zigbee-herdsman:controller:endpoint   Payload: { cmdId: 0, statusCode: 0 },
  zigbee-herdsman:controller:endpoint   Cluster: {
  zigbee-herdsman:controller:endpoint     ID: 1280,
  zigbee-herdsman:controller:endpoint     attributes: {
  zigbee-herdsman:controller:endpoint       zoneState: [Object],
  zigbee-herdsman:controller:endpoint       zoneType: [Object],
  zigbee-herdsman:controller:endpoint       zoneStatus: [Object],
  zigbee-herdsman:controller:endpoint       iasCieAddr: [Object],
  zigbee-herdsman:controller:endpoint       zoneId: [Object],
  zigbee-herdsman:controller:endpoint       numZoneSensitivityLevelsSupported: [Object],
  zigbee-herdsman:controller:endpoint       currentZoneSensitivityLevel: [Object]
  zigbee-herdsman:controller:endpoint     },
  zigbee-herdsman:controller:endpoint     manufacturerCode: undefined,
  zigbee-herdsman:controller:endpoint     name: 'ssIasZone',
  zigbee-herdsman:controller:endpoint     commands: {
  zigbee-herdsman:controller:endpoint       enrollRsp: [Object],
  zigbee-herdsman:controller:endpoint       initNormalOpMode: [Object],
  zigbee-herdsman:controller:endpoint       initTestMode: [Object]
  zigbee-herdsman:controller:endpoint     },
  zigbee-herdsman:controller:endpoint     commandsResponse: { statusChangeNotification: [Object], enrollReq: [Object] },
  zigbee-herdsman:controller:endpoint     getAttribute: [Function: getAttribute],
  zigbee-herdsman:controller:endpoint     hasAttribute: [Function: hasAttribute],
  zigbee-herdsman:controller:endpoint     getCommand: [Function: getCommand],
  zigbee-herdsman:controller:endpoint     getCommandResponse: [Function: getCommandResponse]
[... cut ...]
  zigbee-herdsman:controller:endpoint     hasAttribute: [Function: hasAttribute],
  zigbee-herdsman:controller:endpoint     getCommand: [Function: getCommand],
  zigbee-herdsman:controller:endpoint     getCommandResponse: [Function: getCommandResponse]
  zigbee-herdsman:controller:endpoint   },
  zigbee-herdsman:controller:endpoint   Command: { ID: 11, name: 'defaultRsp', parameters: [ [Object], [Object] ] }
  zigbee-herdsman:controller:endpoint } +1ms
  zigbee-herdsman:controller:endpoint Sending to 0x0015bc0039003a2c/35 when active +4ms

`

Koenkk, 2022-07-26:
You can disable default responses sent by herdsman with the following code: https://github.com/Koenkk/zigbee-herdsman-converters/blob/60ae7793a136f9421d4a400f650156768900bf45/devices/ewelink.js#L35

qvntra, 2022-07-27:
This works. Thank you. I will publish converter here soon to list. Cheers!

Koenkk, 2022-07-27:
You can also directly create a PR, simply click the pencil icon on [this page](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/develco.js)

qvntra, 2022-07-27:
PBTZB-110 converter = https://github.com/Koenkk/zigbee-herdsman-converters/commit/cc8828fc3c9ed556d1ad58219c711fee4d373371
toZigbee = https://github.com/Koenkk/zigbee-herdsman-converters/commit/009fa490a527dcec4193510e016c17bba5f02167

Koenkk, 2022-07-28:
About the `alarm_auto_cancel_delay`, is this a setting? It should also be added to the `exposes`.

qvntra, 2022-08-04:
> About the `alarm_auto_cancel_delay`, is this a setting? It should also be added to the `exposes`.

Let me figure out how to implement.

closed 2022-09-12 by github-actions[bot]