Koenkk/zigbee2mqtt #13123: [New device support]: Orvibo DS20Z - Zigbee Model ccb9f56837ab41dcad366fb1452096b6

hirzel23, 2022-07-13:
### Link

https://orvibo.ae/downlight/

### Database entry

const definition = { 	zigbeeModel: ['ccb9f56837ab41dcad366fb1452096b6'], 	model: 'ccb9f56837ab41dcad366fb1452096b6', 	vendor: 'ORVIBO', 	description: 'ORVIBO Smart Spotlight', 	extend: extend.light_onoff_brightness_colortemp({colorTempRange: [153, 370]}), };

### Comments

Paired device to sonoff zigbee 3.0 usb dongle plus - paired successfully but not supported

Create ccb9f56837ab41dcad366fb1452096b6.js in zigbee2mqtt folder and external converters has been added in zigbee2mqtt configuration file as follow:

```
external_converters:
  - ccb9f56837ab41dcad366fb1452096b6.js
```

here is my current configuration in ccb9f56837ab41dcad366fb1452096b6.js

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
	zigbeeModel: ['ccb9f56837ab41dcad366fb1452096b6'],
	model: 'ccb9f56837ab41dcad366fb1452096b6',
	vendor: 'ORVIBO',
	description: 'ORVIBO Smart Spotlight',
	extend: extend.light_onoff_brightness_colortemp({colorTempRange: [153, 370]}),
};

module.exports = definition;
```

I'm quite new in HA, after I added and run the toggle it shows error 

> No converter available for 'state' ("OFF")
> No converter available for 'state' ("ON")
> No converter available for 'brightness' (13)
> ...

are there anyone can help for this issue? Please Help

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
	zigbeeModel: ['ccb9f56837ab41dcad366fb1452096b6'],
	model: 'ccb9f56837ab41dcad366fb1452096b6',
	vendor: 'ORVIBO',
	description: 'ORVIBO Smart Spotlight',
	extend: extend.light_onoff_brightness_colortemp({colorTempRange: [153, 370]}),
};

module.exports = definition;
```

### Supported color modes

onoff_brightness_colortemp

### Color temperature range

probably 166 to 370, I don't have the exact information yet

hirzel23, 2022-07-14:
here is the debug log when I try to adjust the brightness down

> groupID 0
> Debug 2022-07-14 11:41:52No converter available for 'ccb9f56837ab41dcad366fb1452096b6' with cluster '65280' and type 'attributeReport' and data '{"0":"0x80b300a600000000"}'
> Debug 2022-07-14 11:41:53Received MQTT message on 'zigbee2mqtt/0x4730000025092188/set' with data '{"brightness":8}'
> Debug 2022-07-14 11:41:53Publishing 'set' 'brightness' to '0x4730000025092188'
> Info 2022-07-14 11:41:53MQTT publish: topic 'zigbee2mqtt/0x4730000025092188', payload '{"brightness":8,"color_mode":"color_temp","color_temp":166,"linkquality":69,"state":"ON"}'
> Debug 2022-07-14 11:41:53Received Zigbee message from '0x4730000025092188', type 'attributeReport', cluster '65280', data '{"0":"0x80b300a600000000"}' from endpoint 1 with groupID 0

Koenkk, 2022-07-25:
Could you provide me the herdsman debug logging when changing the brightness?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

hirzel23, 2022-07-26:
> Could you provide me the herdsman debug logging when changing the brightness?
> 
> See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Hi Koen, thanks for your help. here is the log file

[Debug log herdsman - Orvibo DS20Z - 07262022.txt]
(https://github.com/Koenkk/zigbee2mqtt/files/9185724/Debug.log.herdsman.-.Orvibo.DS20Z.-.07262022.txt)

Koenkk, 2022-07-26:
Can you check if it works with the converter below and provide the herdsman debug logging again?

``
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const tzLocal = {
    brightness: {
        key: ['brightness'],
        options: [exposes.options.transition()],
        convertSet: async (entity, key, value, meta) => {
            await entity.command(
                'genLevelCtrl',
                'moveToLevel',
                {level: Number(value), transtime: 0},
                utils.getOptions(meta.mapped, entity),
            );

            return {state: {brightness: value}};
        },
    },
}

const definition = {
	zigbeeModel: ['ccb9f56837ab41dcad366fb1452096b6'],
	model: 'ccb9f56837ab41dcad366fb1452096b6',
	vendor: 'ORVIBO',
	description: 'ORVIBO Smart Spotlight',
    toZigbee: [tz.on_off, tzLocal.brightness],
	extend: extend.light_onoff_brightness_colortemp({colorTempRange: [153, 370]}),
};

module.exports = definition;
```

hirzel23, 2022-07-27:
Hi Koen, I got error message from when changing the brightness shows "utils not defined"
here are the screenshot and the debug log
[Debug log herdsman - Orvibo DS20Z - 07272022.txt](https://github.com/Koenkk/zigbee2mqtt/files/9194011/Debug.log.herdsman.-.Orvibo.DS20Z.-.07272022.txt)

[image]

Koenkk, 2022-07-27:
I updated https://github.com/Koenkk/zigbee2mqtt/issues/13123#issuecomment-1195564194, can you try again?

hirzel23, 2022-07-27:
Hi Koen, the brightness is working with the brightness that I set, but seems I got some errors on z2m console here are the log and screenshots

[Debug log herdsman - Orvibo DS20Z - 07282022.txt](https://github.com/Koenkk/zigbee2mqtt/files/9201481/Debug.log.herdsman.-.Orvibo.DS20Z.-.07282022.txt)

[image]
[image]

I also got slightly delay when I change the brightness multiple times.

hirzel23, 2022-07-27:
sorry and also here is the configuration

[image]

I got some warning on some line

Koenkk, 2022-07-28:
I've updated https://github.com/Koenkk/zigbee2mqtt/issues/13123#issuecomment-1195564194, can you check if the errors are gone now when setting the brightness.

> I got some warning on some line

This can be ignored

hirzel23, 2022-07-28:
Hi Koen,

The error is gone now, fully responsive, no error and fully functional.

Thanks a lot for your help and really appreciated.

hirzel23, 2022-07-28:
Hi Koen,

The error is gone now, fully responsive, logs also no showing any error and fully functional.

Thanks a lot for your help and really appreciated.

Koenkk, 2022-07-28:
Great, added the device!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-07-28 by Koenkk

hirzel23, 2022-07-29:
Hi Koen,

Sorry to bother you again, this morning I did my test again with the color temperature. it seems showing error as follow

[Debug log herdsman - Orvibo DS20Z - 07292022.txt]

(https://gith
[image]
ub.com/Koenkk/zigbee2mqtt/files/9215324/Debug.log.herdsman.-.Orvibo.DS20Z.-.07292022.txt)

the brightness still ok

Koenkk, 2022-07-29:
I updated https://github.com/Koenkk/zigbee2mqtt/issues/13123#issuecomment-1195564194 this should fix it.

hirzel23, 2022-07-29:
> 

Hi Koen,

Thanks for your update. I have update the js as follow:
[image]

After I update it seems the color temperature did not work and showing error as follow
[image]

here is the log file
[Debug.log.herdsman.-.Orvibo.DS20Z.-.07292022.2.txt](https://github.com/Koenkk/zigbee2mqtt/files/9221895/Debug.log.herdsman.-.Orvibo.DS20Z.-.07292022.2.txt)

Am I use the correct js code? please cmiiw

hirzel23, 2022-07-29:
I also try the second test use orvibo.js file

here is the part of the js code
[image]

Also showing error and the color temperature also not change
[image]

here is the log file
[Debug.log.herdsman.-.Orvibo.DS20Z.-.07292022.3.txt](https://github.com/Koenkk/zigbee2mqtt/files/9221946/Debug.log.herdsman.-.Orvibo.DS20Z.-.07292022.3.txt)

I try both, not sure which one is better. really need your suggestion 😄

Koenkk, 2022-07-30:
Should be fixed with https://github.com/Koenkk/zigbee2mqtt/issues/13123#issuecomment-1195564194

hirzel23, 2022-07-30:
> Should be fixed with [#13123 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/13123#issuecomment-1195564194)

Hi Koen,

It seems still not found the converter
[image]

[image]

Here is the log
[Debug log herdsman - Orvibo DS20Z - 07302022.txt](https://github.com/Koenkk/zigbee2mqtt/files/9225881/Debug.log.herdsman.-.Orvibo.DS20Z.-.07302022.txt)

Koenkk, 2022-07-30:
Are you sure you restarted z2m after updating the converter?

hirzel23, 2022-07-30:
> Are you sure you restarted z2m after updating the converter?

Hi Koen,

Yes, I restart z2m every time after updated converter. I also restart the HA but still the same error

hirzel23, 2022-07-30:
Hi Koen,

Sorry I just miss the z2m config on z2m ui config for external converter 
. After update there now all working fine now
[image]

Thanks a lot Koen