Koenkk/zigbee2mqtt #13125: Device available but attributes not readable

MetroAffe, 2022-07-13:
### What happened?

After I changed the batterys of my Ikea Tradfri Motion sensors its not working anymore.
All other devices working without any problem.

It looks like it’s available but offline (sleep mode?!) I‘m not sure…

I already tried everything. Reset, restart, removed, add new device again …. 
No idea where the Problem is. Maybe can someone explain the log to me.

### Zigbee2MQTT version

1.26.0-1

### Adapter firmware version

?

### Adapter

sonoff zigbee 3.0 usb dongle plus

### Debug log

[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_rsp: [<Status.SUCCESS: 0>]
[0x0000:zdo] No handler for ZDO request:ZDOCmd.Mgmt_Permit_Joining_rsp([<Status.SUCCESS: 0>])

New device 0xf4e9 (80:4b:50:ff:fe:8e:39:44) joined the network
[0xf4e9] Scheduling initialization
Received frame on uninitialized device <Device model=None manuf=None nwk=0xF4E9 ieee=80:4b:50:ff:fe:8e:39:44 is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x81\xe9\xf4D9\x8e\xfe\xffPK\x80\x80'
[0xf4e9:zdo] ZDO request ZDOCmd.Device_annce: [0xF4E9, 80:4b:50:ff:fe:8e:39:44, 128]
Tries remaining: 3
[0xf4e9] Requesting 'Node Descriptor'
Tries remaining: 2
[0xf4e9] Extending timeout for 0x08 request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xF4E9 ieee=80:4b:50:ff:fe:8e:39:44 is_initialized=False> from ep 0 to ep 0, cluster 32770: b'\x08\x00\xe9\xf4\x02@\x80|\x11RR\x00\x00,R\x00\x00'
[0xf4e9] Got Node Descriptor: NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)
[0xf4e9] Discovering endpoints
Tries remaining: 3
[0xf4e9] Extending timeout for 0x09 request
[0x20EF](lumi.sensor_magnet.aq2): Device seen - marking the device available and resetting counter
[0x20EF](lumi.sensor_magnet.aq2): Update device availability -  device available: True - new availability: True - changed: False
Received frame on uninitialized device <Device model=None manuf=None nwk=0xF4E9 ieee=80:4b:50:ff:fe:8e:39:44 is_initialized=False> from ep 0 to ep 0, cluster 32773: b'\t\x00\xe9\xf4\x01\x01'
[0xf4e9] Discovered endpoints: [1]
[0xf4e9] Initializing endpoints [<Endpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
[0xf4e9:1] Discovering endpoint information
Tries remaining: 3
[0xf4e9] Extending timeout for 0x0a request
Received frame on uninitialized device <Device model=None manuf=None nwk=0xF4E9 ieee=80:4b:50:ff:fe:8e:39:44 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'\n\x00\xe9\xf4"\x01\x04\x01P\x08\x01\x07\x00\x00\x01\x00\x03\x00\t\x00 \x00\x00\x10|\xfc\x06\x03\x00\x04\x00\x06\x00\x08\x00\x19\x00\x00\x10'
[0xf4e9:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=2128, device_version=1, input_clusters=[0, 1, 3, 9, 32, 4096, 64636], output_clusters=[3, 4, 6, 8, 25, 4096])
[0xF4E9:1:0x0000] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=11, command_id=<GeneralCommand.Read_Attributes: 0>, *is_reply=False)
[0xF4E9:1:0x0000] Sending request: Read_Attributes(attribute_ids=[4, 5])
[0xf4e9] Extending timeout for 0x0b request
[0xF4E9:1:0x0000] Received ZCL frame: b'\x18\x0B\x01\x04\x00\x00\x42\x0E\x49\x4B\x45\x41\x20\x6F\x66\x20\x53\x77\x65\x64\x65\x6E\x05\x00\x00\x42\x15\x54\x52\x41\x44\x46\x52\x49\x20\x6D\x6F\x74\x69\x6F\x6E\x20\x73\x65\x6E\x73\x6F\x72'
[0xF4E9:1:0x0000] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, is_reply=1, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=11, command_id=1, *is_reply=True)
[0xF4E9:1:0x0000] Decoded ZCL frame: Basic:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x0004, status=<Status.SUCCESS: 0>, value=TypeValue(type=CharacterString, value='IKEA of Sweden')), ReadAttributeRecord(attrid=0x0005, status=<Status.SUCCESS: 0>, value=TypeValue(type=CharacterString, value='TRADFRI motion sensor'))])
[0xf4e9] Read model 'TRADFRI motion sensor' and manufacturer 'IKEA of Sweden' from <Endpoint id=1 in=[basic:0x0000, power:0x0001, identify:0x0003, alarms:0x0009, poll_control:0x0020, lightlink:0x1000, manufacturer_specific:0xFC7C] out=[identify:0x0003, groups:0x0004, on_off:0x0006, level:0x0008, ota:0x0019, lightlink:0x1000] status=<Status.ZDO_INIT: 1>>
[0xf4e9] Discovered basic device information for <Device model='TRADFRI motion sensor' manuf='IKEA of Sweden' nwk=0xF4E9 ieee=80:4b:50:ff:fe:8e:39:44 is_initialized=True>
Device is initialized <Device model='TRADFRI motion sensor' manuf='IKEA of Sweden' nwk=0xF4E9 ieee=80:4b:50:ff:fe:8e:39:44 is_initialized=True>
Checking quirks for IKEA of Sweden TRADFRI motion sensor (80:4b:50:ff:fe:8e:39:44)
Considering <class 'zhaquirks.ikea.motionzha.IkeaTradfriMotionE1745'>
Found custom device replacement for 80:4b:50:ff:fe:8e:39:44: <class 'zhaquirks.ikea.motionzha.IkeaTradfriMotionE1745'>
'button' component -> 'ZHAIdentifyButton' using ['identify']
'sensor' component -> 'Battery' using ['power']
'sensor' component -> 'RSSISensor' using ['basic']
'sensor' component -> 'LQISensor' using ['basic']
device - 0xF4E9:80:4b:50:ff:fe:8e:39:44 entering async_device_initialized - is_new_join: True
device - 0xF4E9:80:4b:50:ff:fe:8e:39:44 has joined the ZHA zigbee network
[0xF4E9](TRADFRI motion sensor): started configuration
[0xF4E9:ZDO](TRADFRI motion sensor): 'async_configure' stage succeeded
[... 304 lines cut ...]
[0x21B2:1:0x0006] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=42, command_id=<GeneralCommand.Read_Attributes: 0>, *is_reply=False)
[0x21B2:1:0x0006] Sending request: Read_Attributes(attribute_ids=[16387])
[0x21b2] Extending timeout for 0x2a request
[0x21B2:1:0x0001]: initializing channel: from_cache: False
[0x21B2:1:0x0001]: initializing uncached channel attributes: ['battery_voltage', 'battery_percentage_remaining'] - from cache[False]
[0x21B2:1:0x0001]: Reading attributes in chunks: ['battery_voltage', 'battery_percentage_remaining']
[0x21B2:1:0x0001] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=43, command_id=<GeneralCommand.Read_Attributes: 0>, *is_reply=False)
[0x21B2:1:0x0001] Sending request: Read_Attributes(attribute_ids=[32, 33])
[0x21b2] Extending timeout for 0x2b request
[0x21B2:1:0x0006]: failed to get attributes '['start_up_on_off']' on 'on_off' cluster: Request failed after 5 attempts: <Status.NWK_INVALID_REQUEST: 194>
[0x21B2:1:0x0006]: async_initialize: retryable request #3 failed: Request failed after 5 attempts: <Status.NWK_INVALID_REQUEST: 194>. Retrying in 3.5s
[0x21B2:1:0x0001]: failed to get attributes '['battery_voltage', 'battery_percentage_remaining']' on 'power' cluster: Request failed after 5 attempts: <Status.NWK_INVALID_REQUEST: 194>
[0x21B2:1:0x0001]: async_initialize: retryable request #3 failed: Request failed after 5 attempts: <Status.NWK_INVALID_REQUEST: 194>. Retrying in 2.7s
[0x21B2:1:0x0001]: initializing channel: from_cache: False
[0x21B2:1:0x0001]: initializing uncached channel attributes: ['battery_voltage', 'battery_percentage_remaining'] - from cache[False]
[0x21B2:1:0x0001]: Reading attributes in chunks: ['battery_voltage', 'battery_percentage_remaining']
[0x21B2:1:0x0001] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=44 [... cut ...]
[0x21B2:1:0x0006]: Reading attributes in chunks: ['start_up_on_off']
[0x21B2:1:0x0006] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, is_reply=0, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=45, command_id=<GeneralCommand.Read_Attributes: 0>, *is_reply=False)
[0x21B2:1:0x0006] Sending request: Read_Attributes(attribute_ids=[16387])
[0x21b2] Extending timeout for 0x2d request

sjorge, 2022-07-13:
The output looks like Python, so I'm guessing you're actually running ZHA. Not familier with homeassistant so can't give instructions on how to double check.

Koenkk, 2022-07-13:
This is ZHA, from your log: `Considering <class 'zhaquirks.ikea.motionzha.IkeaTradfriMotionE1745'>` . So not a z2m issue, please create the issue in the ZHA issue tracker.

closed 2022-07-13 by Koenkk