Koenkk/zigbee2mqtt #13126: EMIZB-132 current calibration on phase b/c

toremick, 2022-07-13:
### What happened?

current_calibrate only affects device_current, not device_current_phase_b or c (same for voltage)

### What did you expect to happen?

That current calibrate would calibrate either all phases, or a current_phase_x_calibrate possibility

### How to reproduce it (minimal and precise)

set a value for current_calibrate

### Zigbee2MQTT version

1.26.0

### Adapter firmware version

20210120

### Adapter

CC2652 board

Koenkk, 2022-07-13:
I expect this already works by setting `current_phase_b_calibrate` or `current_phase_c_calibrate`

toremick, 2022-07-14:
> I expect this already works by setting `current_phase_b_calibrate` or `current_phase_c_calibrate`

Im sorry to say, no. (wrote wrong in my report, its not calibrate, its calibration. The 'current_calibration' is added from web, and the one for phase_b is added manually, then z2m is restarted)

from configuration.yaml

  
' 0x0015bc001b02508c':  
friendly_name: '0x0015bc001b02508c'   
interface_mode: kaifa_and_kamstrup  
current_calibration: -99  
current_phase_b_calibration: -99  
power_calibration: 0  
voltage_calibration: 0  

Result:
[image]

Koenkk, 2022-07-14:
Pushed a possible fix, please check. Note that the calibration is applied only after the device reports a new value.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

toremick, 2022-07-15:
> Pushed a possible fix, please check. Note that the calibration is applied only after the device reports a new value.
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

It works (both b/c) !

[image]

Koenkk, 2022-07-15:
Cool! 😄

closed 2022-07-15 by Koenkk