Koenkk/zigbee2mqtt #13135: HEIMAN HS2WD-E ignoring MQTT-parameters partially

snipah, 2022-07-13:
### What happened?

I assume this topic was raised several times but never resolved.

The siren is now properly recognized in Z2M. Alarm can be triggered via MQTT-command, however all parameter are ignored, i.e. the strobe cannot be triggered. Signal can be stopped w/o problems.

The "Exposes" if the WebIF also do the same: You can start and stop the alarm, but other switches and sliders to not work.

From what I read it seems to work with ZHA, which is not an option for me, but this means that we should a able to get it up and running. Signal-level and battery status are working just fine.

Are these the Zigbee raw commands? I found them in a link here in your GitHub. Does this help somehow?

def off() {
    zigbee.command(0x0502, 0x00, "00", "0000", "00", "00")
}

def on() {
    sendEvent(name: "alarm", value: "on", descriptionText: "Device alarming on", type: "digital")
    both()
}

def both() {
    sendEvent(name: "alarm", value: "both", descriptionText: "Device alarming with siren and strobe", type: "digital")
    zigbee.command(0x502, 0x00, "17", DataType.pack(1, DataType.UINT16), "00", "01")
}

def strobe() {
    sendEvent(name: "alarm", value: "strobe", descriptionText: "Device alarming with strobe", type: "digital")
    zigbee.command(0x502, 0x00, "04", DataType.pack(1, DataType.UINT16), "00", "01")
}

def siren() {
    sendEvent(name: "alarm", value: "siren", descriptionText: "Device alarming with siren", type: "digital")
    zigbee.command(0x502, 0x00, "13", DataType.pack(1, DataType.UINT16), "00", "00")

### What did you expect to happen?

I expect to be able to control the siren with all details. Strobe would be great, volume control would be great.

### How to reproduce it (minimal and precise)

Install everything and use it as described above... simply does not work for me!

### Zigbee2MQTT version

1.26.0-dev commit: 994de20

### Adapter firmware version

0x26580700

### Adapter

ConBee2

### Debug log

Can supply whatever you ask for :)

closed 2022-08-20 by github-actions[bot]

bart40140, 2022-08-20:
hello, I have the same problem as you, do you have a solution?

I also have these errors that come back in the log, maybe a track
`Error 2022-08-20 10:18:03No converter available for 'mode' ("")
Error 2022-08-20 10:18:03No converter available for 'level' ("")
Error 2022-08-20 10:18:03No converter available for 'strobe_level' ("")
Error 2022-08-20 10:18:03No converter available for 'strobe' ("")
Error 2022-08-20 10:18:03No converter available for 'strobe_duty_cycle' ("")
Error 2022-08-20 10:18:03No converter available for 'duration' ("")
Error 2022-08-20 10:18:03No converter available for 'battery' ("")`

luixal, 2023-02-10:
Same problem in my case.

Really annoying as I bought this siren instead of a cheaper one because of z2m support shown in docs.