Koenkk/zigbee2mqtt #13144: [New device support]: ADEO ZBEK-09

inoxas78, 2022-07-14:
### Link

https://www.leroymerlin.fr/produits/decoration-eclairage/ampoule-et-led/ampoule-led/ampoule-e27/ampoule-connectee-led-standard-e27-806-lm-60-w-variation-de-blancs-lexman-84372280.html?fwd=3276007273922

### Database entry

{"id":9,"type":"Router","ieeeAddr":"0x50325ffffe208241","nwkAddr":61576,"manufId":4727,"manufName":"Adeo","powerSource":"Mains (single phase)","modelId":"ZBEK-9","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096,64599,64642],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"ZBEK-9","manufacturerName":"Adeo","powerSource":1,"zclVersion":3,"appVersion":2,"stackVersion":2,"hwVersion":1,"dateCode":"20210714","swBuildId":"2.09.00"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":2,"hwVersion":1,"dateCode":"20210714","swBuildId":"2.09.00","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1657792534286,"defaultSendRequestWhen":"immediate"}

### Comments

very similar with ZBEK-12 : IA-CDZFB2AA007NA-MZN-01
Code on box: IA-CDZFB2AA007NA-MZN-02

01 is with an opaque glass
02 is crytal glass with "filaments"

[image]

### Supported color modes

white Color temperature

### Color temperature range

[153, 454]

Koenkk, 2022-07-14:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-07-14 by Koenkk

inoxas78, 2022-07-14:
❤️❤️