Koenkk/zigbee2mqtt #13148: Lidl/Livarnolux TS0502B TS0505B lights double turnoff transition bug

CZDanol, 2022-07-14:
### What happened?

Heyo, I've noticed this annoying thing that causes the TS0502B and TS0505B  lights to flash off right after color temperature transition end (while still reporting the "ON" state). This seems to rather be a firmware bug in the lights itself, just putting it here for discussion.

I've bodged up a workaround that prevents sending multiple consecutive off commands to the light.

### How to reproduce it (minimal and precise)

(tested with Node Red -> Home Assistant -> Z2M setup, 1 second delay after each step, 1 s transition)
* Send off signal to the light
* Send another off signal to the light
* Turn on the light
* Second second turn on signal to the light, changing the color temperature
* -> the light suddenly flashes off at the end of the transition

### Zigbee2MQTT version

1.25.1-dev

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

jpbede, 2022-07-15:
Can also confirm this behavior with the HG08633 (RGB garden spot).

csowada, 2022-07-22:
Same here with all TS050xB variants. Also with [FUT039Z](https://www.zigbee2mqtt.io/devices/FUT039Z.html#miboxer-fut039z) . For me it is enough to double switch off the device. Afterwards I need to switch the device two times on. First time it flickers and shows OFF.

CZDanol, 2022-07-22:
Okay, this seems it's not just a mere Lidl light firmware issue. Either it's a widespread firmware issue, or there could be something in Z2M.

csowada, 2022-07-22:
I guess that all variants from Tuya uses the same firmware. My TS050[2-5]B variants where from AliExpress, cheap but buggy :-)

ianclegg, 2022-07-25:
I don't think this issue affects all TS050xB devices - I have some tuya zigbee tunable white drivers that identify as TS0502B and I have not been able to reproduce this. I've been publishing `{"state":"OFF"}` to the devices topic with various intervals and repetitions, twice, three times - followed by `{"state":"ON"}` with no flicker and the state shows ON. I've been messing with it for 10 minutes, that would seem to agree with your hunch that this is specific to certain TS050xB firmware
The Zigbee manufacturer of my device is _TZ3210_xxxxxx

closed 2022-09-01 by github-actions[bot]