Koenkk/zigbee2mqtt #13163: Gledopto GL-G-003P 7w garden light pro

Hjelen, 2022-07-15:
### Link

https://www.gledopto.com/h-col-381.html

### Database entry

{"id":51,"type":"Router","ieeeAddr":"0x84fd27fffe8aa331","nwkAddr":35268,"manufId":4687,"manufName":"GLEDOPTO","powerSource":"Unknown","modelId":"GL-G-003P","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":2,"dateCode":"20210609","swBuildId":"V_1_8_A1","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1657841278587,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to install the device in zigbee2mqtt, but it says unsupported device. Then I tried to use the code called «gledopto.js» from github (zigbee-herdsman-converters) but I couldn’t find the GL-G-003P in the list. I’m also unshure if I did itj right. If there is a recipe for how I use this code in my homeassistant zigbee2mqtt, it would been very nice. I’m pretty shure that the GL-G-002P model, thats in the file from github, is just the same light as te 003P. Tried to just change the name, but didn’t work. I’m new at this, this is the first time I buy a device that zigbee2mqtt don’t support yet. So if you could publish or send me the exact code that works for me, I would be thrilled. Thank you guys for doing all this work with zigbee!

### Supported color modes

Color temperature, RGB

### Color temperature range

Don’t know this

Koenkk, 2022-07-17:
Please add the color temperature range as described on https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device

Hjelen, 2022-07-17:
Color temperature range: 2000k-6500k

Koenkk, 2022-07-17:
I need the value in mireds, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device

Hjelen, 2022-07-17:
Read result of 'lightingColorCtrl': {"colorTempPhysicalMin":158,"colorTempPhysicalMax":495}

Koenkk, 2022-07-18:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-07-18 by Koenkk

Hjelen, 2022-07-19:
thanx! When will the device be avaliable in the official released versjon? Do I loose my allready installed devices if I install the dev branch? Will I have to deactivate or delete the official released versjon?