Koenkk/zigbee2mqtt #13167: sendZclFrameToEndpointInternal error on End device after Z2M restarts (EFR32 Coordicator)

konnextadmin, 2022-07-16:
### What happened?

**Error message**
sendZclFrameToEndpointInternal error

**Environment**
1. I got two devices in my zigbee network, 0xa4c138e04665fe75 is a 4 gang switch and acts as a router, 0xa4c138c441aaf9b8 is a 2 gang switch that can only act as an end device.
2. I got a EFR32 coordinator, plugged into ethernet and z2m connects to it via ezsp adapter and ip:port
3. There are no other zigbee devices or coordinators around

**Problem**
1. After the first time they get paired, everything's fine. But every time zigbee2mqtt is restarted, it shows "sendZclFrameToEndpointInternal error" when I try to switch on/off the 2 gang switch
2. The 4 gang switch which runs as a router never had such a problem after restart
3. While the 2 gang switch cannot be controlled from z2m, by clicking the physical button on the switch, z2m still can get its state update
4. The problem always disappears after I load the network map once which I suppose scanned network and rebuilt routes, but after a restart, the problem appears again
5. If I remove the 4 gang switch and only pair the 2 gang switch, then of course this end device is connected directly to the coordinator, the problem also disappears.
6. Problem does not exist on the Sonoff CC2652P USB dongle 

### What did you expect to happen?

I expected that restarting z2m is not causing sendZclFrameToEndpointInternal error on the end device attached to the router when I try to control them.

### How to reproduce it (minimal and precise)

Restarting z2m, go to z2m frontend, toggle the 2 gang switch, sendZclFrameToEndpointInternal error shows. Now reload the network map, map is created successfully, go back to toggle the 2 gang switch again, then no error occurs. 

### Zigbee2MQTT version

1.26.0

### Adapter firmware version

6.7.8.0 build 373

### Adapter

EFR32 coordinator

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9125337/log.txt)

kirovilya, 2022-07-16:
Hi @konnextadmin . Please record ["herdsman"-log](https://www.zigbee2mqtt.io/guide/usage/debug.html#enabling-logging) for debug. The device for some reason does not respond to the command... need to look deeper

konnextadmin, 2022-07-16:
> Hi @konnextadmin . Please record ["herdsman"-log](https://www.zigbee2mqtt.io/guide/usage/debug.html#enabling-logging) for debug. The device for some reason does not respond to the command... need to look deeper

Hi, here is the herdsman lo
[herdsman log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9125358/herdsman.log.txt)
g

kirovilya, 2022-07-16:
Thanks. I met the error "Error: Unrecognized nodeId", but I don’t know how to win yet.
I will sort this out.

kirovilya, 2022-07-16:
@konnextadmin Can you check how it will work with the modified ezspAdapter.js file?
need to replace file in zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/ezsp/adapter/
[ezspAdapter.zip](https://github.com/Koenkk/zigbee2mqtt/files/9126238/ezspAdapter.zip)

konnextadmin, 2022-07-16:
@kirovilya Hi kirovilya, I've just tested and think the issue has been resolved! Works perfect, great work!

closed 2022-07-17 by konnextadmin

bubar26, 2023-04-16:
Hi 
I have a similar issue but not sure.
can you say if I have to apply this fix ?

This is my error : 

zlinky config: standard, single_phase, true, TEMPO
zlinky config: standard, single_phase, true, TEMPO
zlinky config: standard, single_phase, true, TEMPO
Zigbee2MQTT:info  2023-04-15 23:19:09: MQTT publish: topic 'zigbee2mqtt/Lixee_Zlinky_TIC', payload '{"active_enerfy_out_d01":9116,"active_enerfy_out_d02":7464,"active_enerfy_out_d03":1122,"active_enerfy_out_d04":2637,"active_power":450,"active_power_max":3080,"active_power_ph_b":502,"apparent_power":513,"available_power":9,"average_rms_voltage_meas_period":232,"current_date":"E230415224258","current_index_tarif":1,"current_price":"HC BLEU","current_summ_delivered":20340,"current_summ_received":141,"current_tarif":"TEMPO","current_tier10_summ_delivered":0,"current_tier1_summ_delivered":4953,"current_tier2_summ_delivered":12963,"current_tier3_summ_delivered":482,"current_tier4_summ_delivered":1153,"current_tier5_summ_delivered":213,"current_tier6_summ_delivered":577,"current_tier7_summ_delivered":0,"current_tier8_summ_delivered":0,"current_tier9_summ_delivered":0,"days_number_current_calendar":0,"days_number_next_calendar":0,"days_profile_current_calendar":"00004001 06004002 16004001 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE","days_profile_next_calendar":"","drawn_v_a_max_n1":4390,"injected_active_load_n":0,"injected_active_load_n1":0,"injected_v_a":0,"injected_v_a_max_n":860,"injected_v_a_max_n1":860,"linkquality":255,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"811961211519","power_threshold":9,"reactive_power":2123,"reactive_power_ph_b":22683,"reactive_power_ph_c":16906,"relais":1,"rms_current":2,"rms_voltage":237,"site_id":"19773806022960","software_revision":2,"start_mobile_point1":0,"start_mobile_point2":0,"start_mobile_point3":0,"status_register":"013A0100","stop_mobile_point1":0,"stop_mobile_point2":0,"stop_mobile_point3":0,"total_reactive_power":8006,"update":{"installed_version":12,"latest_version":13,"state":"available"},"update_available":null}'
Zigbee2MQTT:info  2023-04-15 23:19:09: MQTT publish: topic 'zigbee2mqtt/Lixee_Zlinky_TIC', payload '{"active_enerfy_out_d01":9116,"active_enerfy_out_d02":7464,"active_enerfy_out_d03":1122,"active_enerfy_out_d04":2637,"active_power":450,"active_power_max":3080,"active_power_ph_b":502,"apparent_power":513,"available_power":9,"average_rms_voltage_meas_period":232,"current_date":"E230415224258","current_index_tarif":1,"current_price":"HC BLEU","current_summ_delivered":20340,"current_summ_received":141,"current_tarif":"TEMPO","current_tier10_summ_delivered":0,"current_tier1_summ_delivered":4953,"current_tier2_summ_delivered":12963,"current_tier3_summ_delivered":482,"current_tier4_summ_delivered":1153,"current_tier5_summ_delivered":213,"current_tier6_summ_delivered":577,"current_tier7_summ_delivered":0,"current_tier8_summ_delivered":0,"current_tier9_summ_delivered":0,"days_number_current_calendar":0,"days_number_next_calendar":0,"days_profile_current_calendar":"00004001 06004002 16004001 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE","days_profile_next_calendar":"","drawn_v_a_max_n1":4390,"injected_active_load_n":0,"injected_active_load_n1":0,"injected_v_a":0,"injected_v_a_max_n":860,"injected_v_a_max_n1":860,"linkquality":255,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"811961211519","power_threshold":9,"reactive_power":2123,"reactive_power_ph_b":22683,"reactive_power_ph_c":16906,"relais":1,"rms_current":2,"rms_voltage":237,"site_id":"19773806022960","software_revision":2,"start_mobile_point1":0,"start_mobile_point2":0,"start_mobile_point3":0,"status_register":"013A0100","stop_mobile_point1":0,"stop_mobile_point2":0,"stop_mobile_point3":0,"total_reactive_power":8006,"update":{"installed_version":12,"latest_version":13,"state":"available"},"update_available":null}'
zlinky config: standard, single_phase, true, TEMPO
zlinky config: standard, single_phase, true, TEMPO
zlinky config: standard, single_phase, true, TEMPO
Failed to read zigbee attributes: Error: Read 0x00158d000638e2da/1 liXeePrivate(["currentTarif"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)
zlinky config: standard, single_phase, true, TEMPO
Error: Reset error: Error: {"sequence":-1} after 10000ms
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:298:23
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)