Koenkk/zigbee2mqtt #13168: WS-EUK04 Aqara - can be controlled via Z2M screen but does not register events in log

fdski, 2022-07-16:
### What happened?

WS-EUK04 two button switch. I can see it via HA/Z2M screen - i can change decoupled mode to relay mode and vice versa. It updates all statuses, i know it works as i can hear it click when i change it on the computer. Change of control lights works properly as well. 

However, when i go to log and press buttons - nothing is registered. In either control relay or decoupled mode. 
No idea why that happens, i have 20 switches from Aqara and they have been doing that on and off. Usually they will drop off the network completely and i have to rejoin them, however this time after rejoining it still doesn't register presses. 

Any ideas ?

### What did you expect to happen?

log entry should be present when switch button is pressed. ( viewing in debug mode ). As no messages are present, automations don't work. 

### Zigbee2MQTT version

 1.26.0 commit: unknown

### Adapter firmware version

 20210708

### Adapter

Sonoff ZBDongle-P ZigBee Dongle Plus CC2652P

Koenkk, 2022-07-17:
Try updating your dongle to the latest firmware and repair the switch.

fdski, 2022-07-19:
that fixed it, thank you.

closed 2022-07-19 by fdski