Koenkk/zigbee2mqtt #13180: Mix of timeout / no network route / NWK_TABLE_FULL and a lot of devices indicated offline

lunarok, 2022-07-17:
### What happened?

Everything was working fine until I add a switch connected for ventilation in kitchen. My fault, I plug it the relay on shortcut.
And then, I start to see globally and more than all in the zone where it was plugged issues of communications (shutters not answering, lights also, garden disconnected ...)

So I remove it, and start to take a look with frontend to debug it.
I see a lot of NWK_TABLE_FULL errors. I take a look on google/website and see the potential issue with wifi. I check, no interference found, my wifi is on channel 11 and same for zigbee (so zigbee range is out of the wifi) No neigbhour network interfering too.

Then I upgrade to dev branch. Still same.
I shutdown house, restart all : at beginning a lot of "timeout" happens also.
I shutdown z2m, unplug key for 1h and restart : still same
I upgrade the firmware, first with the version on wiki of february, same, then may version : same

Latest was yesterday, I unplug key for 2h, power off house. Then plug key and start z2m (still dev branch) and start to power on step by step house.
Start few hours after to request actions on devices to update their status, like power off lights, move curtains ...
Still today (24h later) 14 devices indicated as offline (135 devices total, 83 on sector). Some actions are working, sometime very long, sometime not all (including on device online)

How many time is a network supposed to take to rebuild ?

### What did you expect to happen?

That all devices are seen online and no timeout/NWK/route errors

### How to reproduce it (minimal and precise)

No idea

### Zigbee2MQTT version

1.26.0-dev commit: cf8e7b1

### Adapter firmware version

20220507

### Adapter

zStack3x0 ZigStart v4

lunarok, 2022-07-17:
I forget to add debug log, and when I go to get it, I found a message repeating hundred of times per second. It's an Air quality device on battery. I remove it's battery but still logs full of publish concerning it. I'm waiting to see if in some minutes it stops spamming the logs, if yes will see if network become better.

lunarok, 2022-07-17:
[image]

Koenkk, 2022-07-17:
Can you try if this fw improves it? 
[znp_CC1352P_2_LAUNCHXL_tirtos_ccs.hex.zip](https://github.com/Koenkk/zigbee2mqtt/files/9128514/znp_CC1352P_2_LAUNCHXL_tirtos_ccs.hex.zip)

lunarok, 2022-07-17:
Thanks, I just have flash it and restart z2m
I can see still have some timeout and no network routes in logs, mainly when I ask off/on to a device.
Will let it like that this night and see if morning fix the network :)

For information, the multi sensor on battery that was publishing a lot. I remove battery and still messages appear ... So I have force excluded to stop spamming the logs but nothing else has gone better

lunarok, 2022-07-18:
So still for now 9 devices "not avalaible" but for exemple, I have Salon - Rideau Est (curtain device on diret power, no battery)
It's seen as not avalaible, but I can send command and it acts. This is what appears in logs then :
[image]

[image]

lunarok, 2022-07-19:
Thank you @Koenkk , at least if you see where the issue could be that's nice
In the meantime, I receive a USB cable just to confirm and nothing better
Really crazy because a light that is sayed online, at 22.00 I can power it on, and at 23.00 I still see it has online, but when try to power off, NWK errors again

With the number of devices, is it supposed to work with 130-150 devices (maybe will again some more). If there is better key, better I take a move maybe (and kids+wife are still far in holidays for few weeks, so best time of the year for geeky activities :))

lunarok, 2022-07-19:
I see on TI forum request to see the map, I'm not able to generate it on z2m frontend. Keep loading for 5 and then 10 mn, still nothing displayed.(I can see some message in red "cannot get LQI for XXX"
[image]

Koenkk, 2022-07-20:
Can you check with this fw?
[znp_CC1352P_2_LAUNCHXL_tirtos_ccs.hex.zip](https://github.com/Koenkk/zigbee2mqtt/files/9152560/znp_CC1352P_2_LAUNCHXL_tirtos_ccs.hex.zip)

I've changed the max direct possible neighbours of the coordinator to 50 (from 20).

lunarok, 2022-07-20:
Hi,
After 2h with this firmware, still same. Actually 39 devices offline.
I try to power off not everything but a lot to see if it has effet to reconnect, but nothing.
Still same, NWK, timeout and network errors.

Is it possible at least that I get a stable network back by excluding some devices to reduce the size of network ?
It's true that just before the issue happens I have add in the last 2 weeks maybe a dozen at least of lights (handing over all my Yeelight for zigbee)

lunarok, 2022-07-21:
Well, last firmware seems to make the things worst, I have actually 97 devices that are marked as not online ...
Not much is working now, trying to power on/off light is not successfull, same for curtains ...
I didn't keep the firmware you made just before, so will use the one from may to check if it's going better

[image]

Koenkk, 2022-07-21:
Can you try this fw? After updating it may help to toggle the devices once since the availability feature doesn't do any route discovery (and other normal command will). I'm especially interested if this fw still gives the `NWK_TABLE_FULL` errors.

[znp_CC1352P_2_LAUNCHXL_tirtos_ccs.hex.zip](https://github.com/Koenkk/zigbee2mqtt/files/9162318/znp_CC1352P_2_LAUNCHXL_tirtos_ccs.hex.zip)

lunarok, 2022-07-21:
Still NWK_TABLE_FULL. I flash firmware, start back z2m, take the first light in "not avalaible" status and try power on :

[image]

Koenkk, 2022-07-21:
This is really strange, do you maybe have a sniffer available so that we can dive deeper into your network? https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

lunarok, 2022-07-22:
I don't have, but I just ordered one. Must arrived tomorrow.

lunarok, 2022-07-25:
I see a lot of trames with "Bad FCS", do you think it could be related to coordinator and test another key could help ? I have a zzh also somewhere in boxes

Koenkk, 2022-07-25:
I need the pacpng file + nework key (you can send it to me over telegram (@koenkk)). The `Bad FCS` indicates that there is interference, so the CC2531 couldn't receive the whole message. Does it happen for every message? I recommend to sniff close to the coordinator.

lunarok, 2022-07-25:
sended

closed 2022-09-01 by github-actions[bot]