Koenkk/zigbee2mqtt #13184: Issue with ConBee II: `Error: spawn udevadm ENOENT`

SanderDelden, 2022-07-17:
### What happened?

I'm facing some issues with getting zigbee2mqtt to work on my Home Assistant installation. After starting zigbee2mqtt the process exits with the error: `Error: spawn udev adm ENOENT`

My zigbee2mqtt configuration is as follows:
```
enabled: false
master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
options: '-d -d'
log: false
mqtt:
  server: mqtt://core-mosquitto
serial:
  port: >-
    /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2481620-if00
  adapter: deconz
```

### Zigbee2MQTT version

1.26.0

### Adapter firmware version

26720700

### Adapter

ConBee II

### Debug log

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/socat.sh
[21:12:35] INFO: Socat not enabled, marking service as down
cont-init: info: /etc/cont-init.d/socat.sh exited 0
cont-init: info: running /etc/cont-init.d/zigbee2mqtt.sh
cont-init: info: /etc/cont-init.d/zigbee2mqtt.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun socat (no readiness notification)
services-up: info: copying legacy longrun zigbee2mqtt (no readiness notification)
[21:12:43] INFO: Handing over control to Zigbee2MQTT Core ...
> zigbee2mqtt@1.26.0 start
> node index.js
Zigbee2MQTT:info  2022-07-17 21:13:04: Logging to console and directory: '/config/zigbee2mqtt/log/2022-07-17.21-13-04' filename: log.txt
Zigbee2MQTT:info  2022-07-17 21:13:05: Starting Zigbee2MQTT version 1.26.0 (commit #unknown)
Zigbee2MQTT:info  2022-07-17 21:13:05: Starting zigbee-herdsman (0.14.40)
Zigbee2MQTT:error 2022-07-17 21:13:06: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-07-17 21:13:06: Failed to start zigbee
Zigbee2MQTT:error 2022-07-17 21:13:06: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-07-17 21:13:06: Exiting...
Zigbee2MQTT:error 2022-07-17 21:13:06: Error: spawn udevadm ENOENT
    at Process.ChildProcess._handle.onexit (node:internal/child_process:283:19)
    at onErrorNT (node:internal/child_process:478:16)
    at processTicksAndRejections (node:internal/process/task_queues:83:21)
```

Koenkk, 2022-07-17:
Can you provide the herdsman debug log of this?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Bazarov42, 2022-07-19:
Hy, i have same problem with ConBee II.
Is it better that I open another issue or I can put my logs here?

markwoodward86, 2022-07-20:
I also started getting this yesterday with the same hardware

`> zigbee2mqtt@1.26.0 start
> node index.js
2022-07-20T12:21:32.959Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2022-07-20T12:21:32.962Z zigbee-herdsman:deconz:driver Set delay to 0
2022-07-20T12:21:32.962Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2461343-if00","adapter":"deconz"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2022-07-20T12:21:32.963Z zigbee-herdsman:deconz:driver Opening with /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2461343-if00
Zigbee2MQTT:error 2022-07-20 09:21:32: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-07-20 09:21:32: Failed to start zigbee
Zigbee2MQTT:error 2022-07-20 09:21:32: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-07-20 09:21:32: Exiting...
Zigbee2MQTT:error 2022-07-20 09:21:33: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2461343-if00'
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:182:28)
    at SerialPort._error (/app/node_modules/@serialport/stream/lib/index.js:198:14)
    at /app/node_modules/@serialport/stream/lib/index.js:242:12`

markwoodward86, 2022-07-21:
Same error, same hardware started yesterday. 

`> zigbee2mqtt@1.26.0 start
> node index.js
2022-07-21T00:05:37.024Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2022-07-21T00:05:37.027Z zigbee-herdsman:deconz:driver Set delay to 0
2022-07-21T00:05:37.028Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2461343-if00","adapter":"deconz"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2022-07-21T00:05:37.028Z zigbee-herdsman:deconz:driver Opening with /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2461343-if00
2022-07-21T00:05:37.044Z zigbee-herdsman:deconz:driver Serialport opened
2022-07-21T00:05:37.045Z zigbee-herdsman:controller:log Started with result 'resumed'
2022-07-21T00:05:37.046Z zigbee-herdsman:controller:log Injected database: true, adapter: true
2022-07-21T00:05:37.056Z zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 1 paramId: 1
2022-07-21T00:05:40.029Z zigbee-herdsman:deconz:adapter checking coordinator simple descriptor
2022-07-21T00:05:40.048Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0x0 EP:0 SeqNr. 2 request id: 1
2022-07-21T00:05:47.060Z zigbee-herdsman:deconz:driver Timeout for request - CMD: 0xa seqNr: 1
Zigbee2MQTT:error 2022-07-20 21:05:47: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-07-20 21:05:47: Failed to start zigbee
Zigbee2MQTT:error 2022-07-20 21:05:47: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-07-20 21:05:47: Exiting...
Zigbee2MQTT:error 2022-07-20 21:05:47: undefined
2022-07-21T00:05:47.074Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 3`

Koenkk, 2022-07-21:
@markwoodward86 your error is different:

```
2022-07-21T00:05:40.048Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0x0 EP:0 SeqNr. 2 request id: 1
2022-07-21T00:05:47.060Z zigbee-herdsman:deconz:driver Timeout for request - CMD: 0xa seqNr: 1
```

Looks like the Conbee adapter cannot communicate, try replugging or reflashing the firmware

SanderDelden, 2022-07-23:
Hi @Koenkk,

Thank you for the swift reply and my apologies for the late one.. I have tried to produce some debug logging but unfortunately seem unable to do so. My Zigbee2MQTT settings look as follows:

```
enabled: false
master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
options: '-d -d'
log: true
mqtt:
  server: mqtt://core-mosquitto
serial:
  port: >-
    /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2481620-if00
  adapter: deconz
advanced:
  log_level: debug
zigbee_herdsman_debug: true
```

When checking the STDOUT logs of the container no debug logging is found:

```
# docker logs -f --tail 50 addon_45df7312_zigbee2mqtt > log.txt 2>&1

# cat log.txt
> node index.js
Zigbee2MQTT:info  2022-07-23 11:56:33: Logging to console and directory: '/config/zigbee2mqtt/log/2022-07-23.11-56-33' filename: log.txt
Zigbee2MQTT:info  2022-07-23 11:56:33: Starting Zigbee2MQTT version 1.26.0 (commit #unknown)
Zigbee2MQTT:info  2022-07-23 11:56:33: Starting zigbee-herdsman (0.14.40)
Zigbee2MQTT:error 2022-07-23 11:56:34: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-07-23 11:56:34: Failed to start zigbee
Zigbee2MQTT:error 2022-07-23 11:56:34: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-07-23 11:56:34: Exiting...
Zigbee2MQTT:error 2022-07-23 11:56:34: Error: spawn udevadm ENOENT
    at Process.ChildProcess._handle.onexit (node:internal/child_process:283:19)
    at onErrorNT (node:internal/child_process:478:16)
    at processTicksAndRejections (node:internal/process/task_queues:83:21)
[11:56:39] INFO: Handing over control to Zigbee2MQTT Core ...
> zigbee2mqtt@1.26.0 start
```

Leaving out the `log_level: debug` under `advanced` makes no difference. Am I perhaps doing something wrong?

Koenkk, 2022-07-23:
@SanderDelden nowadays the config of the HA addon is split up into 3 sections. The part under `serial:` should move to the serial section like this:

[image]

danka74, 2022-08-11:
Dear All,

have this (or similar or not...) problem with ConBee II since I upgraded my host machine to Ubuntu 22.04.1. My log below:

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/socat.sh
[21:23:29] INFO: Socat not enabled, marking service as down
cont-init: info: /etc/cont-init.d/socat.sh exited 0
cont-init: info: running /etc/cont-init.d/zigbee2mqtt.sh
cont-init: info: /etc/cont-init.d/zigbee2mqtt.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun socat (no readiness notification)
services-up: info: copying legacy longrun zigbee2mqtt (no readiness notification)
[21:23:29] INFO: Zigbee Herdsman debug logging enabled
[21:23:30] INFO: Handing over control to Zigbee2MQTT Core ...
> zigbee2mqtt@1.27.0 start
> node index.js
Zigbee2MQTT:debug 2022-08-11 21:23:31: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2022-08-11 21:23:31: Logging to console and directory: '/config/zigbee2mqtt/log/2022-08-11.21-23-31' filename: log.txt
Zigbee2MQTT:debug 2022-08-11 21:23:31: Removing old log directory '/config/zigbee2mqtt/log/2022-08-11.21-17-23'
Zigbee2MQTT:info  2022-08-11 21:23:31: Starting Zigbee2MQTT version 1.27.0 (commit #unknown)
Zigbee2MQTT:info  2022-08-11 21:23:31: Starting zigbee-herdsman (0.14.46)
Zigbee2MQTT:debug 2022-08-11 21:23:31: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[25],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"deconz","path":"/dev/ttyACM0"}}'
2022-08-11T19:23:32.050Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2022-08-11T19:23:32.051Z zigbee-herdsman:deconz:driver Set delay to 0
2022-08-11T19:23:32.052Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[25]},"serialPort":{"path":"/dev/ttyACM0","adapter":"deconz"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2022-08-11T19:23:32.053Z zigbee-herdsman:deconz:driver Opening with /dev/ttyACM0
2022-08-11T19:23:32.060Z zigbee-herdsman:deconz:driver Serialport opened
2022-08-11T19:23:32.061Z zigbee-herdsman:controller:log Started with result 'resumed'
2022-08-11T19:23:32.061Z zigbee-herdsman:controller:log Injected database: true, adapter: true
2022-08-11T19:23:32.067Z zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 1 paramId: 1
2022-08-11T19:23:35.053Z zigbee-herdsman:deconz:adapter checking coordinator simple descriptor
2022-08-11T19:23:35.056Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0x0 EP:0 SeqNr. 2 request id: 1
2022-08-11T19:23:42.071Z zigbee-herdsman:deconz:driver Timeout for request - CMD: 0xa seqNr: 1
Zigbee2MQTT:error 2022-08-11 21:23:42: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-08-11 21:23:42: Failed to start zigbee
Zigbee2MQTT:error 2022-08-11 21:23:42: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-08-11 21:23:42: Exiting...
Zigbee2MQTT:error 2022-08-11 21:23:42: undefined
2022-08-11T19:23:42.084Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 3
```

closed 2022-09-20 by github-actions[bot]