Koenkk/zigbee2mqtt #13189: Xiaomi WS-USC03 on latest firmware doesn't pass action

jusxon, 2022-07-18:
### What happened?

I am trying to implement the below external converter file for the WS-USC03 (copied from new WS-USC01). 

```
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const ota = require('zigbee-herdsman-converters/lib/ota');
const constants = require('zigbee-herdsman-converters/lib/constants');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const globalStore = require('zigbee-herdsman-converters/lib/store');

const preventReset = async (type, data, device) => {
    if (
        // options.allow_reset ||
        type !== 'message' ||
        data.type !== 'attributeReport' ||
        data.cluster !== 'genBasic' ||
        !data.data[0xfff0] ||
        // eg: [0xaa, 0x10, 0x05, 0x41, 0x87, 0x01, 0x01, 0x10, 0x00]
        !data.data[0xFFF0].slice(0, 5).equals(Buffer.from([0xaa, 0x10, 0x05, 0x41, 0x87]))
    ) {
        return;
    }
    const options = {manufacturerCode: 0x115f};
    const payload = {[0xfff0]: {
        value: [0xaa, 0x10, 0x05, 0x41, 0x47, 0x01, 0x01, 0x10, 0x01],
        type: 0x41,
    }};
    await device.getEndpoint(1).write('genBasic', payload, options);
};

const definition = {
    zigbeeModel: ['lumi.switch.b1naus01'],
    model: 'WS-USC03',
    vendor: 'Xiaomi',
    description: 'Aqara smart wall switch (with neutral, single rocker)',
    fromZigbee: [fz.on_off, fz.xiaomi_multistate_action, fz.aqara_opple],
    toZigbee: [tz.on_off, tz.xiaomi_switch_operation_mode_opple,
        tz.xiaomi_flip_indicator_light, tz.aqara_switch_mode_switch],
    exposes: [e.switch(), e.action(['single', 'double']), e.flip_indicator_light(),
        exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled']).withDescription('Decoupled mode')],
    onEvent: preventReset,
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint1 = device.getEndpoint(1);
        // set "event" mode
        await endpoint1.write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f, disableResponse: true});
    },
};

module.exports = definition;
```

### What did you expect to happen?

I have switches on firmware 07-16-2020 that work perfectly with the external converter, but two switches on firmware 12-28-2021 do not send action events.

### Zigbee2MQTT version

1.26.0-1

### Adapter firmware version

20210218

### Adapter

SMARTLIGHT CC2652P

Koenkk, 2022-07-18:
Try updating the firmware of your smartlight adapter to the latest version and repair the device.

jusxon, 2022-07-20:
Thank you, this worked!

Also, forgot I had changed to a "ZigStar Stick v4" awhile back, but either way updating the coordinating firmware and then re-pairing the switches did the trick.