Koenkk/zigbee2mqtt #13190: Device incorrectly identified

bdnstn, 2022-07-18:
### What happened?

Double Zigbee Power Point model MG-AUZG01 is incorrectly identified as TS011F_plug_1

### What did you expect to happen?

I expected double wall mounted power point model MG-AUZG01 to be added to Zigbee2MQTT. Note, this is an Australian double GPO. Manufacturer is Makegood.

### How to reproduce it (minimal and precise)

Join device to Zigbee network in Zigbee2MQTT.
Front view of device with cover removed to expose PCB.

[image]

Rear view showing model numbers etc.

[image]

### Zigbee2MQTT version

 1.26.0-dev commit: 4ef9ea7

### Adapter firmware version

20220219

### Adapter

usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9129226/log.txt)

Koenkk, 2022-07-18:
Can you provide the data/database.db entry of this device?

bdnstn, 2022-07-19:
Hi Koen,

I guess this is what you are asking for.

{"id":6,"type":"Router","ieeeAddr":"0xa4c138d86d477cb7","nwkAddr":42142,"manufId":4417,"manufName":"_TZ3000_dd8wwzcy","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,6,1794,2820,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�\u0000\u0000\u0000\u0012","65506":54,"65508":0,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_dd8wwzcy","powerSource":1,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":235,"rmsCurrent":0,"activePower":0}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,189]}},"genOnOff":{"attributes":{"onOff":1,"offWaitTime":0,"onTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c83985","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c83985","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c83985","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"2":{"profId":260,"epId":2,"devId":81,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1,"offWaitTime":0,"onTime":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1798312590},"lastSeen":1658237946226,"defaultSendRequestWhen":"immediate"}

Thanks,
Brian

Koenkk, 2022-07-19:
@looknw can you confirm that the `_TZ3000_dd8wwzcy` is not a normal plug but the device shown above?

Koenkk, 2022-07-19:
@looknw
- does it have 1 or 2 switches?
- can you provide the data/database.db entry of this device?

looknw, 2022-07-19:
2 switches , and havent installed it yet

Koenkk, 2022-07-19:
@bdnstn 
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_dd8wwzcy'}],
    model: 'MG-AUZG01',
    vendor: 'Makegood',
    description: 'Double Zigbee power point',
    extend: extend.switch(),
    exposes: [e.switch().withEndpoint('l1'), e.switch().withEndpoint('l2')],
    endpoint: (device) => {
        return {'l1': 1, 'l2': 2};
    },
    meta: {multiEndpoint: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

bdnstn, 2022-07-20:
Hi,
Sorry for the delay replying, I'm in Perth Western Australia GMT+8. 

Its a partial fix.

1. I now have two on/off switches.
2. The manufacturer is correctly displayed as "MakeGood".
3. Control - Child lock - `This entity ('lock.coffee_station_power_child_lock') does not have a unique ID`.
4. Control - Power indicator - `This entity ('select.coffee_station_power_indicator_mode') does not have a unique ID`.
5. Missing Sensors - Energy and Indicator mode.
6. Missing Configurations - Power Outage Memory (x2)
7. Missing Diagnostic - Power and Update.

In total, the device **_had_** 7 entities, screenshot attached - `before update.png`.
I've also attached the diagnostics after the update and before the update (actually taken after rolling back the update).
Thanks
[Befo
[image]

[After - mqtt-e3812a9c09c8b02b66e4998117fd2ccc-New Coffee Power-b19de4e6e00ca4bcb30c0f6ca658fa0a.json.txt](https://github.com/Koenkk/zigbee2mqtt/files/9145556/After.-.mqtt-e3812a9c09c8b02b66e4998117fd2ccc-New.Coffee.Power-b19de4e6e00ca4bcb30c0f6ca658fa0a.json.txt)
re - mqtt-e3812a9c09c8b02b66e4998117fd2ccc-New Coffee Power-b19de4e6e00ca4bcb30c0f6ca658fa0a.json.txt](https://github.com/Koenkk/zigbee2mqtt/files/9145554/Before.-.mqtt-e3812a9c09c8b02b66e4998117fd2ccc-New.Coffee.Power-b19de4e6e00ca4bcb30c0f6ca658fa0a.json.txt)
.

Koenkk, 2022-07-20:
Can you check with:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_dd8wwzcy'}],
    model: 'MG-AUZG01',
    vendor: 'Makegood',
    description: 'Double Zigbee power point',
    fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report, fz.tuya_switch_power_outage_memory,
        fz.ts011f_plug_indicator_mode, fz.ts011f_plug_child_mode],
    toZigbee: [tz.on_off, tz.tuya_switch_power_outage_memory, tz.ts011f_plug_indicator_mode, tz.ts011f_plug_child_mode],
    exposes: [e.switch().withEndpoint('l1'), e.switch().withEndpoint('l2'), e.power(), e.current(), e.voltage().withAccess(ea.STATE),
        e.energy(), exposes.enum('power_outage_memory', ea.ALL, ['on', 'off', 'restore'])
            .withDescription('Recover state after power outage'),
        exposes.enum('indicator_mode', ea.ALL, ['off', 'off/on', 'on/off', 'on'])
            .withDescription('Plug LED indicator mode'), e.child_lock()],

    endpoint: (device) => {
        return {'l1': 1, 'l2': 2};
    },
    meta: {multiEndpoint: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        await endpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
        await reporting.rmsVoltage(endpoint, {change: 5});
        await reporting.rmsCurrent(endpoint, {change: 50});
        await reporting.activePower(endpoint, {change: 10});
        await reporting.currentSummDelivered(endpoint);
        endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
        endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
        device.save();
    },
};

module.exports = definition;

```

bdnstn, 2022-07-21:
Hi Koen,

That is looking much better, though not perfect yet 😊

Here are my observations from the z2m devices/exposes tab:

1. `Power` is 0 w when I would expect > 0.
2. `Current` is also 0 when >0 is expected.
3. `Voltage` is also 0 v when > 0 is expected.
4. `Energy` looks ok. It is reading 0.23 kWh
5. `Power outage memory` shows `on` and can't be changed.
6. `Indicator mode` can't be changed.
7. `Child lock` can't be changed
8. `Linkquality` looks OK = 76 lqi
9. `Endpoint l1` and `l2` look OK and can be set as desired.

It raises the question of how to know what functions the device is capable of exposing? If I pair to with the TuYa app and hub the device joins OK, with the correct picture and exposes:

1. Two on/off power points.
2. Current (ma)
3. Power (W)
4. Voltage (V)
5. In total (KW.h)

So, I'm guessing that the `child lock`, `indicator mode` and `power outage memory` possibly aren't supported. Is there a better way of discovering what the device is capable of? The supply has not been forthcoming.

Debug log attached.

Thanks,
Brian

[z2mDebugLog.txt](https://github.com/Koenkk/zigbee2mqtt/files/9155572/z2mDebugLog.txt)

Koenkk, 2022-07-21:
Can you try with:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_dd8wwzcy'}],
    model: 'MG-AUZG01',
    vendor: 'Makegood',
    description: 'Double Zigbee power point',
    fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report],
    toZigbee: [tz.on_off],
    exposes: [e.switch().withEndpoint('l1'), e.switch().withEndpoint('l2'), e.power(), e.current(), e.voltage().withAccess(ea.STATE),
        e.energy()],

    endpoint: (device) => {
        return {'l1': 1, 'l2': 2};
    },
    meta: {multiEndpoint: true},
    onEvent: tuya.onEventMeasurementPoll,
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        await endpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
        await reporting.rmsVoltage(endpoint, {change: 5});
        await reporting.rmsCurrent(endpoint, {change: 50});
        await reporting.activePower(endpoint, {change: 10});
        await reporting.currentSummDelivered(endpoint);
        endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
        endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
        device.save();
    },
};

module.exports = definition;

```

bdnstn, 2022-07-22:
Working:

1. Current
2. Voltage
3. energy
4. linkquality
5. endpoint state l1
6. endpoint state l2

Not working:

1. power
2. power outage memory
3. indicator mode
4. child lock

Thanks.

Koenkk, 2022-07-23:
- power outage memory -> indicator mode -> child lock: i guess those are not supported since they are also not supported when paired to the TuYa hub
- can you check if the power works when paired with the TuYa hub?

bdnstn, 2022-07-23:
Good question! Yes it does.

[image]

Koenkk, 2022-07-23:
I updated the converter from https://github.com/Koenkk/zigbee2mqtt/issues/13190#issuecomment-1191911273 , can you try again?

bdnstn, 2022-07-23:
Power remains at zero. Interestingly, Current remains at 0.03 A regardless of whether there is a load or not.

[image]

Koenkk, 2022-07-23:
Could you provide me the debug log when connecting a load to the plug (in which case it should report power)

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

bdnstn, 2022-07-23:
I've attached the debug log as requested.
[tail of log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9174848/tail.of.log.txt)

As an experiment, I paired one of these Double GPOs with the iOS Tuya app. To my surprise, it reports 0 for current and power. I've attached the diagnostic log from HA for the Tuya integration. The device is called `"Zigbee Double Power Point"`.

So, with iOS apps, the `Smart Life` app reports current and power, but the `Tuya Smart` app does not. 

[config_entry-tuya-1592afa30f35ea761bd93208348f7bef.json.txt](https://github.com/Koenkk/zigbee2mqtt/files/9174849/config_entry-tuya-1592afa30f35ea761bd93208348f7bef.json.txt)

Koenkk, 2022-07-24:
The `tail of log.txt` file is empty.

bdnstn, 2022-07-25:
Oh, sorry. Here is another one. This log starts just before the device `Double GPO test unit in study` joins. I have a kettle attached, which boils during the period of the log.
[tail of log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9177188/tail.of.log.txt)
Interestingly, this time, different to past times, I'm getting `Null`s for power, current and voltage.
[image]
This is an identical device, called `New Coffee Power`, which joined the network a week ago with a different version of `ext_converter.yaml`, and runs my expresso coffee machine:
[image]

I have a question. I'd like to go back to my Alibaba supplier and ask them for information, but what would be the best question to ask? Can you tell me exactly what information would be most useful, please?

Koenkk, 2022-07-25:
Looks that power is published correctly but is incorrectly exposed:

```
debug 2022-07-25 09:14:17: Received Zigbee message from 'Double GPO test unit in study', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":1436}' from endpoint 1 with groupID 0
info  2022-07-25 09:14:17: MQTT publish: topic 'zigbee2mqtt/Double GPO test unit in study', payload '{"current":null,"current_l1":9.37,"energy":0.27,"linkquality":14,"power":null,"power_l1":1436,"state_l1":"ON","state_l2":"OFF","voltage":null,"voltage_l1":233}'
```

Before fixing that I would like to know if the device can provide power measurements for both channels separately, can you do the same but now plug the load to l2 instead of l1/

bdnstn, 2022-07-25:
Same test as above, but with load plugged into left and side outlet.

[Log l2 load.txt](https://github.com/Koenkk/zigbee2mqtt/files/9185422/Log.l2.load.txt)

bdnstn, 2022-07-26:
It is strange, but following the test earlier this morning (above) I paired the device with the default Smart Life app, which as I posted 3 days ago, was reporting both current and power. This morning it is reporting zero for both power and current for either power outlet. It is as if the device has got itself into a dysfunctional state. I'm going to contact the supplier to ask whether there is a way to reset the device.

bdnstn, 2022-08-05:
I've reached out to the supplier several times, but they are not responding with anything useful. Based on what I see in the default "TuYa Smart Life" app, when paired using a TuYa hub, I think the device only reports `Current (mA)`, `Power (W)`, `Voltage (V)` and `In total (KW.h)` for both outlets together, because that is all that the TuYa app reports.

Koenkk, 2022-08-05:
I see, please check with:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    on_off: {
        cluster: 'genOnOff',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.hasOwnProperty('onOff')) {
                const endpointName = definition.hasOwnProperty('endpoint') ?
                    getKey(definition.endpoint(meta.device), msg.endpoint.ID) : msg.endpoint.ID;
                return {[`state_${endpointName}`]: msg.data['onOff'] === 1 ? 'ON' : 'OFF'};
            }
        },
    }
};

const definition = {
    fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_dd8wwzcy'}],
    model: 'MG-AUZG01',
    vendor: 'Makegood',
    description: 'Double Zigbee power point',
    fromZigbee: [fzLocal.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report],
    toZigbee: [tz.on_off],
    exposes: [e.switch().withEndpoint('l1'), e.switch().withEndpoint('l2'), e.power(), e.current(), e.voltage().withAccess(ea.STATE),
        e.energy()],

    endpoint: (device) => {
        return {'l1': 1, 'l2': 2};
    },
    onEvent: tuya.onEventMeasurementPoll,
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
            await endpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        await endpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
        await reporting.rmsVoltage(endpoint, {change: 5});
        await reporting.rmsCurrent(endpoint, {change: 50});
        await reporting.activePower(endpoint, {change: 10});
        await reporting.currentSummDelivered(endpoint);
        endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
        endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
        device.save();
    },
};

module.exports = definition;

```

bdnstn, 2022-08-06:
Seems to be working perfectly now, with that converter, though I see `Exception while calling fromZigbee converter: getKey is not defined}` when I operate a switch from the zigbee2mqtt dashboard. I'll attach a copy of the debug log.
[z2m error.txt](https://github.com/Koenkk/zigbee2mqtt/files/9274454/z2m.error.txt)

Koenkk, 2022-08-06:
I updated https://github.com/Koenkk/zigbee2mqtt/issues/13190#issuecomment-1206551357

bdnstn, 2022-08-06:
Very good. That fixes the exception! 

1. Will this be included in the next release?
2. Should I get a photo of the device for you? 
3. Can we close the issue now? 
4. Can you tell me why the device was misidentified in the first place?
Thanks for your help!

Koenkk, 2022-08-06:
1. Yes
1. A link (e.g. aliexpress) would be enough 
1. almost :)
1. Because in https://github.com/Koenkk/zigbee2mqtt/issues/7405#issuecomment-1148777470 I was told the `_TZ3000_dd8wwzcy` was a normal plug

Can you do one final check for me and see if power measurements work with (make sure to attach a load to see them updating)

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const utils = require('zigbee-herdsman-converters/lib/utils');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    // MG-AUZG01 requires multiEndpoint only for on_off
    // https://github.com/Koenkk/zigbee2mqtt/issues/13190
    MGAUZG01_on_off: {
        cluster: 'genOnOff',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.hasOwnProperty('onOff')) {
                const endpointName = utils.getKey(model.endpoint(meta.device), msg.endpoint.ID);
                return {[`state_${endpointName}`]: msg.data['onOff'] === 1 ? 'ON' : 'OFF'};
            }
        },
    },
};

const definition = {
    fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_dd8wwzcy'}],
    model: 'MG-AUZG01',
    vendor: 'Makegood',
    description: 'Double Zigbee power point',
    fromZigbee: [fzLocal.MGAUZG01_on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report],
    toZigbee: [tz.on_off],
    exposes: [e.switch().withEndpoint('l1'), e.switch().withEndpoint('l2'), e.power(), e.current(), e.voltage().withAccess(ea.STATE),
        e.energy()],
    endpoint: (device) => {
        return {'l1': 1, 'l2': 2};
    },
    // onEvent: tuya.onEventMeasurementPoll,
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await endpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        await reporting.rmsVoltage(endpoint, {change: 5});
        await reporting.rmsCurrent(endpoint, {change: 50});
        await reporting.activePower(endpoint, {change: 10});
        await reporting.currentSummDelivered(endpoint);
        endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
        endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
        device.save();
    },
};

module.exports = definition;

```

bdnstn, 2022-08-06:
Link to device picture: https://www.alibaba.com/product-detail/AU-Standard-Zigbee-Power-Point-Double_1600481579248.html?spm=a2700.galleryofferlist.normal_offer.d_image.51553add1Sr78j
I attached a jpg also just in case the like requires my account auth:
[image]

Final check done with load, using the converter supplied in your last comment. Confirmed Power, Current, Voltage and Energy all appear correct and update as expected.

.

Koenkk, 2022-08-07:
Great, added the device

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-08-07 by Koenkk

brywithawhy, 2022-10-30:
I have this same socket in Australia however I cannot get it to work completely. Mine reports as _TZ3000_raviyuvk which I changed the external converter to suit. On/Off works however it does not show power data.

I get this error when it tries to configure itself

`Failed to configure 'Bathroom GPO', attempt 2 (Error: ConfigureReporting 0x000d6ffffe9d363c/1 haElectricalMeasurement([{"attribute":"rmsVoltage","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":5}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 5635 - 1 - 6 - 2820 - 7 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:559:17) at processTimers (node:internal/timers:502:7))`

Let me know if there's anything else I can supply to get this working

loryanstrant, 2023-02-18:
Question, where you getting power reporting for both outlets, or just one?

bdnstn, 2023-02-19:
As best I can remember, the device showed up as a single power point instead of a double so I think it was only reporting power for one point.

aynema, 2024-01-02:
Hi there, it seems that when I update Z2M HA Addon from 1.34.0-1 to 1.35.0-1 these TZ3000_dd8wwzcy units lose the power monitoring results. after restoring to a previous update, they are fine again but I'll double check after updating the add-on again to see if it happens again. Has anyone else noticed this?

Jimbobeb66, 2024-03-02:
Mine are all having strange behaviour now also. But only when i try to set the Power outage memory or Indicator mode, On/Off and power monitoring and reporting is working fine though. This is happening to all the Powerpoints of this type i have.

Publish 'set' 'power_outage_memory' to 'PP Garage front' failed: 'Error: Write 0xa4c1389e24ec57e7/1 genOnOff({"moesStartUpOnOff":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

and

Publish 'set' 'indicator_mode' to 'PP Garage front' failed: 'Error: Write 0xa4c1389e24ec57e7/1 genOnOff({"tuyaBacklightMode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'