Koenkk/zigbee2mqtt #13191: [New device support]: Tuya TS011F_plug_3 _TZ3000_okaz9tjs

chinyaolin, 2022-07-18:
### Link

https://item.taobao.com/item.htm?spm=a1z09.2.0.0.50842e8dok54vc&id=670669324162&_u=i2rqr52c91aa

### Database entry

pair OK, and External converter is shown below

### Comments

IT works on my External converter below.
Everything fine, but no **energy** and **ota**

AND here is the Device ICON
`data:image/jpeg;base64,/9j/4AAQSkZJRgABAQAAAQABAAD/2wBDAAICAgICAQICAgIDAgIDAwYEAwMDAwcFBQQGCAcJCAgHCAgJCg0LCQoMCggICw8LDA0ODg8OCQsQERAOEQ0ODg7/2wBDAQIDAwMDAwcEBAcOCQgJDg4ODg4ODg4ODg4ODg4ODg4ODg4ODg4ODg4ODg4ODg4ODg4ODg4ODg4ODg4ODg4ODg7/wAARCAGQAZADAREAAhEBAxEB/8QAHgABAAEFAQEBAQAAAAAAAAAAAAQCAwUGBwEICQr/xABHEAACAQMCAwUFBgUBBQUJAAAAAQIDBBEFIQYSMQcTQVFhInGBkaEIFDKxwdEVI0JSYjMJJEPh8BhykqKyU1Rkc4KkwtLx/8QAHAEBAQEBAQEBAQEAAAAAAAAAAAECAwcGBQgE/8QALREBAAEEAQQBAgUEAwAAAAAAAAECESExAwQFEkEGUWETMnGRoQciQlJigbH/2gAMAwEAAhEDEQA/AP38AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAWalxRpfjmk/Lx+QEKepQw+7pv0c3hARZ6jN/8SMPSMf3AtO9fV16jfpLYCv79lNKvNfEC3KpWm06eo1qT9OVr5NEsLE6usxbdK+p115SoxT+gyLD1fVqfs1I0c/5UXv8AFSKKlruoJtyt6Evc5L9wJENfr7KdlF+bjWf7AXf4++beyljzVVASIa5bN+3RrU/VxTX0ZBOpahZ1sKFeOX0Unyv6lEwAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAALNe4pW9NSqyUU+i8X7kBgLrV5NtRfcw8k/afx/YZGHneycmobN+Ke7AiVbvkf82rGGfBvf5ARnqFLO0pS90f3LE2FP31PG018UQVRuM9HJFtgXo3M0sqTfv2IJEL6cOuUBPp3tOouWolJPzArdvTmuejLH+LZBbVHO3RrqmFe9z/1gWRUqPp9Bse9zthr5oCVRlWopd1VlTX9ucx+Q0MnQv8AbFxFQ/zhvH4rqhcZNSUoqUWmmspplHoAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAGOvr+naU+VYlWayovovV+gGnV7+pWuJNydSb25v0XkgMVcXdK3b72XPV8KcXv8fI0MRV1C4rbKXcw/th+4sLNN5ljDcn8xgZOlYXtWKcLWo4+bjj8yCbHStQwn93f/AIkQXPuN3Be3QnhemS3BRnF4knF+o2L0YppJoWwPe4z7UHyyGBXTuKtGpiax6kGVp14VUsvlmuhLQJdPEpOLwpL6i4vd2iZHvIkUVKAsHJ129wsLtLnpS5qbSi+sH0f7MoylOpGrDK2fin1QFwAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAIF/e07K1c5Nd4/wAMX+fuA0GvXq3d1KWW8vMpNl0MPd6iqalRtJZfSVb9F+42MVRo1rm6jRo05Vq0ntGKy2a0NysOFPZjU1Crh9e5pP8AOX7GbjZqVlZ2kVG2t4Ul5pZb+PUgqk2wKW0gI05Zk0BZkk44aTXqgIdSlBP2VygUxWG0BddONSHLJZQEScJ288rLhnaRcDI29wqkFFv2l0ZBlqM41FiW014ea8yKkcqx0CChldBmw95Vnp8Bse7+RQjKUJqS/F+YGRp1FUjlbNdUBWAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAWa9aFvazq1HiMVn3+gHPr66q3+ot52zsl0S8jUYGvaherErS2eKa2qTX9T8l6E3Is6Zpdzqd53dFclOP+pVkvZgv1foXQ6bp+m2mm2vd29P2n+OpLeU/e/0Mia3swI02s9egFlyXLt4AWJSz7veBaeEmBalLHQCO8sChvH7AVUqy5uSe2+zAmOEZRw0mn4MDFVaU7S5UovNN9H5ehfsMtQrOcIzi8TXQKzdGoqtBSWz6NeTMxZF0oPxYHjWUBRjPgBXFuElKPXxWeoE+E1OCkujAqAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAahrt93lb7tSl7Mevq/8ArYuBqOoXP3Oz7im8XFVe01/TH92PQxmmadW1PU429L2YpZqTxtCPn+yA6ta2lvY2FO2toclOK97b8W/Ugvt56gWpzwsZwBEk92BYcssCzJ7bgWpSAtNvIFD6+nqBbkwI0mstgS7O6/mqjVlu/wADf5AZOrTjUpOM1lNYAxVNztruVKe+N0/NAZqhV7utGon/AC5bTXl6kGXKBB4+hRTjcD1LcC7SqclbD/BJ4fowJoAAAAAAAAAAAAAAAAAAAAAAAAAAAAEO/uVa6bOplc72hnzA59Oooxq3dZ5hFc2/iX0NRnOrdX8qkk51qk9orxb2SQHVdH02Gl6PCls7iftVpecvL3LoQZN5z5gUSaUWBFm8yfXYCNKWX6AWZS222AsyluBbk+m4Ft5b/ICiTxsBZm+qzuUR5P2cLqQRZyecrZgbFpt2rq35JvNaG0vVeDAkXluqlvzwX8yG/vXigLdpNVKHK+jQkZm1qudu4SWZweOvVeZnImJoWFDay8dDQAUt+AFL3T8n1JInUJudH2vxReJfuUXgAAAAAAAAAAAAAAAAAAAAAAAAAAAalr1w53kbeO6gt/e/+WPmWBpms1uS3o2kXu/bn+n7kEvhPT+/1SpfVY5p0Hinnxm1+i/MDob6oChvC/63AjTn1+oEWUuoFiTAsuXh8gLecMCjIFLYFqUt/QCw3vuBYm9vN+AESct8AU2t1K01GnWW6T9qKfWPiBvqcZ0oyi+aEllSXigMRJO11OUVnkl7UQMlRqct7CWcRmuV+8kjKOS8GUeZAZ8uoHje5A3XiUXKUuW8in0msfHqv1AyAAAAAAAAAAAAAAAAAAAAAAAAAAAeSkowcm8JLLYHPak3calOrJ7yk5N+Wf8AkMRFhpV7XdxqlernKcmo+iWyA6ro9mrDhy1t8YqcvPU/7z3f7fADISeyYEectgIs5bARpSy14LzAsOXluBactugFDec+AHje3qBj7rUrC0rxpXV5SoVGsqM54bRbC3Rv7S7lKNrdU67isyUJZwNC5LyyQRpy3KI03sQRJvdsDbuH7vv9JlQk8zoSx/8AS+n6oLMMhqEM28KqW8Jb+5j0i3F89llP2o7ok6GZhNTowmvGOSXFx9clvApbWPMopzl9SDzmS8MgUyk0sr8S3XvQGZhNTpRnH8MkmiioAAAAAAAAAAAAAAAAAAAAAAAAAY/VKjp6FctdZR5V8Xj9QNErz7nS7u46NQfL8dkWdjWdJtlecS2du1mMqq5vct3+RB19vPoBZnLLwBHm/ZfoBDlLM8gcd7XeI9c4b4ShqWh3DpVaNKU5023y1FzJb432LCuL/Z47bOL+0vtq434b4js7WhZ6RpNK6tq1CrUlKpKdx3bUlLZbb7Gb5s6TTEU3h9dN7srkpeVvsgLbe7A+de3qlUlwo7m3qTo16NODjOnNxljLysouYbizlf2S9V1m/wCNe1S31TVbzULa2hYfdad1XlUVHmlcc3Lnpnljn3IzeZqmGq4piIs+zpvEceJr25I8n4EEWb3YESo9nvsBkeH7jueI4028RrRcH7+q/wCvUDfqsFUt5RfSUWgMdZZdJxfuAm2s8WvK3+FtGIgSXL1KKHJJjIwk+JuH6evT0uprVjDUYNKdvK5ipxb8Hvs/TqS8LaWZ5n4vHvZraPebDX0JoZOxnzWPL4wm4fJ7fTBYEwoAAAAAAAAAAAAAAAAAAAAAAAAGD16fLpVKP91VfRNkGjatJR4c5ejnUiv1NC3wlSU+JalXH+lQb+LaX7kHRW8RewEWbzJ+XiBYqyShjOGBBlLdoDi3bJBVuz+6py3TtJL6hqHzT9lC0jQ+0t2mVEvxaBbLp/8AFsx/nLrV+SH3m3jw2NuC3KQFty2e2xdbHC+2hc/B94uv8qP6iW4cf+ynRVHjTtVeOVSWn9PfcHONy6cn5YfYk5eOx09WcEeb3bII1R4+QEKo2BRQrOhf0Ky606kZbe8Drq3e3QDGU4qnqNWHRKYFVPEbu4h0WU0ZvkSW35jAgahOs9O7i2qOnc15qlTmusObrL4JN+8z+gjPTdPtNMVjC2p/dUt6c4KSl5uWerfiWy3Y+xqQ0u9o2dNv+GVp8lKm5Nq2m+kY535JdMdIyxjZ4SMEtlchdGU06WaVaP8Akn81/wAiwrJGkAAAAAAAAAAAAAAAAAAAAAAAADXOIXiFovDMn9F+4Vo+uPGmWsfOo3j4FwibwekquoTfXlhH6tkG5SntgCNOWJPfcCLUl7LfUCG5bAcg7Wt+DbiL/wDdZfmGofPH2W+X/tA9pTju1otqnt53M/2M/wCUulX5Ifbsnt+hpxWnJ9DWhS3jqZHEO2HL4Tu150o/qG4cq+y8scT9qDj/AH2Cf/3BI3Lpyah9ayfXDx8TTgjze37EESb2z4+8ohzeZdMCRGm/YfmQdgsZOro9rV8ZUYv6ICJVxHWJ+TSf0Aok2tTqesEYmBd59thZZWo+3q9snu4xnNfJR/8AyIPz07Yu3Ltw7Q/tPa92S/Zu0ifccM1e54h1+EKeXXj+OHe1WqdGnF5gnvOpKMseyt/ZuD4z2/tPZeHunearTzxeij/jOptGZmdz6piY9vu+w8XYKaJ6judV49Rmf4jMz/EOm9hvaPx5xLU4k7N+1myjYdoOjUueNdUo0pXVLZc0oxbjzxbhJTh7MoyT2aZ491PL09XV108P5dx+j6j5n8b7R0PS9P3fsnJ59NzYmL38avtM5tNpxVmmqJjUw+uLW6+96Ta3T2dajGbXk2llfPJzvHp44z2lS9quvSO3zLAzBpAAAAAAAAAAAAAAAAAAAAAAAAA1riFPNo/6cTz/AOUntWja7vZWeP7pfkVGQ4SeKV/74bfBgbZJrGQIs3mWfECJVlnYCO3hZA0Hjfh2+4k0StZ2U6VGc7dwU60mo5z6JssWWJcm7FOyDiTs37UeNtc1zVdO1C21iyt6FtTsu85qTp1ZzfNzxW2JJLAtF7rMxMWfRmXll0ypb3MilsDiHbDKMOD76UpKMVTj1fvDcacm+yzXhW1/tUlCSmo3Fgm08pbXBmNy1XqH1vJ7eZpyRpy9nBRGqPZARG9/MCNNvPXBB17RnnhXT2+jt4/kBausrWVj+xZAi1JtanN+PIjE7Fan7PghcYTVdYlo6udQVpW1B2+nVqyt6CzOpicMpJe/PToj/X0nBHU9TRw+UU+U2vOofn9d1NfR9Hyc9FE1zTF/GNy4vb9q9KrxdWpaN2aa5QutRrqd/WteHK3LXmo8veVK3LCOUklzSbaR931nx+vj6P8AE5esoqjjj+2meSm8Ru1NN6p36i0PyOy9w6zufVRRy9P+DRN71VTiP4j/AMy5jr97Xt/9sZ2dadYOPd3PA9zW1SMevLFVlBy/8Mce5HnUdPeiee32f1B0FFFf9Jet5OXdPUURR/34TNv3l9eaS2uFrBZ/4W3u5mZeDto0Zt3Fws9Ix/NmoGwGkAAAAAAAAAAAAAAAAAAAAAAAADXuII5tbaX+bX0/5D3A0XW1nSraXiqmPoUXuFZ4ur6D6ypxfyZBts57sCLUkl4gRJPfqBHlPbbqWIFtyb8fkW8CnLbJceEFDefcXAol4AQ6tC2rKXf29KupLDVWkpp/B5JOxZpWlnayqfdLK3tHPHP3FCNPmx0zypZ8QPZMCPJ536ARKkt2h7FiT9kCHVeI+pR2XS4OHDWnwz0t4f8ApRBGud9ba8oJAY2pNfxKpjwikYnaved4/ciILqunxvo1TOFOlc0srz5YyX/pY1LXp8A9sfa79qrsg+1VxmtN4Ur9pPZ7rd3Crw3QnpNW4trSl3UY93Cpb4qU6ikpc8KmVJvmWzP6b7D8e/p/8k+NcH4vPHTdVxRMck+cRVVN5zNNf9tVMxbxmm0xqdPpuz9u6Prq5jqObwj9Yi37tw+zrwF2lXHaRxf299tNt/DuOdfs42elaN3ag9OtI4eHDL7vPLCEYNuSXM5byPJvlHN2bhro7f2ifPi49178p/X3951fWn2XyXvvQR2jg7B2qb8HFPnVV/tXn97XmZn62iNPtajCNtp1C2zlUqcYP1aWH9cnnrydsmhb/epY/tX5moRsBpAAAAAAAAAAAAAAAAAAAAAAAAAw2ux5tGjLG8asfhnb9QNF1OHecOVPOElL6jAx3D9Xu+IYx8KtOUf1/QDdZzwwIkpZn+oEWcll7lwLDe78BMjxvYg8cts9QKW22atgUuUVLHMubyyZFpvfyAtOWwFqT8gLEnvgv3FiTwiCHJ5kUWZeJBEadS4VOKzKUlFfHYtsDulKCp2tOkljkio/JEGHqtS1etLrh4+SAwtSpm/rteMkjE7XUPefruQYnWa1S20aOpUoOrOwqxunCO7lCKaqJevJKT+BJWMshQvqd1ZQubO4VahVjzQqUp7TT6dBMRO2ZWbflr3Hf5Tt6MvZl4TmuiXmovdvzwvMqpXPvlhfTbtBg1pNSbeeeo/okjUIzhpAAAAAAAAAAAAAAAAAAAAAAAAAhajSdbRLmmlmTptxXqt1+QGiuEa9jXo9VODx8VsBqNtN2+oUqyWHTmm/nuX2N8lPnfNF5T3QEeUsJ7/UgjSe4Ftv2mWIgU5wL/QU827IOZdo/GFXh3huqrR4r8u7Tw8vov3LdqIfD/EPbNxppN1O9s9dq2s4ybinGDpr3xaw155HrLcRDuP2SPtV8L/ah7LOKLnTaf3HibhfUlY6va9I16U+b7ve0lltUqqhP2XlxlBpt5izOfbnNr4fWLeFkqLEn1YwI8pPLyUR5y6kEdvqwI05PDecATdBtvvfF9jTazGM+8l7o7/sB2RvEW3jbqBrbqPmqVHs23J5AwcKifNLrmTZzna2eupt1SLAKriScdmn5E9o1S04L4ds9UuLiH8Tlb1ajqfw56rNWcG+qjTik+X/ABcmvDpsGttrdZckIQUadOEeWEIRUYwS6JJbJegTKjvfa659QuXSNMo9xoVtTfXkTfve/wCpuNMpxQAAAAAAAAAAAAAAAAAAAAAAAAHVYe6A5/Wou11arRawoTaWfFdV9MAatfUFR1WqkvYk+aPuZoZywruppcU95Q9l79fIz7FbfrkvoWm+pBbcmXYpcs9C2FPXqB89dslLvdKupLdqTf0RluNPzH7YLydHso4oqRb5oaRdNYeOlCYb1Din+x4vrqn9uztM06FaUbG47PZVa1PO0p07615Hj055fNmYc5f0OylsaZWZSTWPAosTl5kEWbzICzN4iBEqS6oL6bxwVZP/AHvUJLb/AEqb8/GX6BG6XtTu7CpjZtcq+IGs3tTudLk/FrCCxLBxniKT6HNrI6mMrqWIFLqerXwFpLnePDIPO838wXhM0+k7vWra3WcTmub3dX9C2SZdWSSWFsjbL0AAAAAAAAAAAAAAAAAAAAAAAAAANW1+35 [... cut ...]
    exposes: [e.switch(), e.power(), e.current(), e.voltage().withAccess(ea.STATE),
        e.energy(), exposes.enum('power_outage_memory', ea.STATE_SET, ['on', 'off', 'restore'])
            .withDescription('Recover state after power outage'),
        exposes.enum('indicator_mode', ea.ALL, ['off', 'off/on', 'on/off', 'on'])
            .withDescription('Plug LED indicator mode'), e.child_lock()],
    options: [exposes.options.measurement_poll_interval()],
    onEvent: tuya.onEventMeasurementPoll,
};
module.exports = definition;
```

Koenkk, 2022-07-18:
- Does the `power` and `voltage` work?
- Can you provide the `data/dababase.db` entry of this device?

chinyaolin, 2022-07-19:
> * Does the `power` and `voltage` work?
IT work fine.
PS. I remove OTA / Energy in External converter and the current should be **mA** not **A**
[image]

> * Can you provide the `data/dababase.db` entry of this device?
`{"id":23,"type":"Router","ieeeAddr":"0x70b3d52b6000f956","nwkAddr":52641,"manufId":4660,"manufName":"_TZ3000_okaz9tjs","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,6,3,4,5,57345,2820,1794],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_okaz9tjs","powerSource":1,"zclVersion":2,"appVersion":160,"stackVersion":1,"hwVersion":3,"dateCode":"20210625","swBuildId":"20B+TZSKT11BS105"}},"genOnOff":{"attributes":{"32768":1,"onOff":1,"tuyaBacklightMode":1,"moesStartUpOnOff":2}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":115,"rmsCurrent":80,"activePower":4}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":160,"stackVersion":1,"hwVersion":3,"dateCode":"20210625","swBuildId":"20B+TZSKT11BS105","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1658192157577,"defaultSendRequestWhen":"immediate"}`

dsolva, 2023-04-21:
I have some of these plugs but "power_precision" doesn't work. I saw a video from the seller using it with ZHA and reports decimals but I haven't been able to get that with z2m. Should this work on the current converter?