Koenkk/zigbee2mqtt #13195: [New device support]: SMCZB02-DW

jtbrush56, 2022-07-18:
### Link

https://fccid.io/JI5-SMCZB02-DW

### Database entry

{"id":8,"type":"EndDevice","ieeeAddr":"0x000d6f000a7a36e6","nwkAddr":23919,"manufId":4256,"powerSource":"Unknown","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":65535,"inClusterList":[0],"outClusterList":[],"clusters":{"ssIasZone":{"attributes":{}},"genBasic":{"attributes":{"powerSource":0,"zclVersion":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1658153674514,"defaultSendRequestWhen":"immediate"}

### Comments

I've followed the guide for adding new device support, but my device seems to be reporting an empty data object

`zigbee2mqtt      | Zigbee2MQTT:info  2022-07-18 14:14:23: MQTT publish: topic 'zigbee2mqtt/0x000d6f000a7a36e6', payload '{"enrolled":true,"last_seen":"2022-07-18T14:14:23.477Z","linkquality":null}'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-07-18 14:14:23: Received Zigbee message from '0x000d6f000a7a36e6', type 'readResponse', cluster 'ssIasZone', data '{}' from endpoint 1 with groupID null`

Any help would be greatly appreciated. I'm new to zigbee and home automation, and I feel like I'm fishing in the dark a bit here if I can't get any data to return.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    smc_contact_alarm: {
        cluster: 'ssIasZone',
        type: ['commandStatusChangeNotification', 'attributeReport'],
        convert: (model, msg, publish, options, meta) => {
            const zoneStatus = msg.data.zonestatus;
            return {
                zoneStatus,
            };
        },
    },
}

const definition = {
    fingerprint: [
        {type: 'EndDevice', manufacturerID: 4256},
    ],
    model: 'ZB02-DW', // Vendor model number, look on the device for a model number
    vendor: 'SMC', // Vendor of the device (only used for documentation and startup logging)
    description: 'Contact Sensor',
    toZigbee: [],
    exposes: [],
    exposes: [e.contact(), e.battery(), e.voltage()],
    fromZigbee: [fz.battery, fz.ias_enroll, fzLocal.smc_contact_alarm],
    toZigbee: [],
    meta: {battery: {voltageToPercentage: '3V_2500_3200'}},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['ssIasZone']);
        await reporting.batteryVoltage(endpoint);
        await endpoint.read('ssIasZone', ['iasCieAddr', 'zoneState', 'zoneId']);
    },
};

module.exports = definition;
```

jtbrush56, 2022-07-18:
@Koenkk I'm sure I'm messing up about 1000 things here, but I'm happy to try anything you think might work.

jtbrush56, 2022-07-18:
This is when the contact is closed.
`zigbee2mqtt      | Zigbee2MQTT:debug 2022-07-18 19:43:22: Received Zigbee message from '0x000d6f000a7a36e6', type 'readResponse', cluster 'ssIasZone', data '{}' from endpoint 1 with groupID null
zigbee2mqtt      | Zigbee2MQTT:info  2022-07-18 19:43:22: MQTT publish: topic 'zigbee2mqtt/0x000d6f000a7a36e6', payload '{"enrolled":true,"last_seen":"2022-07-18T19:43:22.490Z","linkquality":255}'`

jtbrush56, 2022-07-18:
If we can get this working, I'll submit a PR for the SMC manufacturer.

jtbrush56, 2022-07-19:
Does the `data '{}'` mean that it's not transmitting any information or that I have something configured wrong?

Koenkk, 2022-07-19:
> Does the data '{}' mean that it's not transmitting any information or that I have something configured wrong?

The message doesn't contain anything. The problem is that the device publishes exactly the same message for open and for close. 

If you remove the `configure` method from  your external converter and repair the device, does it still send this empty message?

jtbrush56, 2022-07-19:
I removed the configure block. I force removed the device, restarted z2m and repaired the device. This was the result.
`zigbee2mqtt      | Zigbee2MQTT:info  2022-07-19 16:03:37: Device '0x000d6f000a7a36e6' joined`
`zigbee2mqtt      | Zigbee2MQTT:info  2022-07-19 16:03:37: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f000a7a36e6","ieee_address":"0x000d6f000a7a36e6"},"type":"device_joined"}'`
`zigbee2mqtt      | Zigbee2MQTT:info  2022-07-19 16:03:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x000d6f000a7a36e6"},"type":"device_connected"}'`
`zigbee2mqtt      | Zigbee2MQTT:info  2022-07-19 16:03:37: MQTT publish: topic 'zigbee2mqtt/0x000d6f000a7a36e6', payload '{"last_seen":"2022-07-19T16:03:37.690Z"}'`
`zigbee2mqtt      | Zigbee2MQTT:info  2022-07-19 16:03:37: Starting interview of '0x000d6f000a7a36e6'`
`zigbee2mqtt      | Zigbee2MQTT:info  2022-07-19 16:03:37: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f000a7a36e6","ieee_address":"0x000d6f000a7a36e6","status":"started"},"type":"device_interview"}'`
`zigbee2mqtt      | Zigbee2MQTT:info  2022-07-19 16:03:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x000d6f000a7a36e6"},"type":"pairing"}'`
`zigbee2mqtt      | Zigbee2MQTT:debug 2022-07-19 16:03:38: Received Zigbee message from '0x000d6f000a7a36e6', type 'readResponse', cluster 'ssIasZone', data '{}' from endpoint 1 with groupID null`
`zigbee2mqtt      | Zigbee2MQTT:debug 2022-07-19 16:03:38: Skipping message, definition is undefined and still interviewing`
`zigbee2mqtt      | Zigbee2MQTT:info  2022-07-19 16:03:38: MQTT publish: topic 'zigbee2mqtt/0x000d6f000a7a36e6', payload '{"last_seen":"2022-07-19T16:03:38.135Z","linkquality":244}'`
`zigbee2mqtt      | Zigbee2MQTT:info  2022-07-19 16:03:53: MQTT publish: topic 'zigbee2mqtt/0x000d6f000a7a36e6', payload '{"last_seen":"2022-07-19T16:03:53.164Z","linkquality":244}'`
`zigbee2mqtt      | Zigbee2MQTT:debug 2022-07-19 16:03:53: Received Zigbee message from '0x000d6f000a7a36e6', type 'readResponse', cluster 'ssIasZone', data '{}' from endpoint 1 with groupID null`
`zigbee2mqtt      | Zigbee2MQTT:debug 2022-07-19 16:03:53: Skipping message, definition is undefined and still interviewing`
`zigbee2mqtt      | Zigbee2MQTT:info  2022-07-19 16:03:53: MQTT publish: topic 'zigbee2mqtt/0x000d6f000a7a36e6', payload '{"last_seen":"2022-07-19T16:03:53.652Z","linkquality":216}'`

Koenkk, 2022-07-19:
No improvement, this is really strange and I'm wondering how this can even work with the original hub. Do you have this hub and can you confirm if it works correctly there? If yes it would be interesting to see the send zigbee messages when the contact is closed/opened: https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

jtbrush56, 2022-07-19:
They were part of an ademco alarm system when I bought my house. So I don’t have the original hub, unless it’s the defunct wall panel. There are 10 sensors and they all transmit the same message.

jtbrush56, 2022-07-19:
Could my fzlocal function be modifying the message? Does it look okay to you?

jtbrush56, 2022-07-19:
If I reduce it to this: 
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {type: 'EndDevice', manufacturerID: 4256},
    ],
    model: 'ZB02-DW', // Vendor model number, look on the device for a model number
    vendor: 'SMC', // Vendor of the device (only used for documentation and startup logging)
    description: 'Contact Sensor',
    toZigbee: [],
    exposes: [e.contact()],
    fromZigbee: [fz.contact],
};

module.exports = definition;
```

I get this error when starting z2m:
```
zigbee2mqtt      | Using '/app/data' as data directory
zigbee2mqtt      |
zigbee2mqtt      | /app/node_modules/zigbee-herdsman-converters/index.js:91
zigbee2mqtt      |         if (converter.options) {
zigbee2mqtt      |                       ^
zigbee2mqtt      | TypeError: Cannot read properties of undefined (reading 'options')
zigbee2mqtt      |     at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:91:23)
zigbee2mqtt      |     at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
zigbee2mqtt      |     at new Controller (/app/lib/controller.ts:84:58)
zigbee2mqtt      |     at start (/app/index.js:108:18)
zigbee2mqtt      | Using '/app/data' as data directory
zigbee2mqtt      |
zigbee2mqtt      | /app/node_modules/zigbee-herdsman-converters/index.js:91
zigbee2mqtt      |         if (converter.options) {
zigbee2mqtt      |                       ^
zigbee2mqtt      | TypeError: Cannot read properties of undefined (reading 'options')
zigbee2mqtt      |     at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:91:23)
zigbee2mqtt      |     at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
zigbee2mqtt      |     at new Controller (/app/lib/controller.ts:84:58)
zigbee2mqtt      |     at start (/app/index.js:108:18)
```

Koenkk, 2022-07-19:
Yes everything looks okay, I expect that the issue is on the device side. The converter above doesn't work because `fz.contact` doesn't exist.

jtbrush56, 2022-07-19:
Okay. I cleared out all my devices of that type from my z2m database, cut my converter down to what I believe is close to bare minimum, repaired the device, and perform open and close events.

Converter:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {type: 'EndDevice', manufacturerID: 4256},
    ],
    model: 'ZB02-DW', // Vendor model number, look on the device for a model number
    vendor: 'SMC', // Vendor of the device (only used for documentation and startup logging)
    description: 'Contact Sensor',
    toZigbee: [],
    exposes: [e.contact()],
    fromZigbee: [fz.ias_contact_alarm_1],
};

module.exports = definition;
```
Logs:

```
zigbee2mqtt      | Zigbee2MQTT:info  2022-07-19 18:11:26: Device '0x000d6f000a7a36e6' joined
zigbee2mqtt      | Zigbee2MQTT:info  2022-07-19 18:11:26: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f000a7a36e6","ieee_address":"0x000d6f000a7a36e6"},"type":"device_joined"}'
zigbee2mqtt      | Zigbee2MQTT:info  2022-07-19 18:11:26: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x000d6f000a7a36e6"},"type":"device_connected"}'
zigbee2mqtt      | Zigbee2MQTT:info  2022-07-19 18:11:26: MQTT publish: topic 'zigbee2mqtt/0x000d6f000a7a36e6', payload '{"last_seen":"2022-07-19T18:11:26.316Z"}'
zigbee2mqtt      | Zigbee2MQTT:info  2022-07-19 18:11:26: Starting interview of '0x000d6f000a7a36e6'
zigbee2mqtt      | Zigbee2MQTT:info  2022-07-19 18:11:26: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f000a7a36e6","ieee_address":"0x000d6f000a7a36e6","status":"started"},"type":"device_interview"}'
zigbee2mqtt      | Zigbee2MQTT:info  2022-07-19 18:11:26: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x000d6f000a7a36e6"},"type":"pairing"}'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-07-19 18:11:26: Received Zigbee message from '0x000d6f000a7a36e6', type 'readResponse', cluster 'ssIasZone', data '{}' from endpoint 1 with groupID null
zigbee2mqtt      | Zigbee2MQTT:debug 2022-07-19 18:11:26: Skipping message, definition is undefined and still interviewing
zigbee2mqtt      | Zigbee2MQTT:info  2022-07-19 18:11:26: MQTT publish: topic 'zigbee2mqtt/0x000d6f000a7a36e6', payload '{"last_seen":"2022-07-19T18:11:26.792Z","linkquality":255}'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-07-19 18:11:30: Received Zigbee message from '0x000d6f000a7a36e6', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID null
zigbee2mqtt      | Zigbee2MQTT:debug 2022-07-19 18:11:30: Skipping message, definition is undefined and still interviewing
zigbee2mqtt      | Zigbee2MQTT:info  2022-07-19 18:11:30: MQTT publish: topic 'zigbee2mqtt/0x000d6f000a7a36e6', payload '{"last_seen":"2022-07-19T18:11:30.933Z","linkquality":255}'
zigbee2mqtt      | Zigbee2MQTT:info  2022-07-19 18:11:41: MQTT publish: topic 'zigbee2mqtt/0x000d6f000a7a36e6', payload '{"last_seen":"2022-07-19T18:11:41.759Z","linkquality":255}'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-07-19 18:11:42: Received Zigbee message from '0x000d6f000a7a36e6', type 'readResponse', cluster 'ssIasZone', data '{}' from endpoint 1 with groupID null
zigbee2mqtt      | Zigbee2MQTT:debug 2022-07-19 18:11:42: Skipping message, definition is undefined and still interviewing
zigbee2mqtt      | Zigbee2MQTT:info  2022-07-19 18:11:42: MQTT publish: topic 'zigbee2mqtt/0x000d6f000a7a36e6', payload '{"last_seen":"2022-07-19T18:11:42.263Z","linkquality":252}'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-07-19 18:11:59: Received Zigbee message from '0x000d6f000a7a36e6', type 'readResponse', cluster 'ssIasZone', data '{}' from endpoint 1 with groupID null
zigbee2mqtt      | Zigbee2MQTT:debug 2022-07-19 18:11:59: Skipping message, definition is undefined and still interviewing
zigbee2mqtt      | Zigbee2MQTT:info  2022-07-19 18:11:59: MQTT publish: topic 'zigbee2mqtt/0x000d6f000a7a36e6', payload '{"last_seen":"2022-07-19T18:11:59.780Z","linkquality":255}'
zigbee2mqtt      | Zigbee2MQTT:info  2022-07-19 18:12:27: MQTT publish: topic 'zigbee2mqtt/0x000d6f000a7a36e6', payload '{"last_seen":"2022-07-19T18:12:27.941Z","linkquality":255}'
zigbee2mqtt      | Zigbee2MQTT:debug 2022-07-19 18:12:28: Received Zigbee message from '0x000d6f000a7a36e6', type 'readResponse', cluster 'ssIasZone', data '{}' from endpoint 1 with groupID null
zigbee2mqtt      | Zigbee2MQTT:debug 2022-07-19 18:12:28: Skipping message, definition is undefined and still interviewing
zigbee2mqtt      | Zigbee2MQTT:info  2022-07-19 18:12:28: MQTT publish: topic 'zigbee2mqtt/0x000d6f000a7a36e6', payload '{"last_seen":"2022-07-19T18:12:28.418Z","linkquality":244}'
zigbee2mqtt      | Zigbee2MQTT:info  2022-07-19 18:12:38: MQTT publish: topic 'zigbee2mqtt/0x000d6f000a7a36e6', payload '{"last_seen":"2022-07-19T18:12:38.229Z","linkquality":244}'
[... cut ...]
zigbee2mqtt      | Zigbee2MQTT:debug 2022-07-19 18:13:03: Received Zigbee message from '0x000d6f000a7a36e6', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 1 with groupID null
zigbee2mqtt      | Zigbee2MQTT:debug 2022-07-19 18:13:03: Skipping message, definition is undefined and still interviewing
zigbee2mqtt      | Zigbee2MQTT:info  2022-07-19 18:13:03: MQTT publish: topic 'zigbee2mqtt/0x000d6f000a7a36e6', payload '{"last_seen":"2022-07-19T18:13:03.286Z","linkquality":255}'
```

jtbrush56, 2022-07-19:
What does the log `Skipping message, definition is undefined and still interviewing` mean? Which definition is undefined?

Koenkk, 2022-07-20:
@jtbrush56 this means the device has not been linked to a definition (converter) yet because z2m didn't got enough information from the device to properly identify itself. After the interview it will have all the required info to match it with a converter.

closed 2022-08-27 by github-actions[bot]