Koenkk/zigbee2mqtt #13199: [Feature request]: Support timers for ZVG1 RTX

dominikkarall, 2022-07-18:
### Is your feature request related to a problem? Please describe

This feature request should be used to work together on timer support for ZVG1 RTX.

### Describe the solution you'd like

I try to decode the string from DP 17 which represents the normal timers and would like to share my current status of my elaboration:
 - How do I try to decode DP17 (= Regular Schedule)
I have a Tuya Zigbee gateway which is connected to the irrigation valve. To get the current value of DP17 I'm using the device log on the Tuya IoT plattform. E.g. I set a timer for 08:00 15min every day and check the log. I copy the value and change 15min to 14min to see which values change.

 - Current status
Based on my tests, I could identify that the Tuya string represents the following binary data:
A=000000 (dec 0)
B=000001 (dec 1)
C=..
...
a=011010 (dec 26)
...
0=110100 (dec 52)
...
9=111110 (dec 62)
other values like =, / or + I assume are 111111 - not sure yet if that is correct.

Lets see what that looks like for a timer example: 8:00, 15min, every day
Tuya string from DP17: AAEB4AAPf2QBB+YHEQE=
Converted to binary and padded to groups of 4:
```
0000 0000 0000 0001 0000 0001 1110 0000 0000 0000 0000 1111 0111 1111 0110 0100 0000 0001 0000 0111 1111 0110 0000 0111 0001 0001 0000 0001 0011 1111
```

I could identify the following values within that binary string:
```
0000 0000 0000 0001 HEADER
0000 0001 1110 0000 Minutes after midnight
0000 0000 0000 1111 Duration in minutes
0111 1111 Bits identify each day of the week. Sunday is the right bit, Saturday is represented by the leftmost 1.
0110 0100 0000 0001 0000 0111 1111 0110 0000 0111 0001 0001 0000 0001 0011 1111 FOOTER which changes if you add more timers
```

I couldn't identify the footer change yet, furthermore I"m not quite sure if =, / and + is interpreted correctly. Would be great if other users of that valve could investigate as well so that we can set and read the timer values within z2m.

You can use the following python script to convert from Tuya string to binary:
```
import sys

def convert(s):
    o = ord(s)
    if o >= ord("A") and o <= ord("Z"):
        return o-ord("A")
    if o >= ord("a") and o <= ord("z"):
        return o-ord("a")+26
    if o >= ord("0") and o <= ord("9"):
        return o-ord("0")+26+26
    return 63

def printbin(a):
    binstr = ""
    for x in a:
        binstr += "{:06b}".format(convert(x))
    s=binstr
    newstr=" ".join([s[::-1][i:i+4] for i in range(0, len(s), 4)])[::-1]
    print(newstr)

for x in sys.argv[1:]:
    printbin(x)
```
Copy that to convert.py and start it with
```
python3 convert.py AAEB4AAPf2QBB+YHEQE= 
```

### Describe alternatives you've considered

none

### Additional context

none

dominikkarall, 2022-07-19:
This script interprets the tuya string and displays the schedules:
```
import sys

def convert(s):
    o = ord(s)
    if o >= ord("A") and o <= ord("Z"):
        return o-ord("A")
    if o >= ord("a") and o <= ord("z"):
        return o-ord("a")+26
    if o >= ord("0") and o <= ord("9"):
        return o-ord("0")+26+26
    if o == ord("/"):
        return 63
    return o

def printbin(a):
    binstr = ""
    for x in a:
        binstr += "{:06b}".format(convert(x))
    s=binstr
    newstr=" ".join([s[::-1][i:i+4] for i in range(0, len(s), 4)])[::-1]
    for x in range(int(len(binstr)/(26*4))):
        startnr = x*26*4+12
        starttime = startnr+4
        startduration = starttime+16
        startweek = startduration+16
        startactive = startweek+5*4
        nr = int(binstr[startnr:startnr+4],2)
        time = int(binstr[starttime:starttime+16],2)
        duration = int(binstr[startduration:startduration+16],2)
        week = binstr[startweek+1:startweek+7]
        active = int(binstr[startactive:startactive+4],2)
        print(f"Nr {nr} Time {int(time/60)}:{time%60}h Duration {duration}min week sa-su {week} active {active}")

for x in sys.argv[1:]:
    printbin(x)
```

Example
```
python3 convert.py AAEB4AAPf2QBB+YHEQECAyoAD39kAQfmBxIBAwRXABB3ZAEH5gcSAQ==
Nr 1 Time 8:0h Duration 15min week sa-su 111111 active 1
Nr 2 Time 13:30h Duration 15min week sa-su 111111 active 1
Nr 3 Time 18:31h Duration 16min week sa-su 111011 active 1
```

Next steps:
 - create a script to generate tuya string

Would be great if somebody could test the script with his tuya string to see if it works for everybody.

dominikkarall, 2022-07-20:
Python script to generate Tuya string:
```
# change the line below for your schedules
# value 1: start time in minutes after midnight
# value 2: duration in minutes
# value 3: days you want to be active (sa, fr, th, we, tu, mo, su)
# value 4: active = 1, inactive = 0
schedulelist=["480 15 1111111 1", "810 15 1111111 1", "1111 16 1110111 1", "1200 15 1111111 1"]
schedules = []
for x in schedulelist:
    schedules.append(x.split(" "))

schedule_header = [
        "0000000000000001",
        "0000000100000010",
        "0000000100000011",
        "0000000100000100"
        ]

schedule_footer = [
        "00000122222201100000011100010001",
        "00000111111001100000011100010010",
        "00000111111001100000011100010010",
        "00000110101101100000011100010010"
        ]

footer_end = [
        "0000000100333333",
        "00000001",
        "000000010000333333333333",
        "0000000100333333"
        ]

def converttobin(s, l):
    return "{val:0{l}b}".format(l=l,val=int(s))

def converttime(t):
    return converttobin(t, 16)

def convertduration(d):
    return converttobin(d, 16)

def convertweek(w):
    return "0" + w

def convertactive(a):
    return converttobin(a, 4)

def generatebin(schedule, header, footer):
    unknown = "011001000000"
    return header + converttime(schedule[0]) + convertduration(schedule[1]) + convertweek(schedule[2]) + unknown + convertactive(schedule[3]) + footer

def generateallbin(schedules):
    i=0
    binstr = ""
    for schedule in schedules:
        binstr += generatebin(schedule, schedule_header[i], schedule_footer[i])
        i+=1
    binstr += footer_end[i-1]
    s=binstr
    newstr=" ".join([s[::-1][i:i+4] for i in range(0, len(s), 4)])[::-1]
    return binstr

def binstrtotuyastr(binstr):
    tuyastr = ""
    for x in range(int(len(binstr)/6)):
        tuyastr += bintoalpha(binstr[x*6:x*6+6])
    return tuyastr

def bintoalpha(binstr):
    if binstr == "222222":
        return "+"
    elif binstr == "333333":
        return "="

    val = int(binstr,2)
    if val >= 0 and val <= 25:
        return chr(val+ord("A"))
    if val >= 26 and val <= 51:
        return chr(val-26+ord("a"))
    if val >= 52 and val <= 62:
        return chr(val-26-26+ord("0"))
    if val == 63:
        return "/"
    return "_"

binstr = generateallbin(schedules)
tuyastr = binstrtotuyastr(binstr)
print("Tuya Buffer: " + tuyastr)
```
Run the script with
```
python3 generate.py
```

Next steps:
 - Test sending the generated string to the device and see if it works properly

dominikkarall, 2022-07-20:
Found one bug, sendDataPointStringBuffer requires an array instead of a string. Nevertheless it still doesn't work. Last connection time is not updated either so it seems to me like the data is not being transmitted yet.

dominikkarall, 2022-07-21:
Unfortunately I can't send data to DP 17. I tried following external converter:
```
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different data points
    // it's usually necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE200_2wg5qrjy'
        },
    ],
    model: 'ZVG1',
    vendor: 'RTX',
    description: 'Zigbee smart water valve',
    fromZigbee: [{
                cluster: 'manuSpecificTuya',
                type: ['commandDataResponse', 'commandDataReport'],
                convert: (model, msg, publish, options, meta) => {
            const result = {};
            for (const dpValue of msg.data.dpValues) {
                const dp = dpValue.dp; // First we get the data point ID
                const value = tuya.getDataValue(dpValue); // This function will take care of converting the data to proper JS type
                
                //meta.logger.info(`zigbee-herdsman-converters:RTXZVG1Valve: DP ` + `#${dp} with data ${JSON.stringify(value)}`);
                    
                switch (dp) {
                    case tuya.dataPoints.state: {
                        return {state: value ? 'ON': 'OFF'};
                    }
                    case 5: {
                        // i assume value is reported in fl. oz.
                        return {water_consumed: (value / 33.8140226).toFixed(2)};
                    }
                    case 7: {
                        return {battery: value};
                    }
                    case 11: {
                        // value reported in seconds
                        return {timer_time_left: value / 60};
                    }
                    case 12: {
                        if (value === 0) return {timer_state: 'disabled'};
                        else if (value === 1) return {timer_state: 'active', state: 'ON'};
                        else return {timer_state: 'enabled', state: 'OFF'};
                    }
                    case 15: {
                        // value reported in seconds
                        return {last_valve_open: value / 60};
                    }
                    default: {
                        meta.logger.warn(`zigbee-herdsman-converters: RTXZVG1Valve: NOT RECOGNIZED DP ` +
                            `#${dp} with data ${JSON.stringify(value)}`);
                    }
                }
            }
            return result;
        }
        },
        fz.ignore_tuya_set_time,
        fz.ignore_basic_report,
        //fz.tuya_data_point_dump, // This is a debug converter, it will be described in the next part
    ],
    toZigbee: [
            tz.tuya_switch_state,
                {
            key: ['timer'],
            convertSet: async (entity, key, value, meta) => {
                            const timer = 60 * Math.abs(Math.min(value, 360));
                await tuya.sendDataPointValue(entity, 11, timer); // tuya.sendDataPoint* functions take care of converting the data to proper format
                            return {state: {timer: value}};
            }
            },
            {
            key: ['normal_schedule'],
            convertSet: async (entity, key, value, meta) => {
               //////////////// THIS PART DOES NOT WORK, ANY IDEAS?
               let data = tuya.convertStringToHexArray(value);
               await tuya.sendDataPointStringBuffer(entity, 17, data);
               return {state: {normal_schedule: 0}};
            }
            },
        //tz.tuya_data_point_test, // Another debug converter
    ],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [e.switch().setAccess('state', ea.STATE_SET), e.battery(),
              exposes.enum('timer_state', ea.STATE_GET, ['disabled', 'active', 'enabled']),
              exposes.numeric('timer', ea.STATE_SET).withValueMin(0).withValueMax(360).withUnit('min')
                .withDescription('Auto off after specific time'),
              exposes.numeric('timer_time_left', ea.STATE_GET).withUnit('min')
                .withDescription('Auto off timer time left'),
              exposes.numeric('last_valve_open', ea.STATE_GET).withUnit('min')
                .withDescription('Time the valve was open when state on'),
              exposes.numeric('water_consumed', ea.STATE_GET).withUnit('l')
                .withDescription('Liters of water consumed'),
              exposes.text('normal_schedule', ea.STATE_SET)],
};

module.exports = definition;
```

It seems like DP 17 seems to require a different format of data. If somebody has hardware to sniff it would be great to see which data the app sends to DP 17. If I would get that info it should be easy to decode.

fhempy, 2022-08-01:
I bought a CC2531 to sniff the traffic to the valve. The bits and bytes are more clear now :)

Hex:
```
create new timer for 8am, 15min, every day:
00 02 01 01 e0 00 0f 7f 64 01 07 e6 08 01 01

the values are:
00       header
02       edit / 03 new / 04 delete / 01 report
01       timer nr - only used for edit and delete
01 e0    time in minutes since midnight
00 0e    duration
7f       weekdays bitmask
64       unknown
01       active / 00 inactive
07 e6 08 01 01 footer
```
...and the datapoint is RAW instead of String.

That's it! I tested it successfully by creating 3 timers. Just the delete function for timers doesn't work properly right now.

Next step:
 - create an external converter for testing
 - elaborate further on delete timers function

fhempy, 2022-08-02:
[rtx.js.log](https://github.com/Koenkk/zigbee2mqtt/files/9245200/rtx.js.log)

Find the external converter file attached (remove .log extension):
 - support normal timers 1-4
 - support weather_delay

Next steps:
 - wait for feedback from testers (@ytzelf maybe you would like to test 😄 )
 - create PR to include timers and weather_delay in the next release

oat1, 2022-08-03:
Hi, great work ! I also have this valve, maybe you could help resolving this issue
https://github.com/Koenkk/zigbee2mqtt/issues/12184
Water consumption also does not work correctly on Z2Mqtt

fhempy, 2022-08-03:
> Hi, great work ! I also have this valve, maybe you could help resolving this issue #12184 Water consumption also does not work correctly on Z2Mqtt

Hi,
can you please let me know if timers work for you as soon as you have tested it? Would be great to have some feedback to make sure that it works for everyone.

fhempy, 2022-08-03:
[rtx.js.log](https://github.com/Koenkk/zigbee2mqtt/files/9254856/rtx.js.log)

Updated with cycle timer support 😄 
Afaik there is nothing more that can be integrated for this device. Please report if it works for you. As soon as I have feedback on this, I'm going to start working on a PR.

oat1, 2022-08-04:
> Hi, can you please let me know if timers work for you as soon as you have tested it? Would be great to have some feedback to make sure that it works for everyone.

I would like to help, but I only have conbee II and Z2Mqtt. I do not have tuya app or tuya gateway. I cannot set timer in tuya app. I have installed your file but I do not see any additional feature in Z2MQTT. My timer worked well even before.

[image]

oat1, 2022-08-04:
Ok, I've made a change and now I see additional features
In your file there is only:

```
  fingerprint: [
    {
      modelID: 'TS0601',
      manufacturerName: '_TZE200_2wg5qrjy'
    },
```
My model is : _TZE200_akjefhj5
so i had to change it to oryginal z2mqtt:

```
  fingerprint: [
    {modelID: 'TS0601', manufacturerName: '_TZE200_akjefhj5'}, 
    {modelID: 'TS0601', manufacturerName: '_TZE200_2wg5qrjy'},
    {modelID: 'TS0601', manufacturerName: '_TZE200_81isopgh'},
  ],
```
Also, one thing I am missing,

In the original configuration, I could choose "timer state"
[image]

And now I have only exposed state

[image]

Could you fix it ?

oat1, 2022-08-04:
I've made two additional changes 

every
`last_valve_open`

changed on
`last_valve_open_duration`

and

`exposes.enum('timer_state', ea.STATE, ['disabled', 'active', 'enabled']),`

on

`exposes.enum('timer_state', ea.STATE_SET, ['disabled', 'active', 'enabled']),`

now all works fine :)

fhempy, 2022-08-04:
Hi,
thanks for the adaptations! Are you sure, that timer_state can be set as well? I can't find such an option in the SmartLife app. Can you verify if the timers are inactive when set to disabled?
Thx!

oat1, 2022-08-04:
Yeah I am sure it can be set.
It works on default z2mqtt config look below
[image]

Unfortunately the changes i made to your file rtx.js did not make it work. I'm receiving such an error 
[image]

oat1, 2022-08-05:
I've added:
`  toZigbee: [
    tz.ZVG1_timer_state,`

now it exposes it right, after clicking on option

fhempy, 2022-08-05:
Thanks for the update! Please let me know in the evening if the change has any effect. If not, I would keep it as information only.

fhempy, 2022-08-05:
Thanks for testing!
I'm going to add the other changes you mentioned.

Regarding the timers, you can always deactivate a timer by changing the last value within the timer text field from 1 to 0.

fhempy, 2022-08-08:
PR is ready: https://github.com/Koenkk/zigbee-herdsman-converters/pull/4519

closed 2022-08-08 by fhempy

deviantintegral, 2022-09-03:
@fhempy it looks like I can no longer disable the timer entirely. I'm not sure I actually verified this was working, and it's now showing up as a read-only field instead of a dropdown of states. Did you change it because it's not possible to disable anyways?

fhempy, 2022-09-03:
@deviantintegral do you mean the timer_state setting which shows disabled / active / enabled in the screenshot above? This setting can't be set, even if you send it to the valve it doesn't change anything.
You can disable all timers by changing the last value in the timer string. It's 1 for enabled and 0 for disabled.

deviantintegral, 2022-09-07:
I think I understand this.

- cycle timers let you schedule a timer with a rest period between waterings, such as on for 10 minutes and off for 10 minutes
- normal schedule timers just do a basic on / off
- There is no way to disable the "Auto off" timer for when the valve is turned on manually

[image]

Thanks for your work on this!

fhempy, 2022-09-07:
It's exactly as you described it

Talw3g, 2023-06-30:
Hello,

I don't know if I should post it here, but I tried to use the normal timer with openhab: it works fine for setting the start time and duration, but active days are behaving strangely...
The command received from OH is OK and sent to the device, but the ACK data received back contains the wrong days.
Here is the log when I change the active days in OH:
```
debug 2023-06-30 11:07:55Received MQTT message on 'zigbee2mqtt/prog_arrosage/set' with data '{"normal_schedule_timer_1": "05:30 / 5.0 / WeFrSu / 1"}'
debug 2023-06-30 11:07:55Publishing 'set' 'normal_schedule_timer_1' to 'prog_arrosage'
info 2023-06-30 11:07:55MQTT publish: topic 'zigbee2mqtt/prog_arrosage/state', payload 'OFF'
info 2023-06-30 11:07:55MQTT publish: topic 'zigbee2mqtt/prog_arrosage/timer_state', payload 'enabled'
info 2023-06-30 11:07:55MQTT publish: topic 'zigbee2mqtt/prog_arrosage/battery', payload '50'
info 2023-06-30 11:07:55MQTT publish: topic 'zigbee2mqtt/prog_arrosage/linkquality', payload '149'
info 2023-06-30 11:07:55MQTT publish: topic 'zigbee2mqtt/prog_arrosage/timer_time_left', payload '1'
info 2023-06-30 11:07:55MQTT publish: topic 'zigbee2mqtt/prog_arrosage/last_valve_open_duration', payload '5'
info 2023-06-30 11:07:55MQTT publish: topic 'zigbee2mqtt/prog_arrosage/water_consumed', payload '6.21'
info 2023-06-30 11:07:55MQTT publish: topic 'zigbee2mqtt/prog_arrosage/timer', payload '1'
info 2023-06-30 11:07:55MQTT publish: topic 'zigbee2mqtt/prog_arrosage/cycle_timer_1', payload ''
info 2023-06-30 11:07:55MQTT publish: topic 'zigbee2mqtt/prog_arrosage/cycle_timer_2', payload ''
info 2023-06-30 11:07:55MQTT publish: topic 'zigbee2mqtt/prog_arrosage/cycle_timer_3', payload ''
info 2023-06-30 11:07:55MQTT publish: topic 'zigbee2mqtt/prog_arrosage/cycle_timer_4', payload ''
info 2023-06-30 11:07:55MQTT publish: topic 'zigbee2mqtt/prog_arrosage/normal_schedule_timer_1', payload '05:30 / 5.0 / WeFrSu / 1'
info 2023-06-30 11:07:55MQTT publish: topic 'zigbee2mqtt/prog_arrosage/normal_schedule_timer_2', payload ''
info 2023-06-30 11:07:55MQTT publish: topic 'zigbee2mqtt/prog_arrosage/normal_schedule_timer_3', payload ''
info 2023-06-30 11:07:55MQTT publish: topic 'zigbee2mqtt/prog_arrosage/normal_schedule_timer_4', payload ''
debug 2023-06-30 11:07:55Received Zigbee message from 'prog_arrosage', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,1,1,74,0,5,21,64,1,7,230,8,1,1],"type":"Buffer"},"datatype":0,"dp":17}],"seq":4608}' from endpoint 1 with groupID 0
info 2023-06-30 11:07:55MQTT publish: topic 'zigbee2mqtt/prog_arrosage/state', payload 'OFF'
info 2023-06-30 11:07:55MQTT publish: topic 'zigbee2mqtt/prog_arrosage/timer_state', payload 'enabled'
info 2023-06-30 11:07:55MQTT publish: topic 'zigbee2mqtt/prog_arrosage/battery', payload '50'
info 2023-06-30 11:07:55MQTT publish: topic 'zigbee2mqtt/prog_arrosage/linkquality', payload '149'
info 2023-06-30 11:07:55MQTT publish: topic 'zigbee2mqtt/prog_arrosage/timer_time_left', payload '1'
info 2023-06-30 11:07:55MQTT publish: topic 'zigbee2mqtt/prog_arrosage/last_valve_open_duration', payload '5'
info 2023-06-30 11:07:55MQTT publish: topic 'zigbee2mqtt/prog_arrosage/water_consumed', payload '6.21'
info 2023-06-30 11:07:55MQTT publish: topic 'zigbee2mqtt/prog_arrosage/timer', payload '1'
info 2023-06-30 11:07:55MQTT publish: topic 'zigbee2mqtt/prog_arrosage/cycle_timer_1', payload ''
info 2023-06-30 11:07:55MQTT publish: topic 'zigbee2mqtt/prog_arrosage/cycle_timer_2', payload ''
info 2023-06-30 11:07:55MQTT publish: topic 'zigbee2mqtt/prog_arrosage/cycle_timer_3', payload ''
info 2023-06-30 11:07:55MQTT publish: topic 'zigbee2mqtt/prog_arrosage/cycle_timer_4', payload ''
info 2023-06-30 11:07:55MQTT publish: topic 'zigbee2mqtt/prog_arrosage/normal_schedule_timer_1', payload '05:30 / 5 / ThTuSu / 1'
info 2023-06-30 11:07:55MQTT publish: topic 'zigbee2mqtt/prog_arrosage/normal_schedule_timer_2', payload ''
info 2023-06-30 11:07:55MQTT publish: topic 'zigbee2mqtt/prog_arrosage/normal_schedule_timer_3', payload ''
info 2023-06-30 11:07:55MQTT publish: topic 'zigbee2mqtt/prog_arrosage/normal_schedule_timer_4', payload ''
```

Please let me know if I should post it elsewhere or if you need other info.

Thanks,
Talw3g