Koenkk/zigbee2mqtt #13215: sonoff 3.0 wont join zigbee netwoork

gzcwnk, 2022-07-19:
### What happened?

sonoff wont join network as a router

### What did you expect to happen?

automatic pairing

### How to reproduce it (minimal and precise)

Flashed to router hex and went to "permit join and plugged it in

### Zigbee2MQTT version

1.26.0

### Adapter firmware version

20220125

### Adapter

sonoff 3.0

### Debug log

I got this one time,  but it has not come up in the logs since despite re-flashing.

=====
2022-07-19 20:53:37Successfully interviewed 'Coordinator', device has successfully been paired
info 2022-07-19 20:53:37Device 'Coordinator' is supported, identified as: Custom devices (DiY) Texas Instruments router (ti.router)
info 2022-07-19 20:53:37MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Texas Instruments router","exposes":[{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"ti.router","options":[],"supports_ota":false,"vendor":"Custom devices (DiY)"},"friendly_name":"Coordinator","ieee_address":"0x00124b0026b749a1","status":"successful","supported":true},"type":"device_interview"}'
info 2022-07-19 20:53:37Configuring 'Coordinator'
info 2022-07-19 20:53:37MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Texas Instruments router","friendly_name":"Coordinator","model":"ti.router","supported":true,"vendor":"Custom devices (DiY)"},"type":"pairing"}'
info 2022-07-19 20:53:37MQTT publish: topic 'homeassistant/sensor/0x00124b0026b749a1/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_undefined"],"manufacturer":"Custom devices (DiY)","model":"Texas Instruments router (ti.router)","name":"Coordinator","sw_version":"20220125"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/Coordinator","name":"Coordinator linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/Coordinator","unique_id":"undefined_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
info 2022-07-19 20:53:37MQTT publish: topic 'homeassistant/sensor/0x00124b0026b749a1/linkquality/config', payload 'null'
info 2022-07-19 20:53:37Successfully configured 'Coordinator'
info 2022-07-19 20:53:57Zigbee: allowing new devices to join.
=====

gzcwnk, 2022-07-19:
Can I get the older/ last firmware?

What I want to try is flash it back, look at the logs to see the old pre-20220125 version.  Try that and see if it works.  If not flasgh to 20220125 again and make sure that the number has changed, that will prove the flashing actually worked.

ashokaiyar, 2022-07-20:
Have you tried pushing either one of the two buttons on the dongle (you'll have to remove it from the case) during pairing?

ashokaiyar, 2022-07-20:
You can get the previous launchpad_router firmware here:

https://github.com/Koenkk/Z-Stack-firmware/blob/a4fbab342c131e8ec387e02ac9662d5344c48224/router/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_router_20210128.zip

closed 2022-08-27 by github-actions[bot]