Koenkk/zigbee2mqtt #13218: Tuya Zigbee Smart Socket 2 Gang Model : DMP202 not reporting any feedback when using physical switch

onewaylifestreet, 2022-07-20:
### What happened?

Tuya Zigbee Smart Socket 2 Gang Model : DMP202 not reporting any feedback when using physical switch

Device type
 : Router
Zigbee Model
 : TS011F
Zigbee Manufacturer
 : _TZ3000_mvn6jl7x
Description
 : 2 gang wall outlet
Support status
 : Supported

IEEE Address
0x84fd27fffed6b0d4
Network address
0x3384
Manufacturer
[TuYa](https://www.zigbee2mqtt.io/supported-devices/#v=TuYa)
Model
[TS011F_2_gang_wall](https://www.zigbee2mqtt.io/devices/TS011F_2_gang_wall.html#tuya-ts011f_2_gang_wall)

### What did you expect to happen?

When pressing the physical button on the device, the status on zigbee2mqtt should change but it is not.
Tried using ZHA and it is working correctly.

### How to reproduce it (minimal and precise)

Make the switch status working correctly when pressing physical button on the device.

### Zigbee2MQTT version

1.26.0-1

### Adapter firmware version

 20220219

### Adapter

CC2652R Vision

### Debug log

nothing show on the log when pressing physical button.

Debug 2022-07-20 14:32:14Received MQTT message on 'zigbee2mqtt/0x84fd27fffed6b0d4/set' with data '{"state_l1":"ON"}'
Debug 2022-07-20 14:32:14Publishing 'set' 'state' to '0x84fd27fffed6b0d4'
Info 2022-07-20 14:32:14MQTT publish: topic 'zigbee2mqtt/0x84fd27fffed6b0d4', payload '{"backlight_mode":null,"linkquality":51,"power_on_behavior":null,"state_l1":"ON","state_l2":"OFF"}'
Info 2022-07-20 14:32:14MQTT publish: topic 'zigbee2mqtt/0x84fd27fffed6b0d4/mode', payload '{"backlight":null}'
Debug 2022-07-20 14:32:14Received MQTT message on 'zigbee2mqtt/0x84fd27fffed6b0d4/set' with data '{"state_l2":"ON"}'
Debug 2022-07-20 14:32:14Publishing 'set' 'state' to '0x84fd27fffed6b0d4'
Info 2022-07-20 14:32:14MQTT publish: topic 'zigbee2mqtt/0x84fd27fffed6b0d4', payload '{"backlight_mode":null,"linkquality":61,"power_on_behavior":null,"state_l1":"ON","state_l2":"ON"}'
Info 2022-07-20 14:32:15MQTT publish: topic 'zigbee2mqtt/0x84fd27fffed6b0d4/mode', payload '{"backlight":null}'
Debug 2022-07-20 14:32:16Received MQTT message on 'zigbee2mqtt/0x84fd27fffed6b0d4/set' with data '{"state_l2":"OFF"}'
Debug 2022-07-20 14:32:16Publishing 'set' 'state' to '0x84fd27fffed6b0d4'
Info 2022-07-20 14:32:16MQTT publish: topic 'zigbee2mqtt/0x84fd27fffed6b0d4', payload '{"backlight_mode":null,"linkquality":61,"power_on_behavior":null,"state_l1":"ON","state_l2":"OFF"}'
Info 2022-07-20 14:32:16MQTT publish: topic 'zigbee2mqtt/0x84fd27fffed6b0d4/mode', payload '{"backlight":null}'
Debug 2022-07-20 14:32:16Received MQTT message on 'zigbee2mqtt/0x84fd27fffed6b0d4/set' with data '{"state_l1":"OFF"}'
Debug 2022-07-20 14:32:16Publishing 'set' 'state' to '0x84fd27fffed6b0d4'
Info 2022-07-20 14:32:16MQTT publish: topic 'zigbee2mqtt/0x84fd27fffed6b0d4', payload '{"backlight_mode":null,"linkquality":61,"power_on_behavior":null,"state_l1":"OFF","state_l2":"OFF"}'
Info 2022-07-20 14:32:16MQTT publish: topic 'zigbee2mqtt/0x84fd27fffed6b0d4/mode', payload '{"backlight":null}'

Koenkk, 2022-07-20:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_mvn6jl7x'},
        {modelID: 'TS011F', manufacturerName: '_TZ3000_raviyuvk'}, {modelID: 'TS011F', manufacturerName: '_TYZB01_hlla45kx'},
        {modelID: 'TS011F', manufacturerName: '_TZ3000_92qd4sqa'}, {modelID: 'TS011F', manufacturerName: '_TZ3000_zwaadvus'}],
    model: 'TS011F_2_gang_wall',
    vendor: 'TuYa',
    description: '2 gang wall outlet',
    toZigbee: extend.switch().toZigbee.concat([tz.moes_power_on_behavior, tz.tuya_backlight_mode]),
    fromZigbee: extend.switch().fromZigbee.concat([fz.moes_power_on_behavior, fz.tuya_backlight_mode]),
    exposes: [e.switch().withEndpoint('l1'), e.switch().withEndpoint('l2'),
        exposes.enum('power_on_behavior', ea.ALL, ['on', 'off', 'previous']),
        exposes.enum('backlight_mode', ea.ALL, ['LOW', 'MEDIUM', 'HIGH'])],
    whiteLabel: [{vendor: 'ClickSmart+', model: 'CMA30036'}],
    endpoint: (device) => {
        return {'l1': 1, 'l2': 2};
    },
    meta: {multiEndpoint: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await endpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

onewaylifestreet, 2022-07-20:
Dear Koen, the issue is fixed after using the new external converter.

Thank you.

Koenkk, 2022-07-21:
Pushed the fix!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-07-21 by Koenkk