Koenkk/zigbee2mqtt #13225: Zigbee binding not working

imuncas, 2022-07-20:
### What happened?

Migrating from Deconz to Z2M, migrated one Philips HUE dimmer and one Philips Hue light.
Pairing is successfull, however binding the dimmer to the light doesn't works.
What I did:
- created a group for the light
- added the light in the group. (controlling the light from the group UI works as expected)
- binded the dimmer to the group (Report successfull)
- actions on the dimmer are properly logges and available in MQTT
- actions on the dimmer have no impact on the lights however

### What did you expect to happen?

Actions on dimmer turn the lights on/off.

### Zigbee2MQTT version

1.26.0

### Adapter firmware version

20220219

### Adapter

SMARTLIGHT CC2652P Zigbee USB Adapter SLZB-02

imuncas, 2022-07-20:
> Sounds similar to this [https://github.com/[/issues/13050](https://github.com/Koenkk/zigbee2mqtt/issues/13050)] #13050 What's the behaviour like on removing the devices?

This is something different. The bulbs and dimmer operate perfctly. Removing, readding the devices also work. The only thing which is not working is binding. If I bind the dimmer to a group, or directly to the light, dimmer actions have no impact on the group members/light. The actions are still reported in logs/MQTT.

MattWestb, 2022-07-20:
HUE dimmers need being unbond (from the coordinator that is being made then paring the device) before using new bindings (can only being bonded to one device / group).

imuncas, 2022-07-20:
Thank you, that was the issue. Now working as expected.

closed 2022-07-20 by imuncas