Koenkk/zigbee2mqtt #13228: [New device support]: Tuya 4CH Zigbee Smart Switch Module with RF433

stefangries, 2022-07-21:
### Link

https://www.aliexpress.com/item/1005004036534972.html

### Database entry

{"id":58,"type":"Router","ieeeAddr":"0xa4c138c1bcead9ee","nwkAddr":21958,"manufId":4417,"manufName":"_TZ3000_excgg5kb","powerSource":"Mains (single phase)","modelId":"TS0004","epList":[1,2,242,3,4],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"57344":{"attributes":{"53251":"AAAAAgAABAAABgAA"}},"genBasic":{"attributes":{"modelId":"TS0004","manufacturerName":"_TZ3000_excgg5kb","powerSource":1,"zclVersion":3,"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":0,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":0,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff0670b0","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff0670b0","endpointID":1}],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":256,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff0670b0","endpointID":1}],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":256,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff0670b0","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1658392060992,"defaultSendRequestWhen":"immediate"}

### Comments

[image]

Currently, the device gets identified as this device:
https://www.zigbee2mqtt.io/devices/TS0004.html

The two implemented devices differ in the possibility to set the power on behaviour per button or per device and the possibility to set the switch type.

As my device has a button to switch the type of the switch (toggle, momentary, only once at a time), maybe this device is more appropriate:
https://www.zigbee2mqtt.io/devices/TS0004_switch_module.html

Additionally, I somentimes got some errors when switching, but I cannot tell if this is a Z2M problem, or a problem with zigbee coverage:
Zigbee2MQTT:error 2022-07-21 09:21:00: Publish 'set' 'state' to 'Bewässerungssteuerung' failed: 'Error: Command `0xa4c138c1bcead9ee/2 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'`

nmaster2042, 2022-07-21:
Same issue here.

stefangries, 2022-07-24:
I have now tested whether the definition of the power on behavior works per channel, or is defined globally for all channels. For my device the individual definition per channel works fine.

evangelo18, 2022-07-25:
I have the TS0004 switch. Whenever an action is triggered on a single switch it signals all 4 switches to turn on or off.

rickzed, 2022-08-03:
> 我有 TS0004 交换机。每当在单个开关上触发动作时，它都会向所有4个开关发出打开或关闭信号。

try this.

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    zigbeeModel: ['TS0004'],
    model: 'TS0004',
    vendor: 'TuYa',
    description: 'Smart light switch - 4 gang with neutral wire CUSTOM',
    extend: extend.switch(),
    exposes: [e.switch().withEndpoint('l1'), e.switch().withEndpoint('l2'), e.switch().withEndpoint('l3'), e.switch().withEndpoint('l4')],
    endpoint: (device) => {
        return {'l1': 1, 'l2': 2, 'l3': 3, 'l4': 4};
    },
    whiteLabel: [{vendor: 'BSEED', model: 'TS0004', description: 'Zigbee switch'}],
    meta: {multiEndpoint: true, disableDefaultResponse: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await endpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(3), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(4), coordinatorEndpoint, ['genOnOff']);
    },
};

module.exports = definition;

evangelo18, 2022-08-22:
Excuse my ignorance, but which file do I add this code to?

Regards,

Evan Pavlakos
m: 0404 276 320

danieledwardgeorgehitchcock, 2022-08-24:
I am working on a full converter for this device as we speak. I have all datapoints probed and am just writing the toZigbee and fromZigbee entries to add the additional functionality to the standard TS0004 converter

closed 2022-10-01 by github-actions[bot]