Koenkk/zigbee2mqtt #13230: Problem join devices

lxeka, 2022-07-21:
### What happened?

I am using a zigbee EFR32MG1 adapter which is part of EZSP integration. I start pairing the  device and it fails to interview. I have tested it with other adapters and works pervectly

### What did you expect to happen?

I expected to pair  the devices normally.

### Zigbee2MQTT version

1.26.0

### Adapter firmware version

6.6.5.0 build 204

### Adapter

EFR32MG1

Koenkk, 2022-07-21:
Can you provide the herdsman debug log of this? (CC @kirovilya )

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

lxeka, 2022-07-22:
This is the output from log when i try to pair the device.
[image]

kirovilya, 2022-07-22:
I don't see any problems in the log.
device joined

lxeka, 2022-07-22:
No, the interview fails. 
[image]

lxeka, 2022-07-22:
[image]
It says that the device can not be configured

kirovilya, 2022-07-22:
I'm seeing a strange error that shouldn't be "Unexpected Packet Sequence". But even then everything goes well. I don’t see where the command is sent and what the device is responding to. Need more log :)

Do you have the ability to update the firmware on the EFR32? at least up to version 6.7.8

lxeka, 2022-07-22:
i am checking with the supplier if i can update the firmware. Meanwhile i am providing the log file where i tried to pair 2 devices and both of them the adapter could found bout the  interview on both of them failed.
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9166197/log.txt)

lxeka, 2022-07-22:
Also on the UI i have this  info for the  device. The device is  Bitron AV2010/32 which is developed from side.
[image]

kirovilya, 2022-07-22:
Your log is short - there is only the start of the adapter and there is no device pairing process.
I also see that the protocol version is 7 - I only supported version 8 of the protocol, so I don’t know how the adapter will behave. I recommend updating the firmware.

Have you used z2m with this stick before?

lxeka, 2022-07-22:
i will try to update the firmware. I am using z2m with sonoff dongle and works perfectly with this device.  Meanwhile i pass to the next zigbee adapter i hava those problems

kirovilya, 2022-07-22:
https://github.com/Koenkk/zigbee-herdsman/issues/319

I think if you update the firmware, then everything will work. EZSP adapter is not yet as popular for Z2M as ZNP :) but are gradually starting to use it.

Hedda, 2022-08-02:
FYI, since EZSP support is still experimental recommend trying to keep discussions to that issue-> https://github.com/Koenkk/zigbee-herdsman/issues/319

PS: Many more other tips on that specific adapter here -> https://community.home-assistant.io/t/itead-zigbee-3-0-usb-dongle-stick-adapter-based-on-silicon-labs-efr32mg21/271144

PPS: ITead have just released a new EFR32MG21 based USB adapter to replace that and solving its HW issues, see -> https://github.com/Koenkk/zigbee2mqtt/discussions/13373

closed 2022-09-09 by github-actions[bot]