Koenkk/zigbee2mqtt #13235: [new device] Heiman HS3AQ-EFA-3.0

asgothian, 2022-07-21:
### Link

https://www.heimantech.com/product/?type=detail&id=106

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0x84fd27fffe07f13a","nwkAddr":15109,"manufId":4619,"manufName":"HEIMAN","powerSource":"DC Source","modelId":"HS3AQ-EFA-3.0","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,1,3,9,10,1026,1029,1037,2821,64641],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"HS3AQ-EFA-3.0","manufacturerName":"HEIMAN","powerSource":4,"zclVersion":3,"appVersion":20,"stackVersion":2,"hwVersion":16,"dateCode":"2021.06.29"}},"genAlarms":{"attributes":{}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2570,"minMeasuredValue":-1000,"maxMeasuredValue":6000,"tolerance":100}},"msRelativeHumidity":{"attributes":{"measuredValue":5504,"tolerance":1000}},"genPowerCfg":{"attributes":{"batteryVoltage":0,"batteryPercentageRemaining":144}},"genTime":{"attributes":{"time":69664,"timeStatus":0,"timeZone":0}},"msCO2":{"attributes":{"measuredValue":0.00039999998989515007,"minMeasuredValue":0,"maxMeasuredValue":7.006492321624085e-42}},"haDiagnostic":{"attributes":{"lastMessageLqi":184,"lastMessageRssi":-54,"averageMacRetryPerApsMessageSent":0}},"heimanSpecificAirQuality":{"attributes":{}}},"binds":[{"cluster":1029,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c0ac6e","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c0ac6e","endpointID":1},{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c0ac6e","endpointID":1},{"cluster":1037,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c0ac6e","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0},{"cluster":1026,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":100},{"cluster":1029,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":100},{"cluster":1037,"attrId":0,"minRepIntval":5,"maxRepIntval":300,"repChange":0.00005}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":20,"stackVersion":2,"hwVersion":16,"dateCode":"2021.06.29","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2086395323},"lastSeen":1658429621659,"defaultSendRequestWhen":"immediate"}

### Comments

Got the device working fine. It lists the HeimanSpecificAirQuality Cluster as supported but does not react to report the values from this cluster and any attempt to read the values returns an "UNSUPPORTED CLUSTER"
The same is true for the haDiagnostic cluster.

On the genTime cluster reading and setting the time (in seconds) is possible, but i did not think this important enough to make an expose for that. Being able to get/set that as device specific configuration should be enough.

### External converter

```shell
const exposes = require('../lib/exposes');
const fz = require('../converters/fromZigbee');
const tz = require('../converters/toZigbee');
const constants = require('../lib/constants');
const reporting = require('../lib/reporting');
const extend = require('../lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('../lib/tuya');

module.exports = [
    {
        zigbeeModel: ['HS3AQ-EFA-3.0'],
        model: 'HS3AQ',
        vendor: 'HEIMAN',
        description: 'Smart Air Quality Monitor',
        fromZigbee: [fz.co2, fz.humidity, fz.battery, fz.temperature, fz.heiman_air_quality],
        toZigbee: [],
        configure: async (device, coordinatorEndpoint, logger) => {
            const heiman = {
                configureReporting: {
                     msCO2: async (endpoint, overrides) => {
                        const payload = reporting.payload('measuredValue', 5, constants.repInterval.MINUTES_5, 0.00005); // 50 ppm change
                        await endpoint.configureReporting('msCO2', payload);
                    }
        
                },
            };

            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['msRelativeHumidity','genPowerCfg', 'msTemperatureMeasurement', 'msCO2']);
            await reporting.batteryPercentageRemaining(endpoint);
            await reporting.temperature(endpoint, {  min:1, max: constants.repInterval.MINUTES_5, change:10 });
            await reporting.humidity(endpoint, { min:1, max: constants.repInterval.MINUTES_5, change:10 });
            await heiman.configureReporting.msCO2(endpoint);
        },
        exposes: [e.co2(), e.battery(), e.humidity(), e.temperature()],
    },

];
```

Koenkk, 2022-07-21:
If everything works, can you please make a PR?

asgothian, 2022-07-21:
Will do, but not tonight.

closed 2022-07-22 by asgothian

SebOM, 2024-07-24:
can you add the time because the module display goes into standby mode at night?

asgothian, 2024-07-24:
What do you mean with "add the time" ? I did try to have it report the time, but the device refused to configure reporting for the time.
What does work for me is to write the current time's "seconds since start of day" to the `time` attribute of the `genTime` to set the time correctly.

when I did the integration I did try to add that as a regular repeating event for the device but could not get that to work as I couldn't find the correct converter for it. 

Since I had a working workaround I gave up after a while.

lgjunk17, 2024-09-07:
Hello
I strugle to expose the "set" genTime.
Is anybody here that can help ?

With my conbee2 on Jeedom (plugin jeezigbee that use zigbee2mqtt), the clock always show a wrong time.

If genTime is exposed the clock could be set on the wrigh time periodicaly

Thanks in advance