Koenkk/zigbee2mqtt #13243: Very quick requests on groups lead to error

pannal, 2022-07-22:
### What happened?

Using `emulated hue` HA integration and the android app `iLightShow` leads to errors on groups `(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)'))`

Using the same combo of app and integration on two separate lights works.

```
zigbee           | Zigbee2MQTT:error 2022-07-22 04:04:49: Publish 'set' 'color' to 'dach_licht_native' failed: 'Error: Command 1 lightingColorCtrl.moveToColor({"transtime":0,"colorx":37424,"colory":12775}) failed (SREQ '--> AF - dataRequestExt - {"dstaddrmode":1,"dstaddr":"0x0000000000000001","destendpoint":255,"dstpanid":0,"srcendpoint":1,"clusterid":768,"transid":159,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[17,43,7,48,146,231,49,0,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)'))'
zigbee           | Zigbee2MQTT:debug 2022-07-22 04:04:49: Error: Command 1 lightingColorCtrl.moveToColor({"transtime":0,"colorx":37424,"colory":12775}) failed (SREQ '--> AF - dataRequestExt - {"dstaddrmode":1,"dstaddr":"0x0000000000000001","destendpoint":255,"dstpanid":0,"srcendpoint":1,"clusterid":768,"transid":159,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[17,43,7,48,146,231,49,0,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)'))
zigbee           | Zigbee2MQTT:info  2022-07-22 04:04:49: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'color' to 'dach_licht_native' failed: 'Error: Command 1 lightingColorCtrl.moveToColor({\"transtime\":0,\"colorx\":37424,\"colory\":12775}) failed (SREQ '--> AF - dataRequestExt - {\"dstaddrmode\":1,\"dstaddr\":\"0x0000000000000001\",\"destendpoint\":255,\"dstpanid\":0,\"srcendpoint\":1,\"clusterid\":768,\"transid\":159,\"options\":0,\"radius\":30,\"len\":9,\"data\":{\"type\":\"Buffer\",\"data\":[17,43,7,48,146,231,49,0,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)'))'","meta":{"friendly_name":"dach_licht_native"},"type":"zigbee_publish_error"}'
```

### What did you expect to happen?

No errors

### How to reproduce it (minimal and precise)

Add two lights to a zigbee group in z2m, add that entity in HA via emulated hue, then address them in iLightShow (iOS/Android).

### Zigbee2MQTT version

 1.26.0

### Adapter firmware version

20220219

### Adapter

zzh!

pannal, 2022-08-22:
Ping

sjorge, 2022-08-23:
BUFFER_FULL is the zigbee adaptor running out of space to queue the incoming and outgoing messages, aside form reducing the amount of messages being send (or perhaps tweaking the firmware memory allocations, but you probably will need to sacrifice other features) there is not a lot that can be done.

Well you could also try and reduce the messages in the mesh, if for example all the lights in the group have reporting enabled you can try and raise the minimalReporting time or disable reporting for those lights altogether.

Also note that when using groups the traffic on the mesh is broadcast  and not unicast. So if it's only a few lights, you could also try sending the commands to them directly.

closed 2022-09-30 by github-actions[bot]

arjanhouben, 2022-11-19:
This is effectively preventing me from running the homebridge/homekit integration, as just opening the homekit screen kills the z2m process.

Even if the zigbee adapter is running out of memory, it is still well behaved, as it returns a sensible message. Is there any reason why z2m couldn't just acknowledge the fact that the device is busy and carry on as best as it can?