Koenkk/zigbee2mqtt #13245: Adurosmart 81848 with powermeasurement cannot be used in Home assistant Energy Dashboard

Madfox8, 2022-07-22:
### What happened?

I've added 2 Adurosmart powermeasurement plugs ([81848](https://www.zigbee2mqtt.io/devices/81848.html#adurosmart-81848)) tot my zigbee2mqtt instance (running on docker koenkk/zigbee2mqtt). The plugs report powermeasurements in Watt, but this entity cannot be used in the Homeassistant dashboard. The dashboard refers the following page in order to use this entity for statistics: [Expected data source not listed](https://www.home-assistant.io/more-info/statistics/).
As far as I understand it the state_class must be properly declared:

> Entities not representing a total amount

Home Assistant tracks the min, max and mean value during the statistics period. The state_class property must be set to measurement, and the device_class must not be either of energy, gas, or monetary< [link](https://developers.home-assistant.io/docs/core/entity/sensor/#long-term-statistics)

### What did you expect to happen?

The Entities could be added to the home assistant dashboard without extra steps

### Zigbee2MQTT version

1.26.0

### Adapter firmware version

 20210708

### Adapter

CC2652

sjorge, 2022-07-22:
`power` is usually the current live consumption and not the consumption over time which is usually labled `energy`.

Looking at the device converts it looks like it only supports `power`, so the message from home assistent seems correct to me.

Madfox8, 2022-07-22:
Indeed the device only supplies power in W - however I expect that home assistant will convert this to energy if the device_class is set correctly, I have not been able to find out how this is set in Z2M for these devices.
[Long term Statistics](https://developers.home-assistant.io/docs/core/entity/sensor/#long-term-statistics)
`Home Assistant has support for storing sensors as long-term statistics if the entity has the right properties. To opt-in for statistics, the sensor must have state_class set to one of the valid state classes: measurement, total or total_increasing. For certain device classes, the unit of the statistics is normalized to for example make it possible to plot several sensors in a single graph.`

sjorge, 2022-07-22:
https://github.com/Koenkk/zigbee2mqtt/blob/cd23aeaa4677efc402115be66e524eb40abe3ce3/lib/extension/homeassistant.ts#L669

I think it's being set to `measurement` but it's also has the class set to energy so I think HA is refusing to track it because of that, probably because tracking that would give too much of an incorrect picture.

Madfox8, 2022-07-22:
Ok, I've been able to create energy entities that calculate kWh based on power readings with an addon: [powercalc](https://github.com/bramstroker/homeassistant-powercalc/blob/master/README.md) to increase accuracy I've increased reporting interval in Z2M from 5 to 1 sec. I thought HA would do this natively but this way seems to work

closed 2022-07-23 by Koenkk