Koenkk/zigbee2mqtt #13246: New Device Suport: Power-Plug Zigbee 20A model 'TS011F' and manufacturer name '_TZ3000_dksbtrzs'

FragMenthor, 2022-07-22:
### Link

https://www.aliexpress.com/item/1005004408927536.html?spm=a2g0o.order_list.0.0.78cacaa4WL6mpF&gatewayAdapt=glo2bra

### Database entry

{"id":38,"type":"Router","ieeeAddr":"0xa4c13818be19a6b4","nwkAddr":5602,"manufId":4417,"manufName":"_TZ3000_dksbtrzs","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":"","zclVersion":3}},"seMetering":{"attributes":{"currentSummDelivered":[0,0]}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":223,"rmsCurrent":0,"activePower":0}},"genOnOff":{"attributes":{"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":2619665,"53251":123,"53252":11944,"53253":1374,"powerOnBehavior":2,"switchType":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":76,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1658492156840,"defaultSendRequestWhen":"immediate"}

### Comments

I tryed on old converter that I had used before, and I can get it to work, but I don't know if all settings are correct, or even if all of them are available. Maybe this is just another TS011F_plug1 like the others and only missing the "vendor"? Very noob in this, sorry.
I "mixed and matched" an old converter that I had for inicial support of Tuya TS011F with the code from "zigbee-herdsman-converters", and got the code below that "works". Not sure if it's correct.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;
const ota = require('zigbee-herdsman-converters/lib/ota');

const definition = {
        fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_dksbtrzs'},],
        model: 'TS011F_plug_1',
        description: 'Smart EU Plug 20A',
        vendor: 'TuYa',
        ota: ota.zigbeeOTA,
        fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report, fz.tuya_switch_power_outage_memory, fz.ts011f_plug_indicator_mode, fz.ts011f_plug_child_mode],
        toZigbee: [tz.on_off, tz.tuya_switch_power_outage_memory, tz.ts011f_plug_indicator_mode, tz.ts011f_plug_child_mode],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await endpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
            await reporting.rmsVoltage(endpoint, {change: 5});
            await reporting.rmsCurrent(endpoint, {change: 50});
            await reporting.activePower(endpoint, {change: 10});
            await reporting.currentSummDelivered(endpoint);
            endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
            endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
            device.save();
        },
        exposes: [e.switch(), e.power(), e.current(), e.voltage().withAccess(ea.STATE), 
            e.energy(), exposes.enum('power_outage_memory', ea.ALL, ['on', 'off', 'restore'])
                .withDescription('Recover state after power outage'),
            exposes.enum('indicator_mode', ea.ALL, ['off', 'off/on', 'on/off', 'on'])
                .withDescription('Plug LED indicator mode'), e.child_lock()],
};

module.exports = definition;
```

FragMenthor, 2022-07-22:
I get these settings exposed:
[image]

But, for instance, when I call to update the "power_outage memory" status I get an error:
[image]

Thanks in advance!

malosaa, 2022-07-22:
These sockets have an issue, buy the round ones of the 20a 
They do turn off power, even after the update.

I did test them and have an issue, so i sended them back.

Maybe this is from an other fabric, so try to go to OTA and update trough Z2M and see if it works then.

regards

FragMenthor, 2022-07-22:
> @FragMenthor
> 
> These sockets have an issue, buy the round ones of the 20a They do turn off power, even after the update. That power_outage_memory problem, is also an issue,
> 
> I did test them and have an issue, so i sended them back.
> 
> Maybe this is from an other fabric, so try to go to OTA and update trough Z2M and see if it works then.
> 
> regards

I left home with absolutly no "exposes" on the device, and "not supported" status. With this converter I got it "working" in Z2M in HA remotely, but I can't confirm if it responding correctly, I left it connected but with no device attached. When I get home I will test all the settings and post some feedback.
I added the 2 lines to allow OTA, and it reports an update available, but since I don't know if all the settings are correct (the OTA also), maybe i should stand put until someone with more expertise reviews this. PS: I got 3 of these, for about 6€ each... They where cheap, but I intend to make them work :) Thanks!!!

malosaa, 2022-07-22:
U can just OTA them.
Like your screenshot shows, there are no other settings for it.

I have many of these plugs, better update them first.

The not supported status somethimes happens when it did not pair correctly, just press the orange button and it will repair again.
I had this with 2 of my plugs, it will eventually pair and say supported.

[image]

FragMenthor, 2022-07-22:
> The not supported status somethimes happens when it did not pair correctly, just press the orange button and it will repair again.
> I had this with 2 of my plugs, it will eventually pair and say supported.

I think the "not supported" status was only due to the manufacturer id being "new", it did pair correctly the first time, but only got it recognised with the external converted that I posted, that contains that manufacturer id; the rest I just copied. The only thing I see different than yours is that you don't have a "power_outage_memory" update button (the blue one with arrows), maybe it is not supposed to exist there, but "my" converter exposed it. On a note, at least remotly, it seems to turn on/off correctly, at least reports the state change and I "think" (not being there) that is physically toggling the switch. Only later can confirm.

malosaa, 2022-07-22:
Hi,

the blue one with arrows only shows up when u change that setting. 
Thats why its not in my screenshot.

regards

FragMenthor, 2022-07-25:
Another note: This plug behaves a little strange with the converter I posted, don't know if it's related, but I get, every 3-4 minutes, a power consuption value of 0W, when I know the device is consuming. Don't know if it's converter-related of bad device funcionality. I did the OTA update, also don't know if it was the better option. Still have 2 more to test, will report as soon as possible.

[image]

[image]

Anyone else with this behavior?
Thanks in advance!

jenshorn, 2022-08-07:
> Another note: This plug behaves a little strange with the converter I posted, don't know if it's related, but I get, every 3-4 minutes, a power consuption value of 0W, when I know the device is [...]
I have the same type of plug (dksbtrzs) and been trying a few different settings for the converter but ended up with the same as your initial post. This is the first time setting up a converter so I'll just go by trial and error. Of the two plugs I have tested so far one powers off every now and then and it also has the same behavior with reporting 0W/0A on a regular interval, although it's powering a laptop so should be under constant load. The other plug is for the coffee machine which is expected to report periods of 0W and so far I haven't had any issues with it turning off. Both have AppVersion: 76 from database.db and no update currently available.

malosaa, 2022-08-07:
I hve the same issue @FragMenthor  @jenshorn 
What you have to do is go to reporting and change the settings.

I changed it and now it doesnt show up the 0w anymore 
Before i get every update a 0 power value, not anymore :) 

```
 2022-08-07 20:45:30MQTT publish: topic 'zigbee2mqtt/c', payload '{"child_lock":"LOCK","current":0.6,"energy":9.32,"indicator_mode":"off/on","linkquality":18,"power":119,"power_outage_memory":"on","state":"ON","update":{"state":"idle"},"update_available":null,"voltage":218}'
Info 2022-08-07 20:45:54MQTT publish: topic 'zigbee2mqtt/c', payload '{"child_lock":"LOCK","current":0.6,"energy":9.32,"indicator_mode":"off/on","linkquality":18,"power":120,"power_outage_memory":"on","state":"ON","update":{"state":"idle"},"update_available":null,"voltage":218}'
Info 2022-08-07 20:46:24MQTT publish: topic 'zigbee2mqtt/c', payload '{"child_lock":"LOCK","current":0.6,"energy":9.32,"indicator_mode":"off/on","linkquality":18,"power":119,"power_outage_memory":"on","state":"ON","update":{"state":"idle"},"update_available":null,"voltage":218}'
Info 2022-08-07 20:46:29MQTT publish: topic 'zigbee2mqtt/c', payload '{"child_lock":"LOCK","current":0.6,"energy":9.33,"indicator_mode":"off/on","linkquality":14,"power":119,"power_outage_memory":"on","state":"ON","update":{"state":"idle"},"update_available":null,"voltage":218}'
Info 2022-08-07 20:46:33MQTT publish: topic 'zigbee2mqtt/c', payload '{"child_lock":"LOCK","current":0.6,"energy":9.33,"indicator_mode":"off/on","linkquality":21,"power":119,"power_outage_memory":"on","state":"ON","update":{"state":"idle"},"update_available":null,"voltage":214}'
Info 2022-08-07 20:46:35MQTT publish: topic 'zigbee2mqtt/c', payload '{"child_lock":"LOCK","current":0.6,"energy":9.33,"indicator_mode":"off/on","linkquality":21,"power":117,"power_outage_memory":"on","state":"ON","update":{"state":"idle"},"update_available":null,"voltage":214}'
Info 2022-08-07 20:46:38MQTT publish: topic 'zigbee2mqtt/c', payload '{"child_lock":"LOCK","current":0.6,"energy":9.33,"indicator_mode":"off/on","linkquality":18,"power":117,"power_outage_memory":"on","state":"ON","update":{"state":"idle"},"update_available":null,"voltage":219}'
Info 2022-08-07 20:46:42MQTT publish: topic 'zigbee2mqtt/c', payload '{"child_lock":"LOCK","current":0.6,"energy":9.33,"indicator_mode":"off/on","linkquality":18,"power":119,"power_outage_memory":"on","state":"ON","update":{"state":"idle"},"update_available":null,"voltage":219}'
Info 2022-08-07 20:46:54MQTT publish: topic 'zigbee2mqtt/c', payload '{"child_lock":"LOCK","current":0.6,"energy":9.33,"indicator_mode":"off/on","linkquality":21,"power":123,"power_outage_memory":"on","state":"ON","update":{"state":"idle"},"update_available":null,"voltage":219}'
Info 2022-08-07 20:47:24MQTT publish: topic 'zigbee2mqtt/c', payload '{"child_lock":"LOCK","current":0.6,"energy":9.33,"indicator_mode":"off/on","linkquality":18,"power":122,"power_outage_memory":"on","state":"ON","update":{"state":"idle"},"update_available":null,"voltage":219}'
```

[image]

jenshorn, 2022-08-07:
> I hve the same issue @FragMenthor @jenshorn What you have to do is go to reporting and change the settings.
> 
> I changed it and now it doesnt show up the 0w anymore Before i get every update a 0 power value, not anymore :)
> 
I see that you have changed rmsVoltage and activePower to 1 instead of 5 and 10 respectively as in the initial post, are both needed to change to get rid of the 0W as far as you can tell?

malosaa, 2022-08-10:
> I see that you have changed rmsVoltage and activePower to 1 instead of 5 and 10 respectively as in the initial post, are both needed to change to get rid of the 0W as far as you can tell?

Well it decreases a lot.
I did today had a pworbel where my server went to hot and my whole system went off.
So i restarted everything later and did get 2x a 0 power and after that nothing anymore.
So i think it will be there but like 90% less than before.

So if u set it up like my screenshot it should work, but u can ofcourse try different combinations and look at the logs and see what happens.

regards

jenshorn, 2022-08-11:
I tried a few different settings but it didn't make a huge difference, I still get the occasional drops to 0w.
[image]
Regarding the firmware, I did an update from AppVersion 76 to 77 and although I still get the occasional drop to 0**V** at least now the plug doesn't turn off anymore. I've ordered some additional round Tuya 20A Zigbee plugs to compare with, hopefully they are a bit more stable until the firmware of these have improved. Or perhaps I'll just return these _dksbtrzs_.

malosaa, 2022-08-11:
I did order them all, all plugs available on Aliexpress, the ones i posted they work, but have that 0W issue (firmware)

I did see somewhere on this board that a guy is talking with the factory to get a firmware out.
Lets wait for that.

malosaa, 2022-08-11:
Well today i gonna get in contact with the factory, i will let u guys know when there will be a possible update for these plugs.
I let you guys know, 
Don't send them back yet i try to get an answer within a week.

regards

phalox, 2022-08-14:
Here another one with these plugs... Looking forward to more updates.

Also, noob question: how do you OTA update these without the tuya stuff?

malosaa, 2022-08-16:
> Here another one with these plugs... Looking forward to more updates.
> 
> Also, noob question: how do you OTA update these without the tuya stuff?

Hi,

Here check my screenshot,

regards 

[image]

gyrex, 2022-08-20:
I have the same issue. I bought Australian plugs from a seller on Aliexpress: https://trade.aliexpress.com/order_detail.htm?orderId=8153763973618261

Current and voltage is being reported but power usage is 0W despite there being 0.13A of current being measured @ 240V.

Have there been any updates to this issue from the manufacturer?

malosaa, 2022-08-22:
> I have the same issue. I bought Australian plugs from a seller on Aliexpress: https://trade.aliexpress.com/order_detail.htm?orderId=8153763973618261
>
> Current and voltage is being reported but power [...]

Hi

The link doesnt work, 
as if they are aubess plugs i am already in contact with them, the company told me they gonna try to let the factory deploy an update, but the thing is he thinks they wont help us as there need to be more people reporting this issue because "its a big project" the company says, but i'm on it and let everyone know if i hear something. 

I gonna order 8 more today, so i hope money speeds up the process..

gyrex, 2022-08-22:
> Hi
>
> The link doesnt work, as if they are aubess plugs i am already in contact with them, the company told me they gonna try to let the factory deploy an update, but the thing is he thinks they wont [...]

Apologies, here is the link: https://www.aliexpress.com/item/1005004505864316.html

Here's the entry from database.db too:

`{"id":6,"type":"Router","ieeeAddr":"0xa4c138cf5e6f6532","nwkAddr":37648,"manufId":4417,"manufName":"_TZ3000_typdpbpg","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57345,57344,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�Ւ*\u0003","65506":54,"65508":0,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_typdpbpg","powerSource":1,"zclVersion":3,"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":239,"rmsCurrent":137,"activePower":0}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,2]}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":0,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":1,"53249":0,"53250":3850787,"53251":124,"53252":7366,"53253":934,"powerOnBehavior":0,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c6d1b6","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c6d1b6","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c6d1b6","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1798312590},"lastSeen":1660971938850,"defaultSendRequestWhen":"immediate"}
`

malosaa, 2022-08-22:
> Apologies, here is the link: https://www.aliexpress.com/item/1005004505864316.html
>
> Here's the entry from database.db too: [...]

I see its different, i use these (in europe), but i think they all have same firmware.
https://nl.aliexpress.com/item/1005004124722011.html

But what if u go to zigbee2mqtt and tab on logs and filter out your device and wait a bit.
Can you post that here like i did above, so i can see how it reports.

regards

gyrex, 2022-08-22:
Here's the log:

`Debug 2022-08-23 00:16:09Received Zigbee message from 'Power01', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":77}' from endpoint 1 with groupID 0
Debug 2022-08-23 00:18:32Received Zigbee message from 'Power01', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,20]}' from endpoint 1 with groupID 0
Info 2022-08-23 00:18:32MQTT publish: topic 'zigbee2mqtt/Power01', payload '{"child_lock":"UNLOCK","current":0.13,"energy":0.2,"indicator_mode":"off/on","linkquality":196,"power":0,"power_outage_memory":"off","state":"ON","update":{"state":"idle"},"update_available":false,"voltage":233}'
Debug 2022-08-23 00:18:32Received Zigbee message from 'Power01', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":4,"rmsCurrent":127,"rmsVoltage":235}' from endpoint 1 with groupID 0
Info 2022-08-23 00:18:32MQTT publish: topic 'zigbee2mqtt/Power01', payload '{"child_lock":"UNLOCK","current":0.13,"energy":0.2,"indicator_mode":"off/on","linkquality":196,"power":4,"power_outage_memory":"off","state":"ON","update":{"state":"idle"},"update_available":false,"voltage":235}'`

malosaa, 2022-08-22:
@gyrex 

you have indeed the same firmware issue, so all those devices have the same firmware.

try this :
https://github.com/Koenkk/zigbee2mqtt/issues/13246#issuecomment-1207464798

I did set it up like this and the 0 power usage only came once a 8 or 20 minutes instead every update the device does, atleast for my devices.

The rest we need to wait for a firmware update..

malosaa, 2022-08-22:
well thinking about the update, i think we need to ask tuya about it, they release the firmware for the devices..
thats what the aliexpress store said..
SO if we can get in touch with the turya firmware developers they should be able to fix that.

gyrex, 2022-08-22:
> @gyrex
> 
> you have indeed the same firmware issue, so all those devices have the same firmware.
> 
> try this : [#13246 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/13246#issuecomment-1207464798)
> 
> I did set it up like this and the 0 power usage only came once a 8 or 20 minutes instead every update the device does, atleast for my devices.
> 
> The rest we need to wait for a firmware update..

I tried this but it didn't fix anything unfortunately. Power was still being reported as 0W

jenshorn, 2022-08-23:
I have opened a support ticket at https://iot.tuya.com/ but so far the response I have received is that they need a Device ID from a Tuya Smart Gateway to start the investigation. I do not have any such gateway and I have asked them if they don't support other bridges/gateway. And even if I did have such gateway they would probably not have any success as all my Wi-Fi IoT devices are on an isolated VLAN without internet access... I'll keep you posted in case there are any updates.

gyrex, 2022-08-23:
> I have opened a support ticket at https://iot.tuya.com/ but so far the response I have received is that they need a Device ID from a Tuya Smart Gateway to start the investigation. I do not have any [...]

Interesting technical support response. It doesn't matter what gateway you're running, if the device is reporting 0W, that's the end of the story.

jenshorn, 2022-08-24:
It appears that the developers at Tuya can't do anything without a device ID and the only way to have such ID is to connect the plug to a Tuya Smart Gateway. This is the response I have received so far:
> Hello, I hope to be able to help you; but you are not connected to Tuya Smart Gateway, and I cannot provide any help without the device ID; if you can provide a ZigBee device of the same model and provide the device ID; the device will have a device ID after it is added to the Tuya gateway; this device ID is the unique identifier of the Tuya smart module. According to the device ID, we can query related issues, and only push the upgrade according to the device ID.

1. I don't own that kind of gateway. Does anyone here have a Tuya gateway that we could use to get a device ID for one of the _TZ3000_dksbtrzs plugs?
2. I'm not so sure that this problem exists when connected to a Tuya gateway? I think I read somewhere about some other plug that had issues when connected to another bridge but those issues disappeared when connected to Tuyas own gateway... But I might be totally wrong about this.

phalox, 2022-09-09:
Anyone has an update? I also don't own a gateway, and I'd like to start using these plugs in HA. But if they're not reliable, I might need to buy other ones...

martinkasa, 2022-09-10:
I am using them with this external script. But I would appreciate to have it supported out of box.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");
const ota = require('zigbee-herdsman-converters/lib/ota');

const transactionStore = {};

const hasAlreadyProcessedMessage = (msg, transaction=null, key=null) => {
    const current = transaction !== null ? transaction : msg.meta.zclTransactionSequenceNumber;
    key = key || msg.device.ieeeAddr;
    if (transactionStore[key] === current) return true;
    transactionStore[key] = current;
    return false;
};

const metering_skip_duplicate = {
        ...fz.metering,
        convert: (model, msg, publish, options, meta) => {
            if (hasAlreadyProcessedMessage(msg)) return;
            return fz.metering.convert(model, msg, publish, options, meta);
        },
    };

const electrical_measurement_skip_duplicate = {
        ...fz.electrical_measurement,
        convert: (model, msg, publish, options, meta) => {
            if (hasAlreadyProcessedMessage(msg)) return;
            return fz.electrical_measurement.convert(model, msg, publish, options, meta);
        },
    };

const definition = {
    fingerprint: [
        {
            modelID: 'TS011F',
            manufacturerName: '_TZ3000_dksbtrzs'
        },
    ],
    model: 'TS011F',
    vendor: '_TZ3000_dksbtrzs', 
    description: 'Zigbee plug',
    ota: ota.zigbeeOTA,
    fromZigbee: [fz.on_off, electrical_measurement_skip_duplicate, metering_skip_duplicate, fz.ignore_basic_report,
            fz.tuya_switch_power_outage_memory, fz.ts011f_plug_indicator_mode, fz.ts011f_plug_child_mode],
        toZigbee: [tz.on_off, tz.tuya_switch_power_outage_memory, tz.ts011f_plug_indicator_mode, tz.ts011f_plug_child_mode],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await endpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
            await reporting.rmsVoltage(endpoint, {change: 5});
            await reporting.rmsCurrent(endpoint, {change: 50});
            await reporting.activePower(endpoint, {change: 10});
            await reporting.currentSummDelivered(endpoint);
            endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
            endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
            device.save();
        },
        exposes: [e.switch(), e.power(), e.current(), e.voltage().withAccess(ea.STATE),
            e.energy(), exposes.enum('power_outage_memory', ea.ALL, ['on', 'off', 'restore'])
                .withDescription('Recover state after power outage'),
            exposes.enum('indicator_mode', ea.ALL, ['off', 'off/on', 'on/off', 'on'])
                .withDescription('Plug LED indicator mode'), e.child_lock()],
};

module.exports = definition;
```

FragMenthor, 2022-09-14:
> I am using them with this external script. But I would appreciate to have it supported out of box.
>
> ```
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = [...]

Hello! Did this solved for you the 0W recurring problem?

ngosang, 2022-09-16:
I bough this device on 10/09/2022 from Aliexpress and it's working fine after using the template from the first message.
* The device is up-to-date according to the OTA information in Zigbee2Mqtt
* All metrics from power meter are working
* Power switch and Child lock are working
* Power state restore is working
* Led color on/off is working
* The SmartPlug is detected as Zigbee Router :)
* All of it if working well from Home Assistant

All is working fine and the device only costs 8€ with shipping costs. Could you add it to supported devices so I don't have to keep a separated file?
Thank you!

martinkasa, 2022-09-24:
@FragMenthor 0W problem is not there, nice straight line.

gyrex, 2022-09-25:
> @FragMenthor 0W problem is not there, nice straight line.

Where do you run the script from?

martinkasa, 2022-09-25:
@gyrex 
These were my steps:
1. enable ssh by addon
2. login to HA by ssh and create file /etc/config/zigbee2mqtt/TS011F.js with content I posted in this thread above.
3. in zigbee2mqtt in settings there is External scripts tab. There I just added the filename 'TS011F.js'
4. restart zigbee2mqtt

FragMenthor, 2022-09-25:
> @FragMenthor 0W problem is not there, nice straight line.

Probably not exactly the same hardware, as this issue for many still remains. Good for you :) ;)

malosaa, 2022-09-28:
> 

I only see external converters tab. 
Is that the correct one ?

FragMenthor, 2022-09-28:
> > 
> 
> @martinkasa I only see external converters tab. Is that the correct one ?

yes!

phalox, 2022-10-10:
Did an OTA update today, and I can clearly see the 0w drops. So far it didn't turn off yet, but I'll test that more tomorrow.

[image]

[image]

Nobody has an update from Tuya?

malosaa, 2022-10-11:
> 

Sadly the 0 watt problem still there

phalox, 2022-11-14:
So, my plugs don't seem to turn off, but only report 0w once in a while. I'm currently not using the active consumption, but I do keep track of total consumption which seems to work.

Anyway, there's still something wrong with the firmware. it would be great if Tuya would fix it.

If anyone has a tuya gateway, let us know!

malosaa, 2022-12-19:
But does someone have tested these plugs with the tuya gateway and does it report 0 watt there ?

phalox, 2022-12-19:
We didn't find anyone with a gateway. But I don't think that's the problem (unless in the gateway they have some kind of smart filter to take out these problem values).

My plugs don't reset, but I do see the 0w on a regular basis. I'm not using the power measurement, I'm only looking at the overall power consumption which doesn't go to 0 (but probably it's not very accurate).

Tuya only wants to debug on their gateway (I assume they can log in) to check what's up. So if you know someone with a tuya gateway and plugs.... we might be able to get a fixed firmware for these plugs.

malosaa, 2022-12-19:
I have a gateway, i will see if i can do this, as all my plugs are connected to stuff i can't disconnect. 
I will see what i can do to test.

regards

ngosang, 2022-12-19:
I see the drops to 0 w but the plug is working well overall. The instant power is correct (when it's not 0).
[image]

malosaa, 2022-12-19:
Thats great, but if it doesnt report 0 on the tuya app with the tuya gateway, than its a z2m issue, we will see.. i let you guys know if i can test and see what plug i gonna use first.

regards

malosaa, 2022-12-19:
Actually i have a plug, but someone had accidentally water over it and it started to making noises.. so i put it to dry for a week, dunno if it will work . if this one works again ill test this one out. 
But kinda scare to test.

if not ill buy new ones and test in tuya.

regards

Tontze, 2022-12-22:
Hello ! I have few of these [AUBESS](https://www.aliexpress.com/item/1005004945143502.html?spm=a2g0o.new_account_index.0.0.725f25b9ddluIr) plugs too.

The problem is that i dont see anything in herdsman ? I can pair the plugs in tuya hub and they do work, no FW updates available. 
I was looking forward to measure some consumptions in HASS .. 

What should i do to make them visible to z2m ?

malosaa, 2022-12-23:
> Hello ! I have few of these [AUBESS](https://www.aliexpress.com/item/1005004945143502.html?spm=a2g0o.new_account_index.0.0.725f25b9ddluIr) plugs too.

DO you have these plugs paired in the tuya app with the tuya gateway?
If so, can you please check if they report a 0 watt usage after every update ?
Because no one of us could test that.

Yes i can help you updating thise aubess plugs, thats the one i have.

regards.

closed 2023-02-01 by github-actions[bot]

ngosang, 2023-02-10:
The solution posted in the first comment is not working in zigbee2mqtt **1.30.1**
This is the updated external_converter:
```js
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
        fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_dksbtrzs'},],
        model: 'TS011F_plug_1',
        description: 'Smart EU Plug 20A',
        vendor: 'TuYa',
        ota: ota.zigbeeOTA,
        extend: tuya.extend.switch({electricalMeasurements: true, powerOutageMemory: true, indicatorMode: true, childLock: true}),
        configure: async (device, coordinatorEndpoint, logger) => {
            await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
            await reporting.rmsVoltage(endpoint, {change: 5});
            await reporting.rmsCurrent(endpoint, {change: 50});
            await reporting.activePower(endpoint, {change: 10});
            await reporting.currentSummDelivered(endpoint);
            endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
            endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
            device.save();
        },
};

module.exports = definition;

```

gyrex, 2023-02-28:
My 'TS011F_plug_1' power monitor appears to be working now and reporting correct power usage. Is everyone else experiencing the same? What's changed?

[image]

malosaa, 2023-03-09:
It would be great if someone can pair a socket to a tuya bridge and see if the 0 is also reporting in there, if now we maybe can fix this in z2m

iculookn, 2023-03-09:
> It would be great if someone can pair a socket on a tuya bridge and see if the 0 is also reporting in there, if not we maybe can fix this in z2m 

I have a _TZ3000_r6buo8ba connected to a Tuya hub and these are the only screens I get. Not sure if this helps

[image]

[image]

FragMenthor, 2023-03-10:
> > It would be great if someone can pair a socket on a tuya bridge and see if the 0 is also reporting in there, if not we maybe can fix this in z2m
> 
> I have a _TZ3000_r6buo8ba connected to a Tuya hub and these are the only screens I get. Not sure if this helps
> 
> [image]
> 
> [image]

You could closelly monitor your Power (W) and Voltage (V) on the first screen, while the device is consuming energy, and for about 10 minutes, to see if in any moment the values drop to 0 unexpectedly. In my case, the drop occours every 4 and a half minutes, more or less. Should not take the eyes of the values, because it's only for a moment.

[image]

Thanks!

FragMenthor, 2023-03-10:
Althought it's not quite the same model.

malosaa, 2023-03-18:
> Althought it's not quite the same model.

thanks fot the info, now we know the drop takes place also on tuya, so it's firmware related.