Koenkk/zigbee2mqtt #13251: [New device support]: TS011F _TZ3000_aim0ztek

Gugulator, 2022-07-23:
### Link

https://he.aliexpress.com/item/1005002679735460.html

### Database entry

not found it. newbie to HA instance on VM

### Comments

This is switch for boiler to heating water, with power monitoring option.
Bought from LoraTap official store at AliExpress.
Works fine with Tuya/SmartHome systems.
Problem to complete setup in HA due to unsupported model/manufacturer.

Device type: Router
Zigbee Model: TS011F
Zigbee Manufacturer: _TZ3000_aim0ztek
IEEE Address: 0xa4c1384690f973a1
Network address: 0x9281

Gugulator, 2022-08-20:
still not fixed in any system update...

danieledwardgeorgehitchcock, 2022-09-06:
I'll have a look at this tomorrow. Can't promise anything though

closed 2022-10-15 by github-actions[bot]

morhimi, 2023-06-25:
any chance this will be supported? right now the device is identified is a Plug