Koenkk/zigbee2mqtt #13252: [New device support]: Aqara E1 Water Leak Sensor SJCGQ13LM

1000010000, 2022-07-23:
### Link

https://www.tomtop.com/p-pas0916w.html

### Database entry

lumi.flood.acn001

### Comments

I added an external converter by referring to this [issue](https://github.com/Koenkk/zigbee2mqtt/issues/11285) and [SJCGQ12LM](https://github.com/Koenkk/zigbee-herdsman-converters/blob/3a581af48ba7d97444fdc71f6e384a9f34fb8bb3/devices/xiaomi.js#L964), but the system can't get any attributes, or even get the device battery.
When I try to find this problem on Google, I see a similar [problem.](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5725) It seems that this device is different from SJCGQ12LM.
[image]
[image]

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.flood.acn001'],
    model: 'SJCGQ13LM',
    vendor: 'Xiaomi',
    description: 'Aqara E1 water leak sensor',
    meta: {battery: {voltageToPercentage: '3V_2850_3200'}},
    fromZigbee: [fz.xiaomi_basic, fz.ias_water_leak_alarm_1],
    toZigbee: [],
    exposes: [e.battery(), e.water_leak(), e.battery_low(), e.battery_voltage()],
    ota: ota.zigbeeOTA,
};

module.exports = definition;
```

Otnow, 2022-07-23:
Tell me what version of the coordinator firmware do you have?

1000010000, 2022-07-23:
> Tell me what version of the coordinator firmware do you have?
Is my version too old？
[image]

Otnow, 2022-07-23:
> Is my version too old？

Yes, you need to [update](https://www.zigbee2mqtt.io/devices/MCCGQ14LM.html#adapter-firmware) the firmware of the coordinator, then re-pair the sensor and try to use this external converter:

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.flood.acn001'],
    model: 'SJCGQ13LM',
    vendor: 'Xiaomi',
    description: 'Aqara E1 water leak sensor',
    fromZigbee: [fz.ias_water_leak_alarm_1, fz.aqara_opple, fz.battery],
    toZigbee: [],
    exposes: [e.water_leak(), e.battery_low(), e.battery(), e.battery_voltage(), e.device_temperature(), e.power_outage_count(false)],
    meta: {battery: {voltageToPercentage: '3V_2850_3200'}},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await endpoint.read('genPowerCfg', ['batteryVoltage']);
    },
    ota: ota.zigbeeOTA,
};

module.exports = definition;
```

1000010000, 2022-07-24:
> > Is my version too old？
> 
> Yes, you need to [update](https://www.zigbee2mqtt.io/devices/MCCGQ14LM.html#adapter-firmware) the firmware of the coordinator, then re-pair the sensor and try to use this external converter:
> 
> ```js
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const ota = require('zigbee-herdsman-converters/lib/ota');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const definition = {
>     zigbeeModel: ['lumi.flood.acn001'],
>     model: 'SJCGQ13LM',
>     vendor: 'Xiaomi',
>     description: 'Aqara E1 water leak sensor',
>     fromZigbee: [fz.ias_water_leak_alarm_1, fz.aqara_opple, fz.battery],
>     toZigbee: [],
>     exposes: [e.water_leak(), e.battery_low(), e.battery(), e.battery_voltage(), e.device_temperature(), e.power_outage_count(false)],
>     meta: {battery: {voltageToPercentage: '3V_2850_3200'}},
>     configure: async (device, coordinatorEndpoint, logger) => {
>         const endpoint = device.getEndpoint(1);
>         await endpoint.read('genPowerCfg', ['batteryVoltage']);
>     },
>     ota: ota.zigbeeOTA,
> };
> 
> module.exports = definition;
> ```

I updated my coordinator firmware to 20220219. After using this external converter, my sensor works normally,this is awesome,Thank you very much for your help! but now there are two less important problems:

1. OTA doesn't work properly, of course, it doesn't matter.
2. The status of battery is a little strange. When I first connected, only 14% of battery was left. After a period of time, battery became 30%. This is a brand new sensor. I may need to buy a new battery to verify this problem.

Otnow, 2022-07-24:
> I updated my coordinator firmware to 20220219. After using this external converter, my sensor works normally,this is awesome,Thank you very much for your help!

Fine 👍

---

>     1. OTA doesn't work properly, of course, it doesn't matter.

Can you clarify what exactly is wrong with OTA?

---

>     2. The status of battery is a little strange. When I first connected, only 14% of battery was left. After a period of time, battery became 30%. This is a brand new sensor. I may need to buy a new battery to verify this problem.

Please provide a [debug log](https://www.zigbee2mqtt.io/guide/usage/debug.html#zigbee2mqtt-debug-logging) (`log_level: debug`), which will contain the following actions:

- sensor pairing
- press the pairing button once (heartbeat-message (`{"247":{"data":...`) from the sensor should appear in the log)
- close and hold in this state the sensor contacts (simulating a leak)
- press the pairing button once (heartbeat-message (`{"247":{"data":...`) from the sensor should appear in the log)
- open the sensor contacts
- press the pairing button once (heartbeat-message (`{"247":{"data":...`) from the sensor should appear in the log)

1000010000, 2022-07-24:
> Can you clarify what exactly is wrong with OTA?

When I click the update button of Ota, an error message appears. I see the following information in the log:
Error:failed to check if update available for '浸水传感器' (Device didn't respond to OTA request）

**debug log**
[image]

Otnow, 2022-07-24:
> When I click the update button of Ota, an error message appears. I see the following information in the log:
> Error:failed to check if update available for '浸水传感器' (Device didn't respond to OTA request）

Because the sensor is battery powered, then immediately after the command you need to press the pairing button so that it wakes up and accepts the command.

---

> The battery power has been changing from 14% of the initial value to 34% and now back to 25%.
> **debug log**

Unfortunately, this log does not contain the necessary information.

Let's do it a little differently and split it into 2 debug logs:

- Log 1:
  - sensor pairing
    - starts with a message: `Device '0x...' joined`
    - ends with the message: `Successfully configured '0x...'`

Because this sensor apparently belongs to those that for some time after the pairing do not report a heartbeat-message (`{"247":{"data":...`) when the pairing button is pressed, then I suggest waiting after the moment of pairing for about 1-2 hours and then write the second debug log:

- Log 2:
  - press the pairing button once (heartbeat-message (`{"247":{"data":...`) from the sensor should appear in the log)
  - close and hold in this state the sensor contacts (simulating a leak)
  - press the pairing button once (heartbeat-message (`{"247":{"data":...`) from the sensor should appear in the log)
  - open the sensor contacts
  - press the pairing button once (heartbeat-message (`{"247":{"data":...`) from the sensor should appear in the log)

1000010000, 2022-07-24:
> Because the sensor is battery powered, then immediately after the command you need to press the pairing button so that it wakes up and accepts the command.

Error :Failed to check if update available for '浸水传感器' (No image available for imageType '10763')

I feel like I should get a valid log this time.
Received Zigbee message from '浸水传感器', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[1,33,89,11,3,40,25,4,33,0,0,5,33,1,0,8,33,8,0,10,33,50,237,12,32,30,100,16,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
[image]

Otnow, 2022-07-24:
> Error :Failed to check if update available for '浸水传感器' (No image available for imageType '10763')

This means that the OTA check worked and there is currently no update file available.

---

> I feel like I should get a valid log this time.

Not really, because we need to see the heartbeat-message during the period when the sensor contacts are closed (simulating a leak):

- close and hold in this state the sensor contacts (simulating a leak)
- press the pairing button once and wait for the heartbeat-message (`{"247":{"data":...`) to appear in the log 

I also see message `{"252":0}` appearing in the log and maybe it periodically reports the status of the leak, so let's try to check this when the sensor contacts are closed (simulating a leak):

- close and hold in this state the sensor contacts (simulating a leak)
- wait for the message `{"252":...}` to appear in the log (the value is expected to be 1)

Otnow, 2022-07-25:
> data'{"247":{"data":[1,33,115,11,3,40,25,4,33,0,0,5,33,1,0,8,33,8,0,10,33,47,247,12,32,30,100,16,1],"type":"Buffer"}}'
> 
> I feel like I understand a little bit, when I keep the leak state, two numbers in this set of numbers change, the third digit of the change should be the signal strength, the last digit should be the leak state, and 1 is Leakage, 0 means no leakage, so the 100 in the third-to-last digit should be the value of the battery.

This is the log we needed. In this heartbeat-message:

- _attribute 1_ (`1,33,115,11`) is the battery voltage
- _attribute 100_ (`100,16,1`) is the leak status

Judging by the voltage of the battery, it is not new and is already partially discharged, but the percentage of charge displayed at the moment is approximate, because until now, the formula for its calculation is the subject of [discussion](https://github.com/Koenkk/zigbee2mqtt/issues/8499).

---

Were you able to log message `{"252":...}`?:

> I also see message `{"252":0}` appearing in the log and maybe it periodically reports the status of the leak, so let's try to check this when the sensor contacts are closed (simulating a leak): [...]

1000010000, 2022-07-25:
No matter how I try, the value of {"252":...} will not change to 1. Only when I long press the pairing button will {"252": 0} appear. When I simulate the water leakage state, the value of type'commandstatuschangenotification', cluster'ssiaszone', data '{"extendedstatus": 0, "zonestatus": 256}' has changed. When I simulate the water leakage state, and short press, long press, {"252":...} will not appear again

[image]

Otnow, 2022-07-25:
> No matter how I try, the value of {"252":...} will not change to 1. Only when I long press the pairing button will {"252": 0} appear. When I simulate the water leakage state, the value of [...]

Ok, so message `{"252":...}` is not related to the leak status and means something else.

---

> And I don't quite understand why the battery voltage has been fluctuating significantly.
> 
> [image]

In this log, voltage fluctuations are small and this is normal:

- `1,33,63,11` - it's _2879 mV_
- `1,33,70,11` - it's _2886 mV_
- `1,33,64,11` - it's _2881 mV_
- `1,33,59,11` - it's _2875 mV_
- `1,33,66,11` - it's _2882 mV_

The displayed low battery percentage immediately after pairing is due to the fact that at the sensor configuration stage the `[genPowerCfg] batteryVoltage` attribute is read, the value of which is the battery voltage in Volts with an accuracy of tenths (for example, _2.8 V = 2800 mV_). And then about 55 minutes later a heartbeat-message arrives containing a more accurate battery voltage value (for example, _2879 mV_) and the displayed battery percentage, respectively, increases noticeably.

---

Created a [PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/4484) to add support for this sensor.

1000010000, 2022-07-26:
> In this log, voltage fluctuations are small and this is normal:
> 
> 1,33,63,11 - it's 2879 mV
> 1,33,70,11 - it's 2886 mV
> 1,33,64,11 - it's 2881 mV
> 1,33,59,11 - it's 2875 mV
> 1,33,66,11 - it's 2882 mV

I see. Thank you very much. I think there is something wrong with my battery.

closed 2022-07-26 by Koenkk