Koenkk/zigbee2mqtt #13258: Bitron Smart Plug with Metering Power false

hko123, 2022-07-23:
### What happened?

Zigbee Modell: 902010/25
Zigbee Hersteller:Bitron Home

The Power value must still be divided by 10. This is no longer done.

Issue: Problem report "Unsupported Device: Bitron Smart Plug with Metering - seMetering support #770" the value was still handled accordingly.

### Zigbee2MQTT version

1.26.0 commit: cd23aea

### Adapter firmware version

20220219

### Adapter

zStack3x0

Koenkk, 2022-07-23:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['AV2010/25', '902010/25'],
    model: 'AV2010/25',
    vendor: 'SMaBiT (Bitron Video)',
    description: 'Wireless socket with metering',
    fromZigbee: [fz.on_off, fz.metering],
    toZigbee: [tz.on_off],
    exposes: [e.switch(), e.power(), e.energy()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'seMetering']);
        await reporting.instantaneousDemand(endpoint);
        await reporting.currentSummDelivered(endpoint);
        await reporting.currentSummReceived(endpoint);
        endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100000, multiplier: 1});
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

hko123, 2022-07-23:
I installed the fix.

There is no change. These error messages are in the log.

`info  2022-07-23 19:14:38: Successfully interviewed '0x00124b0007305e57', device has successfully been paired
info  2022-07-23 19:14:38: Device '0x00124b0007305e57' is supported, identified as: SMaBiT (Bitron Video) Wireless socket with metering (AV2010/25)
info  2022-07-23 19:14:38: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Wireless socket with metering","exposes":[{"features":[{"access":7,"description":"On/off state of the switch","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"description":"Instantaneous measured power","name":"power","property":"power","type":"numeric","unit":"W"},{"access":1,"description":"Sum of consumed energy","name":"energy","property":"energy","type":"numeric","unit":"kWh"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"AV2010/25","options":[],"supports_ota":false,"vendor":"SMaBiT (Bitron Video)"},"friendly_name":"0x00124b0007305e57","ieee_address":"0x00124b0007305e57","status":"successful","supported":true},"type":"device_interview"}'
info  2022-07-23 19:14:38: Configuring '0x00124b0007305e57'
info  2022-07-23 19:14:38: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Wireless socket with metering","friendly_name":"0x00124b0007305e57","model":"AV2010/25","supported":true,"vendor":"SMaBiT (Bitron Video)"},"type":"pairing"}'
error 2022-07-23 19:14:38: Failed to configure '0x00124b0007305e57', attempt 2 (Error: ConfigureReporting 0x00124b0007305e57/1 seMetering([{"attribute":"currentSummReceived","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":[1,1]}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28)
    at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:694:22)
    at Object.currentSummReceived (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:123:9)
    at Object.configure (/app/dist/util/externally-loaded.js:24:9)
    at Configure.configure (/app/lib/extension/configure.ts:115:13))`

hko123, 2022-07-23:
Oh sorry I just see that the error in the configuration is also there without the fix.

Therefore, I assume that my plugs are no longer properly supported.

I had already learned the plugs about 1 or 2 years ago and they worked without any problems.

My server died and now I'm reinstalling everything. Hence the problems.

Koenkk, 2022-07-24:
Can you try with the following?

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['AV2010/25', '902010/25'],
    model: 'AV2010/25',
    vendor: 'SMaBiT (Bitron Video)',
    description: 'Wireless socket with metering',
    fromZigbee: [fz.on_off, fz.metering],
    toZigbee: [tz.on_off],
    exposes: [e.switch(), e.power(), e.energy()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'seMetering']);
        await reporting.instantaneousDemand(endpoint);
        await reporting.currentSummDelivered(endpoint);
        //await reporting.currentSummReceived(endpoint);
        endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 10000, multiplier: 1});
    },
};

module.exports = definition;
```

hko123, 2022-07-24:
Now there are no more errors. But no power either.

`info  2022-07-24 18:52:16: Device '0x00124b0007305e57' joined
info  2022-07-24 18:52:16: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b0007305e57","ieee_address":"0x0
0124b0007305e57"},"type":"device_joined"}'
info  2022-07-24 18:52:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00124b0007305e57"},"type":"device_co
nnected"}'
info  2022-07-24 18:52:16: Starting interview of '0x00124b0007305e57'
info  2022-07-24 18:52:16: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b0007305e57","ieee_address":"0x0
0124b0007305e57","status":"started"},"type":"device_interview"}'
info  2022-07-24 18:52:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00124b000
7305e57"},"type":"pairing"}'
info  2022-07-24 18:52:16: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b0007305e57","ieee_address":"0x0
0124b0007305e57"},"type":"device_announce"}'
info  2022-07-24 18:52:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00124b0007305e57"}
,"type":"device_announced"}'
info  2022-07-24 18:52:16: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b0007305e57","ieee_address":"0x0
0124b0007305e57"},"type":"device_announce"}'
info  2022-07-24 18:52:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00124b0007305e57"}
,"type":"device_announced"}'
info  2022-07-24 18:52:16: Successfully interviewed '0x00124b0007305e57', device has successfully been paired
info  2022-07-24 18:52:16: Device '0x00124b0007305e57' is supported, identified as: SMaBiT (Bitron Video) Wireless socket with metering (AV2010/25)
info  2022-07-24 18:52:16: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Wireless socket with meteri
ng","exposes":[{"features":[{"access":7,"description":"On/off state of the switch","name":"state","property":"state","type":"binary","value_off":"OFF
","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"description":"Instantaneous measured power","name":"power","property":"powe
r","type":"numeric","unit":"W"},{"access":1,"description":"Sum of consumed energy","name":"energy","property":"energy","type":"numeric","unit":"kWh"}
,{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":2
55,"value_min":0}],"model":"AV2010/25","options":[],"supports_ota":false,"vendor":"SMaBiT (Bitron Video)"},"friendly_name":"0x00124b0007305e57","ieee
_address":"0x00124b0007305e57","status":"successful","supported":true},"type":"device_interview"}'
info  2022-07-24 18:52:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Wireless s
ocket with metering","friendly_name":"0x00124b0007305e57","model":"AV2010/25","supported":true,"vendor":"SMaBiT (Bitron Video)"},"type":"pairing"}'
info  2022-07-24 18:52:18: MQTT publish: topic 'zigbee2mqtt/0x00124b0007305e57', payload '{"linkquality":167,"state":"OFF"}'
nfo  2022-07-24 18:55:05: MQTT publish: topic 'zigbee2mqtt/0x00124b0007305e57', payload '{"linkquality":163,"state":"OFF"}'
info  2022-07-24 18:55:06: MQTT publish: topic 'zigbee2mqtt/0x00124b0007305e57', payload '{"linkquality":160,"state":"ON"}'
`

Koenkk, 2022-07-24:
Can you try to configure it via the frontend? Press the yellow refresh button on the device page.

hko123, 2022-07-25:
I thought a reboot was enough. Excuse me. 

Now there are errors again
But power is displayed again but with the wrong decimal place.

`info  2022-07-25 05:56:08: Configuring '0x00124b0007305e57'
error 2022-07-25 05:56:11: Failed to configure '0x00124b0007305e57', attempt 4 (Error: ConfigureReporting 0x00124b0007305e57/1 seMetering([{"attribut
e":"currentSummReceived","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":[1,1]}], {"sendWhen":"immediate","timeout":10000,"
disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":nul
l,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28)
    at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:694:22)
    at Object.currentSummReceived (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:123:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/bitron.js:139:13)
    at Configure.configure (/app/lib/extension/configure.ts:115:13)
    at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21))
info  2022-07-25 05:56:11: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"0x00124b0007305e57"},"error":"
Failed to configure (ConfigureReporting 0x00124b0007305e57/1 seMetering([{\"attribute\":\"currentSummReceived\",\"minimumReportInterval\":5,\"maximum
ReportInterval\":3600,\"reportableChange\":[1,1]}], {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false
,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null
,\"writeUndiv\":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE'))","status":"error","transaction":"gb02f-1"}'
info  2022-07-25 05:56:17: MQTT publish: topic 'zigbee2mqtt/0x00124b0007305e57', payload '{"linkquality":91,"power":0,"state":"OFF"}'`

The device has 250.1 W

`info  2022-07-25 06:14:27: MQTT publish: topic 'zigbee2mqtt/0x00124b0007305e57', payload '{"linkquality":94,"power":2501,"state":"ON"}'`

Koenkk, 2022-07-25:
The configure fail for `currentSummReceived` is unexpected since this shouldn't happen anymore with the converter from https://github.com/Koenkk/zigbee2mqtt/issues/13258#issuecomment-1193335036 , I've updated the converter; can you check if the converter is loaded  correctly? The description in the frontend should end with `CUSTOM` now.

hko123, 2022-07-25:
You were right. My mistake. The parameters in configuration.yaml had disappeared. Unfortunately, I cannot understand why. 

Sorry for the false reports.

Now everything works perfectly. 

Power shows the correct value and the total energy is also displayed.

Thank you for your patience.

Koenkk, 2022-07-25:
Thanks pushed the fix. 
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)