Koenkk/zigbee2mqtt #13263: Ikea Devices won't pair

andrewcfitz, 2022-07-23:
### What happened?

I can't get any of my zigbee ikea devices to pair to zigbee2mqtt, which I am running via the HA official addon.  I am trying to pair a TRADFRI signal repeater (E1746) and a FYRTUR roller blind (E1757), but nothing happens when I put zigbee2mqtt in pairing mode.  The devices are almost touching the sonoff dongle when pairing.

I was able to pair a Aqara Wireless Mini Switch and it worked fine.  My existing ikea devices that are already connected work as expected.

### What did you expect to happen?

I expect the devices to pair.

### How to reproduce it (minimal and precise)

1. Turn on pairing mode in zigbee2mqtt
2. Device
  a. Ikea Repeater - Hold the pair button for 5 seconds, and the pair led starts blinking
  b. Ikea Blinds - Hold both buttons for 5 seconds, and the pair led starts blinking
3. Nothing happens in zigbee2mqtt

### Zigbee2MQTT version

1.26.0-1

### Adapter firmware version

20210708

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Debug log

Info 2022-07-23 12:19:38Zigbee: allowing new devices to join.
Info 2022-07-23 12:19:39MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"d9ol3-1"}'

Koenkk, 2022-07-24:
In case you have a CC2531 adapter lying around, could you make a sniff when trying to pair the device? https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

andrewcfitz, 2022-07-24:
I do, and I will give it a try.

eigentlichich, 2022-08-03:
Exact same thing for me using a Sonoff Zigbee Dongle 3.0.

closed 2022-09-10 by github-actions[bot]