Koenkk/zigbee2mqtt #13264: Neo NAS-PD07  Motion Sensor

Profixit, 2022-07-23:
### What happened?

After the latest update my Neo NAS-PD07 Motion Sensor is not recognised properly. It's recognised as a NEO camcool ZigBee Siren Alarm Tuya. Before the update it was fine.

Zigbee2mqtt Dev branch recognises it perfectly, so that's what I've started using.

[image]
.
Above Motion Sensor Recognised as below Siren.

[image]

### What did you expect to happen?

After updating, I didn't expect the device type to change to another device. 

### How to reproduce it (minimal and precise)

Every time you pair the device using Zigbee2mqtt standard.

### Zigbee2MQTT version

1.26.0-1

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

Koenkk, 2022-07-24:
Probably a duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/12710 and already fixed in the dev branch  (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Profixit, 2022-07-24:
Thanks for the reply.
I'm using the dev branch since yesterday, easy to switch with my setup. Odroid N2+. Installed the add-on, stopped the standard one, started the Dev branch and it took over. Very impressed.
It is recognised but not actually working. I was hoping it would populate exposed values over night. Unfortunately not. No occupancy showing just N/A on all exposed.
No biggie, I only have one.
Thanks again.

Profixit, 2022-07-24:
I've given the other issue a look. I need time to make sense of it. I'll give it some study. I can see you've pushed the fix through in Dev branch.
Recognised, but not working. I'll give more testing...

petergerdes, 2022-08-21:
Same issue here on docker-dev-image (pulled this morning), only state passed is linkquality.

closed 2022-09-28 by github-actions[bot]

Profixit, 2022-12-08:
No cure. Even the with the latest Version Standard or Dev. Still the same. No Biggie. Only got one of these.